margelo/react-native-vision-camera #1869: ❓ FrameProcessors are disabled because you're using an incompatible version of reanimated

hmyslf, 2023-09-27:
### Question

My apps worked before, but after I tried the troubleshooting solution from this site: https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting/ , suddenly it stopped working and showed this warning.

I've been trying to figure out how to fix this, but nothing has been successful, so I decided to ask here, here's the screenshot of the problem:

[image]

### What I tried

I've tried upgrading and downgrading the reanimated and vision camera version, but it still cant run the frameprocessor, what could be possibly causing this?

### VisionCamera Version

2.16.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-09-30:
Hi yea you need REA 3.5

closed 2023-09-30 by mrousavy

hmyslf, 2023-10-02:
But it worked fine before? when upgraded to said version, I got this error: 

`Execution failed for task ':react-native-vision-camera:configureCMakeDebug[arm64-v8a]'.
> [CXX1405] error when building with cmake using /Document/android/mobile-sris-2.0/node_modules/react-native-vision-camera/android/CMakeLists.txt: Build command failed.
  Error while executing process /Document/cmake/3.18.1/bin/cmake with arguments {-H/Document/android/mobile-sris-2.0/node_modules/react-native-vision-camera/android -DCMAKE_SYSTEM_NAME=Android -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_SYSTEM_VERSION=21 -DANDROID_PLATFORM=android-21 -DANDROID_ABI=arm64-v8a -DCMAKE_ANDROID_ARCH_ABI=arm64-v8a -DANDROID_NDK=/Document/ndk/21.4.7075529 -DCMAKE_ANDROID_NDK=/Document/ndk/21.4.7075529 -DCMAKE_TOOLCHAIN_FILE=/Document/ndk/21.4.7075529/build/cmake/android.toolchain.cmake -DCMAKE_MAKE_PROGRAM=/Document/cmake/3.18.1/bin/ninja -DCMAKE_CXX_FLAGS=-fexceptions -frtti -std=c++1y -DONANDROID -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/Document/android/mobile-sris-2.0/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/30265s55/obj/arm64-v8a -DCMAKE_RUNTIME_OUTPUT_DIRECTORY=/Document/android/mobile-sris-2.0/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/30265s55/obj/arm64-v8a -DCMAKE_BUILD_TYPE=Debug -B/Document/android/mobile-sris-2.0/node_modules/react-native-vision-camera/android/.cxx/Debug/30265s55/arm64-v8a -GNinja -DANDROID_STL=c++_shared -DREACT_NATIVE_VERSION=70 -DNODE_MODULES_DIR=/Document/android/mobile-sris-2.0/node_modules -DFOR_HERMES=true -DJS_RUNTIME_DIR=/Document/android/mobile-sris-2.0/node_modules/react-native/sdks/hermes}
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /Document/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64/bin/clang - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /Document/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Configuring incomplete, errors occurred!
  See also "/Document/android/mobile-sris-2.0/node_modules/react-native-vision-camera/android/.cxx/Debug/30265s55/arm64-v8a/CMakeFiles/CMakeOutput.log".
  
  CMake Error at CMakeLists.txt:208 (find_package):
    Could not find a package configuration file provided by
    "react-native-reanimated" with any of the following names:
  
      react-native-reanimatedConfig.cmake
      react-native-reanimated-config.cmake
`

edersonmberti, 2023-10-04:
> But it worked fine before? when upgraded to said version, I got this error:
>
> `Execution failed for task ':react-native-vision-camera:configureCMakeDebug[arm64-v8a]'.
>
>
> CMake Error at [...]

I'm having the same issue after update React Native Reanimated to 3.5.4 version.

Version:
"react-native": "0.68.6"
"react-native-reanimated": "3.5.4",
"react-native-vision-camera": "^2.16.1"

I tried to use the CMake 3.18.1 and 3.22.1 version and both didn't work. 🥲

mshirzad2, 2023-10-21:
Same error!!!!

mssoliveira, 2023-11-07:
same problem!