margelo/react-native-vision-camera #1870: 🐛 App crashes in release version

MrSolimanKing, 2023-09-27:
### What's happening?

I am using the latest version of react-native-vision-camera@3.1.0, The app is working fine on a Physical iPhone 12 and iPhone 13 pro max device, but It crashes in the Xcode simulator I didn't care since it working on actual physical device, "In knowing that the older version @3.0.0 didn't crashes in the simulator". The version @3.1.0 I tried to build the app in release mode and When I opened the camera the app crashed, 

This Error came from Firebase Crashlytics -> 
closure #4 in CameraView.didSetProps(_:)
NSInvalidArgumentException - -[__NSCFBoolean _fastCStringContents:]: unrecognized selector sent to instance <address>
```

Full Stack Trace

### Fatal Exception: NSInvalidArgumentException
0  CoreFoundation                 0x9cb4 __exceptionPreprocess
1  libobjc.A.dylib                0x183d0 objc_exception_throw
2  CoreFoundation                 0x17eab8 +[NSObject(NSObject) _copyDescription]
3  CoreFoundation                 0x200e8 ___forwarding___
4  CoreFoundation                 0x86900 _CF_forwarding_prep_0
5  libswiftCore.dylib             0x1fe1b0 _bridgeCocoaString(_:)
6  libswiftCore.dylib             0x1fe3e0 String.init(_cocoaString:)
7  Foundation                     0x26f660 static String._unconditionallyBridgeFromObjectiveC(_:)
8  App_name                      0x3c29a4 closure #4 in CameraView.didSetProps(_:) + 211 (CameraView.swift:211)
9  App_name                      0x3c323c partial apply for closure #4 in CameraView.didSetProps(_:) (<compiler-generated>)
10 App_name                      0x2d410 thunk for @escaping @callee_guaranteed () -> () (<compiler-generated>)
11 libdispatch.dylib              0x2320 _dispatch_call_block_and_release
12 libdispatch.dylib              0x3eac _dispatch_client_callout
13 libdispatch.dylib              0xb534 _dispatch_lane_serial_drain
14 libdispatch.dylib              0xc0a4 _dispatch_lane_invoke
15 libdispatch.dylib              0x16cdc _dispatch_workloop_worker_thread
16 libsystem_pthread.dylib        0xddc _pthread_wqthread
17 libsystem_pthread.dylib        0xb7c start_wqthread 
###

This Error causes the simulator crash:
#7	0x000000010517aba8 in CameraDevicesManager.constantsToExport() at app_name/node_modules/react-native-vision-camera/ios/CameraDevicesManager.swift:51

 override func constantsToExport() -> [AnyHashable: Any]! {
    let devices = getDevicesJson()
    let preferredDevice: [String: Any]
    // TODO: Remove this #if once Xcode 15 is rolled out
    #if swift(>=5.9)
      if #available(iOS 17.0, *),
         let userPreferred = AVCaptureDevice.userPreferredCamera {
        preferredDevice = userPreferred.toDictionary()
      } else {
        preferredDevice = devices[0]
      }
    #else
      preferredDevice = devices[0]
    #endif

  

### Reproduceable Code

```tsx
function CameraComponent(props){
    const { cameraType, flashMode, cameraRef, enableAudio, cameraZoomValue, isReady, selectedSound, captureAudio, videoStabilization } = props;

    useEffect(async () => {
      await Camera.requestCameraPermission()
      await Camera.requestMicrophonePermission()
    }, [])

    

    const [enableNightMode, setEnableNightMode] = useState(false);
    
    const dimensions = Dimensions.get("window")
    const screenWidth = dimensions.width;
    const height = Math.round((screenWidth * 16) / 8.25);

    const insets = useSafeAreaInsets();

    const device = useCameraDevice(cameraType === 1 ? 'back' : 'front', { physicalDevices: ['wide-angle-camera'] })
    const flash = flashMode === 0 ? 'off' : 'on'

    const format = useMemo(() => {
        return device?.formats.reduce((prev, curr) => {
        // console.log(prev)
        
          if (prev == null) return curr
          // // if (curr.maxFps > prev.maxFps && curr.videoWidth === 1080 && curr.videoHeight === 1920) return curr
          if (curr.maxFps > (prev.maxFps + 30)) return curr
          // if (curr.maxFps > 120) return curr
          else return prev

      

        }, undefined)
        
      }, [device?.formats])

      async function handlerGestureEvent(event){
        if(event.nativeEvent.state === State.BEGAN){
         
        }
        else if(event.nativeEvent.state === State.ACTIVE){
            
            await cameraRef.current.focus({ x: event.nativeEvent.x, y: event.nativeEvent.y })
        } else if(event.nativeEvent.state === State.END){
          
        }
    }

    

    const supportsVideoStabilization = format.videoStabilizationModes.includes(videoStabilization)

    return (
        <View style={{ top: insets.top }}>
          <TapGestureHandler onHandlerStateChange={handlerGestureEvent}>
      
          <ReanimatedCamera
            style={[{           
                height: height,
                width: "100%",
                zIndex: -2,
                borderRadius: 30,
                overflow: 'hidden'
            }]}
            ref={cameraRef}
            lowLightBoost={device.supportsLowLightBoost && enableNightMode}
            device={device}
            isActive={true}
            torch={flash}
            photo={true}
            zoom={Math.abs(cameraZoomValue * 1000)}
            video={true}
            audio={captureAudio}
            format={format}
            videoStabilizationMode={supportsVideoStabilization}
            fps={30}
            enableZoomGesture
            orientation="portrait"

        />
        </TapGestureHandler>
        </View>
    )
}

export default CameraComponent;
```

### Relevant log output

```shell
Error Came from the simulator 

2023-09-27 12:13:42.757019+0300 app_name[27573:149699] [javascript] true
Swift/ContiguousArrayBuffer.swift:600: Fatal error: Index out of range
2023-09-27 12:13:42.858933+0300 app_name[27573:149699] Swift/ContiguousArrayBuffer.swift:600: Fatal error: Index out of range
```

### Camera Device

```json
Because there are no logs in release mode, I can't see the camera formats
```

### Device

iPhone 12 16.6.1

### VisionCamera Version

3.1.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-27:
Format your code properly. Use three backticks: ```

mrousavy, 2023-09-27:
If you're talking about the simulator crash, I just fixed that.

closed 2023-09-27 by mrousavy

MrSolimanKing, 2023-09-27:
Oh, my bad I write on GitHub rarely, Thank you.

MrSolimanKing, 2023-09-27:
Not Only The simulator but the app in release mode on Physical devices (iPhone 12 and iPhone 13 pro max) I tried the app on Development mode and it worked perfectly but When I tried the app on TestFlight after archiving (also Release build) it crashed

MrSolimanKing, 2023-09-27:
@mrousavy 

I tried to run The app on my Physical device iPhone 12 and as connected to Xcode, When I opened the camera the app crashed this was out of Xcod Log   

```
3   app_name                           0x00000001010e2928 -[SRRunLoopThread runLoop] + 36
4   app_name                           0x00000001010df1c4 +[NSRunLoop(SRWebSocket) SR_networkRunLoop] + 40
5   app_name                           0x00000001010e1b40 -[SRProxyConnect _openConnection] + 56
6   app_name                           0x00000001010e1444 -[SRProxyConnect _configureProxy] + 468
7   app_name                           0x00000001010e3780 -[SRWebSocket open] + 296
8   app_name                           0x0000000100fed748 -[RCTWebSocketModule connect:protocols:options:socketID:] + 744
9   CoreFoundation                      0x000000019ca25c04 A900B459-0127-379E-9CBA-0EAB9C5D559F + 470020
10  CoreFoundation                      0x000000019c9d3cb4 A900B459-0127-379E-9CBA-0EAB9C5D559F + 134324
11  CoreFoundation                      0x000000019c9d36cc A900B459-0127-379E-9CBA-0EAB9C5D559F + 132812
12  app_name                           0x0000000100fa84b0 -[RCTModuleMethod invokeWithBridge:module:arguments:] + 388
13  app_name                           0x0000000100faa694 _ZN8facebook5reactL11invokeInnerEP9RCTBridgeP13RCTModuleDatajRKN5folly7dynamicEiN12_GLOBAL__N_117SchedulingContextE + 456
14  app_name                           0x0000000100faa31c ___ZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEi_block_invoke + 112
15  libdispatch.dylib                   0x00000001032d8520 _dispatch_call_block_and_release + 32
16  libdispatch.dylib                   0x00000001032da038 _dispatch_client_callout + 20
17  libdispatch.dylib                   0x00000001032ea798 _dispatch_main_queue_drain + 1196
18  libdispatch.dylib                   0x00000001032ea2dc _dispatch_main_queue_callback_4CF + 44
19  CoreFoundation                      0x000000019ca4bc28 A900B459-0127-379E-9CBA-0EAB9C5D559F + 625704
20  CoreFoundation                      0x000000019ca2d560 A900B459-0127-379E-9CBA-0EAB9C5D559F + 501088
21  CoreFoundation                      0x000000019ca323ec CFRunLoopRunSpecific + 612
22  GraphicsServices                    0x00000001d7f4835c GSEventRunModal + 164
23  UIKitCore                           0x000000019edbef58 7D57A1D1-856F-338D-97DB-880C4EC8B02E + 3788632
24  UIKitCore                           0x000000019edbebbc UIApplicationMain + 340
25  app_name                           0x0000000100d5455c main + 80
26  dyld                                0x00000001bbf64dec C3FC2EE4-367F-3086-BEB8-420AD442BF88 + 89580
2023-09-27 15:05:23.926135+0300 app_name[27457:3616403] [javascript] source.uri should not be an empty string
2023-09-27 15:05:24.027733+0300 app_name[27457:3616403] [expo] 🟢 Creating JS object for module 'ExpoVideoView'
2023-09-27 15:05:27.598402+0300 app_name[27457:3616392] -[__NSCFBoolean _fastCStringContents:]: unrecognized selector sent to instance 0x1f4a78380
2023-09-27 15:05:27.599504+0300 app_name[27457:3616392] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFBoolean _fastCStringContents:]: unrecognized selector sent to instance 0x1f4a78380'
*** First throw call stack:
(0x19c9bccb4 0x195a543d0 0x19cb31ab8 0x19c9d30e8 0x19ca39900 0x1968da1b0 0x1968da3e0 0x196ed8660 0x1011129ac 0x101113244 0x100d7d410 0x1032d8520 0x1032da038 0x1032e20b0 0x1032e2df4 0x1032efc74 0x1fc7f1ddc 0x1fc7f1b7c)
libc++abi: terminating due to uncaught exception of type NSException
terminating due to uncaught exception of type NSException
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFBoolean _fastCStringContents:]: unrecognized selector sent to instance 0x1f4a78380'

Thread 9: "-[__NSCFBoolean _fastCStringContents:]: unrecognized selector sent to instance 0x1f4a78380"

libsystem_kernel.dylib`:
    0x1db979570 <+0>:  mov    x16, #0x148
    0x1db979574 <+4>:  svc    #0x80
->  0x1db979578 <+8>:  b.lo   0x1db979598               ; <+40>
    0x1db97957c <+12>: pacibsp 
    0x1db979580 <+16>: stp    x29, x30, [sp, #-0x10]!
    0x1db979584 <+20>: mov    x29, sp
    0x1db979588 <+24>: bl     0x1db9749a4               ; cerror_nocancel
    0x1db97958c <+28>: mov    sp, x29
    0x1db979590 <+32>: ldp    x29, x30, [sp], #0x10
    0x1db979594 <+36>: retab  
    0x1db979598 <+40>: ret    
```

MrSolimanKing, 2023-09-27:
I was able to fix the problem by removing the `VideoStabilization` 

 `const supportsVideoStabilization = format.videoStabilizationModes.includes(videoStabilization)`

`videoStabilizationMode={supportsVideoStabilization}`.

mrousavy, 2023-09-27:
Interesting, honestly I have no idea what the crash log means.

Siliconvelly, 2023-10-17:
I am also facing same issue after successfully installation of react-native-vision-camera i run my project on android it works fine but on ios it given error when i press run on xcode and build time i see the error 'userPreferredCamera' is unavailable in iOS.
i tried everything to fix the issue but nothing happened so please fix as soon as possible

mrousavy, 2023-10-17:
> so please fix as soon as possible

pay me.

mrousavy, 2023-10-17:
You need to upgrade to Xcode 15.

> i tried everything to fix the issue

except upgrading to Xcode 15 I assume

commit fb6ebd9e (2023-09-27): fix: Allow nil `preferredDevice`

Fixes #1870