margelo/react-native-vision-camera #1871: ❓ How to align the camera view with the screen orientation?

najamulsaqib, 2023-09-27:
### Question

When using the React Native Vision Camera in my app, I encounter an unexpected behavior where the camera rotation does not align correctly with the screen orientation. Specifically, when the device screen is switched to landscape mode, the camera rotates 90 degrees clockwise, causing an inconvenience for users trying to capture content in landscape orientation.

### What I tried

Here, in portrait mode (left image), it works fine, but when I turn my phone's orientation to landscape mode (right image), the camera view somehow rotates 90 degrees clockwise. When I turn the phone, the camera should not rotate; it should align with the orientation of my phone.

 Is there anything I'm missing?

[image]

[image]

### VisionCamera Version

2.15.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

tiste, 2023-09-27:
Hello, same behaviour here with camera or usb camera. On landscape mode, or default, in the `app.json`.

najamulsaqib, 2023-09-28:
Hey @tiste, I'm using the iPhone selfie camera, but it happens on both the front and back cameras, but only in landscape mode.

jkerb8, 2023-09-29:
I'm experiencing the same behavior in version 3.1.0 with RN 72.5 on both iOS and Android. When rotated in landscape, the video even appears warped. The result of the video appears normal (except it's in portrait orientation when it should be landscape), but the preview does not show the same result. These tests were done on a Pixel 6 running Android 13. The phone is locked to landscape using react-native-orientation-locker during this test.

Here a screenshot when I'm viewing the preview while holding the phone in landscape orientation:
[image]

If I rotate the device to portrait orientation, this is what I see. 
[image]

Here is the video of the end result:

https://github.com/mrousavy/react-native-vision-camera/assets/15618621/cfb7d867-f3b8-4887-8601-9ed7d3c70283

mrousavy, 2023-09-30:
Hey yea Orientation is quite tricky, I'll create a separate issue

closed 2023-09-30 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Orientation in this feature request/issue now: https://github.com/mrousavy/react-native-vision-camera/issues/1891 ✨

nijatkazimliPG, 2025-01-18:
@mrousavy, this still happens on Android.
My screen rotates, but preview output not. Resulting in 90 degree rotated image after changing orientation.