margelo/react-native-vision-camera #1879: ❓ Turning off torch mid-recording

choonghwanlee, 2023-09-28:
### Question

I'm currently using expo-camera for a project that requires me to a) turn torch upon recording start and b) turn torch off N seconds into the recording. Expo Camera unfortunately cannot support this feature so I'm thinking of ejecting into a bare React Native app and using React-Native-Vision-Camera. 

If I start a recording using camera.current.startRecording with the flash property of RecordVideoOptions set to "on", is it possible to toggle this off N seconds into the recording? What would this look like? I just want confirmation that this feature would work before committing to ejecting. 

### VisionCamera Version

3.1.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jkaufman, 2023-09-28:
I have an app that offers a torch toggle during video recording. Works well.

choonghwanlee, 2023-09-28:
> I have an app that offers a torch toggle during video recording. Works well.

@jkaufman Awesome, that's great to know! Just a quick question –– I was wondering how you configured the toggle. is it a on/off state variable that changes the RecordVideoOptions of camera.current.startRecording, or the CameraProps of the camera object itself?

jkaufman, 2023-09-28:
The latter. The button simply toggles the torch prop.

closed 2023-09-28 by choonghwanlee

victorkvarghese, 2023-09-29:
Do we need additional permissions to enable flash? I tried the example code with flash prop. Flash is not working for photo/video. Tried 3 different devices.

Any help would be appreciated

jkaufman, 2023-10-07:
I wanted to follow up and clarify that I can only confirm the behavior on v2, not v3.

zoobibackups, 2024-05-07:
how I on torch in react native Vision camera by click on button