margelo/react-native-vision-camera #1880: 🔧 Example app crashed on ios release build

basukiwinoto, 2023-09-28:
### How were you trying to build the app?

Example app crashed on ios release build.

### Full build logs

```tsx
2023-09-28 18:21:53.238066-0400 VisionCameraExample[25171:6547125] [native] Running application VisionCameraExample ({
    initialProps =     {
    };
    rootTag = 1;
})
2023-09-28 18:21:53.260174-0400 VisionCameraExample[25171:6547433] [javascript] Running "VisionCameraExample
2023-09-28 18:21:53.261669-0400 VisionCameraExample[25171:6547433] [javascript] Re-rendering Navigator. Camera: undefined | Microphone: undefined
2023-09-28 18:21:53.261991-0400 VisionCameraExample[25171:6547433] [javascript] Loading react-native-worklets-core...
2023-09-28 18:21:53.262525-0400 VisionCameraExample[25171:6547433] [javascript] Worklets loaded successfully
2023-09-28 18:21:53.262922-0400 VisionCameraExample[25171:6547433] VisionCameraProxy: Creating Worklet Context...
2023-09-28 18:21:53.263693-0400 VisionCameraExample[25171:6547433] VisionCameraProxy: Worklet Context Created!
2023-09-28 18:21:53.267375-0400 VisionCameraExample[25171:6547433] [javascript] Re-rendering Navigator. Camera: granted | Microphone: undefined
2023-09-28 18:21:53.267914-0400 VisionCameraExample[25171:6547433] [javascript] Re-rendering Navigator. Camera: granted | Microphone: granted
2023-09-28 18:21:53.318013-0400 VisionCameraExample[25171:6547433] [javascript] Re-rendering camera page with active camera. Device: "Back Triple Camera" (4224x2376 photo / 1920x1080 video @ 60fps)
2023-09-28 18:21:53.334218-0400 VisionCameraExample[25171:6547433] [javascript] Re-rendering camera page with active camera. Device: "Back Triple Camera" (4224x2376 photo / 1920x1080 video @ 60fps)
2023-09-28 18:21:53.385445-0400 VisionCameraExample[25171:6547433] [javascript] Camera initialized!
2023-09-28 18:21:53.385680-0400 VisionCameraExample[25171:6547433] [javascript] Re-rendering camera page with active camera. Device: "Back Triple Camera" (4224x2376 photo / 1920x1080 video @ 60fps)
libc++abi: terminating due to uncaught exception of type facebook::jsi::JSINativeException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
(lldb)
```

### Project dependencies

```json
the package.json from the example app
```

### VisionCamera Version

3.0.0

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rodgomesc, 2023-09-28:
hey @basukiwinot there's a few duplicates of this issue in #1776 #1714

basukiwinoto, 2023-09-28:
Maybe related but not the same. The issue I'm having is on ios release build. Debug build works fine.

closed 2023-09-28 by basukiwinoto