margelo/react-native-vision-camera #1881: Android: Execution failed for task ':app:checkDevDebugDuplicateClasses'

vvodicka, 2023-09-29:
### How were you trying to build the app?

`npx react-native run-android --variant devDebug --appIdSuffix dev --active-arch-only`

Nothing special...perhaps V3 is not meant to be used with older RN? If yes, maybe it's wort mentioning it in requirements.

### Full build logs

```tsx
1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:checkDevDebugDuplicateClasses'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckDuplicatesRunnable
   > Duplicate class com.facebook.debug.debugoverlay.model.DebugOverlayTag found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.debug.holder.NoopPrinter found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.debug.holder.Printer found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.debug.holder.PrinterHolder found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.debug.tags.ReactDebugOverlayTags found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.fbreact.specs.NativeAccessibilityInfoSpec found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.fbreact.specs.NativeAccessibilityManagerSpec found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.fbreact.specs.NativeActionSheetManagerSpec found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.fbreact.specs.NativeAlertManagerSpec found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     Duplicate class com.facebook.fbreact.specs.NativeAnimatedModuleSpec found in modules jetified-react-android-0.72-debug.5-debug-runtime (com.facebook.react:react-android:0.72.5) and jetified-react-native-0.68.3-runtime (com.facebook.react:react-native:0.68.3)
     ...

Execution failed for task ':react-native-vision-camera:configureCMakeDebug'.
> [CXX1405] error when building with cmake using /Users/vladislavvodicka/work/powerful_medical/cardio/node_modules/react-native-vision-camera/android/CMakeLists.txt: Build command failed.
  Error while executing java process with main class com.google.prefab.cli.AppKt with arguments {--build-system cmake --platform android --abi x86_64 --os-version 28 --stl c++_shared --ndk-version 21 --output /Users/vladislavvodicka/work/powerful_medical/cardio/node_modules/react-native-vision-camera/android/.cxx/Debug/45r1v224/prefab/x86_64/prefab /Users/vladislavvodicka/.gradle/caches/transforms-3/700f3c1069c300241d9bab24ea29ea28/transformed/jetified-react-android-0.72.5-debug/prefab /Users/vladislavvodicka/.gradle/caches/transforms-3/6ca9e354b9be47c245bd4a36fb1123be/transformed/jetified-fbjni-0.3.0/prefab}
  
  Exception in thread "main" java.lang.IllegalArgumentException: Only schema_version 1 is supported. ReactAndroid uses version 2.
        at com.google.prefab.api.Package.<init>(Package.kt:46)
        at com.google.prefab.cli.Cli$packages$2.invoke(Cli.kt:124)
        at com.google.prefab.cli.Cli$packages$2.invoke(Cli.kt:95)
        at kotlin.SynchronizedLazyImpl.getValue(LazyJVM.kt:74)
        at com.google.prefab.cli.Cli.getPackages(Cli.kt)
        at com.google.prefab.cli.Cli.validate(Cli.kt:172)
        at com.google.prefab.cli.Cli.run(Cli.kt:189)
        at com.github.ajalt.clikt.parsers.Parser.parse(Parser.kt:168)
        at com.github.ajalt.clikt.parsers.Parser.parse(Parser.kt:16)
        at com.github.ajalt.clikt.core.CliktCommand.parse(CliktCommand.kt:258)
        at com.github.ajalt.clikt.core.CliktCommand.parse$default(CliktCommand.kt:255)
        at com.github.ajalt.clikt.core.CliktCommand.main(CliktCommand.kt:273)
        at com.github.ajalt.clikt.core.CliktCommand.main(CliktCommand.kt:298)
        at com.google.prefab.cli.AppKt.main(App.kt:28)
```

### Project dependencies

```json
"dependencies": {
    "@aws-sdk/client-cognito-identity-provider": "^3.56.0",
    "@babel/runtime": "^7.17.9",
    "@expo-google-fonts/ibm-plex-sans": "^0.1.0",
    "@hookform/resolvers": "^0.1.1",
    "@jsonforms/core": "^2.5.2",
    "@jsonforms/react": "^2.5.2",
    "@notifee/react-native": "5.7.0",
    "@qeepsake/react-native-image-exif": "^1.0.1",
    "@react-native-async-storage/async-storage": "~1.15.17",
    "@react-native-clipboard/clipboard": "^1.11.1",
    "@react-native-community/blur": "^4.2.0",
    "@react-native-community/netinfo": "7.1.3",
    "@react-native-firebase/analytics": "^14.7.0",
    "@react-native-firebase/app": "^14.7.0",
    "@react-native-firebase/messaging": "^14.7.0",
    "@react-native-masked-view/masked-view": "^0.2.6",
    "@react-navigation/bottom-tabs": "^6.2.0",
    "@react-navigation/elements": "^1.3.1",
    "@react-navigation/native": "^6.0.8",
    "@react-navigation/stack": "^6.1.1",
    "@sentry/react-native": "5.6.0",
    "ajv": "^8.11.0",
    "ajv-i18n": "^4.2.0",
    "axios": "^1.4.0",
    "camelcase-keys": "^6.2.2",
    "d3fc-sample": "^3.0.5",
    "dotenv-parse-variables": "^2.0.0",
    "expo": "^46.0.15",
    "expo-linear-gradient": "~11.4.0",
    "expo-screen-orientation": "^4.3.0",
    "expo-secure-store": "~11.1.1",
    "expo-splash-screen": "^0.16.2",
    "expo-status-bar": "~1.2.0",
    "i18next": "^19.9.2",
    "intl": "^1.2.5",
    "json-schema-ref-parser": "7.1.3",
    "jwt-decode": "^3.1.2",
    "lodash": "^4.17.21",
    "promise.allsettled": "^1.0.5",
    "react": "17.0.2",
    "react-hook-form": "6.15.8",
    "react-i18next": "11.8.10",
    "react-native": "0.68.3",
    "react-native-auto-scrolling": "3.0.0",
    "react-native-blob-util": "^0.14.1",
    "react-native-charts-wrapper": "^0.5.7",
    "react-native-device-info": "^8.7.1",
    "react-native-dotenv": "^2.6.2",
    "react-native-elements": "3.3.2",
    "react-native-email-link": "^1.13.1",
    "react-native-error-boundary": "^1.1.12",
    "react-native-fs": "^2.19.0",
    "react-native-gesture-handler": "^2.9.0",
    "react-native-get-random-values": "~1.7.2",
    "react-native-iap": "^12.10.7",
    "react-native-image-crop-picker": "^0.38.0",
    "react-native-image-picker": "^4.8.2",
    "react-native-keyboard-state-provider": "^1.0.1",
    "react-native-localize": "^2.2.1",
    "react-native-modal": "^13.0.1",
    "react-native-notifications": "^4.3.3",
    "react-native-pager-view": "^6.2.0",
    "react-native-pdf": "6.6.2",
    "react-native-reanimated": "~2.12.0",
    "react-native-safe-area-context": "3.3.2",
    "react-native-screens": "~3.10.2",
    "react-native-sensors": "^7.3.6",
    "react-native-share": "^7.3.9",
    "react-native-svg": "^13.0.0",
    "react-native-swipe-list-view": "^3.2.9",
    "react-native-tab-view": "^3.1.1",
    "react-native-url-polyfill": "^1.3.0",
    "react-native-vector-icons": "^9.2.0",
    "react-native-vision-camera": "^3.1.0",
    "react-native-webview": "11.23.1",
    "react-query": "3.17.0",
    "snakecase-keys": "^3.2.1",
    "styled-components": "^5.3.5",
    "url": "^0.11.0",
    "yup": "^0.29.3"
  },
```

### VisionCamera Version

3.1.0

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-30:
This should be fixed with latest

closed 2023-09-30 by mrousavy

vvodicka, 2023-10-02:
> This should be fixed with latest

with latest you mean 3.2.2? because I tried it and the issue is still there.

mrousavy, 2023-10-03:
I dont see any error related to Visioncamera in your logs though

vvodicka, 2023-10-09:
with 3.3.1 the first problem with duplicate classes disappear but the second one `Only schema_version 1 is supported. ReactAndroid uses version 2.` stays. 

```Execution failed for task ':react-native-vision-camera:configureCMakeDebug'.
> [CXX1405] error when building with cmake using [...]/node_modules/react-native-vision-camera/android/CMakeLists.txt: Build command failed.
  Error while executing java process with main class com.google.prefab.cli.AppKt with arguments {--build-system cmake --platform android --abi x86_64 --os-version 28 --stl c++_shared --ndk-version 21 --output [...]/node_modules/react-native-vision-camera/android/.cxx/Debug/45r1v224/prefab/x86_64/prefab [...]/.gradle/caches/transforms-3/34f0f525c4d305edfe2d9d9d24bd6b0f/transformed/jetified-react-android-0.73.0-rc.1-debug/prefab [...]/.gradle/caches/transforms-3/f35e3e913dc614162c5b29b3a1444056/transformed/jetified-fbjni-0.5.1/prefab}
  
  Exception in thread "main" java.lang.IllegalArgumentException: Only schema_version 1 is supported. ReactAndroid uses version 2.
```

vvodicka, 2023-10-09:
additional info: when I tried to create minimal example project for this issue I encountered other issue:

```Could not determine the dependencies of task ':app:processDebugResources'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Searched in the following locations:
       - file:/Users/vladislavvodicka/others/camera/node_modules/react-native/android/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/vladislavvodicka/others/camera/node_modules/jsc-android/dist/com/facebook/react/react-android/maven-metadata.xml
       - https://dl.google.com/dl/android/maven2/com/facebook/react/react-android/maven-metadata.xml
       - https://www.jitpack.io/com/facebook/react/react-android/maven-metadata.xml
     Required by:
         project :app > project :react-native-vision-camera
```

created repo for it: https://github.com/vvodicka/react-native-vision-camera-android-build-error

iskandarzhilmi, 2023-10-19:
@vvodicka Are you by any chance using monorepo?

vvodicka, 2023-10-19:
Nope, I am not.

nikoloz110, 2023-11-09:
I have the same issue, anybody got fix or updates regarding this? @vvodicka