margelo/react-native-vision-camera #1883: 🐛 Front camera not detected at all on Xiaomi Mi Mix 2S

fobos531, 2023-09-29:
### What's happening?

Hello, 

I'm trying to integrate this library into my app over `expo-camera`. iOS works great for both back and rear camera, but on Android on this specific device (Xiaomi Mi Mix 2S) I am only able to get the back camera recognized. Front camera always defaults to the back camera instead.

### Reproduceable Code

```tsx
const cameraDevice = useCameraDevice("front")

 const cameraFormat = useCameraFormat(cameraDevice, [
    {
      videoStabilizationMode: 'cinematic-extended',
    },
  ]);

<Camera
ref={camera.cameraRef}
          device={cameraDevice}
          isActive={true}
          video={true}
          audio={true}
          format={cameraFormat}
          videoStabilizationMode={
            cameraFormat?.videoStabilizationModes.includes('cinematic-extended') ? 'cinematic-extended' : 'off'
          }
```

### Relevant log output

```shell
This is when I open the Camera screen and explicitly specify that I want to fetch the "front" camera devices.

https://gist.github.com/fobos531/f1f5766ef8a04158253b8cc5b21cfa7d
```

### Camera Device

```json
cameraDevice {
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Xiaomi Mi Mix 2S

### VisionCamera Version

3.1.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fobos531, 2023-09-29:
To reiterate - it seems VisionCamera ONLY recognizes the back camera, it does not at all recognize the front camera. The front camera is recognized normally in `expo-camera`.

mrousavy, 2023-09-29:
Wait in `expo-camera` it does actually find the front Camera, but in VisionCamera it doesn't? This sounds odd, normally VisionCamera detects more Cameras than expo-camera.

Can you share the output of this code:

```ts
const devices = Camera.getAvailableCameraDevices()
console.log(`VisionCamera detected ${devices.length} devices:`)
for (const d of devices) {
  d.formats = []
}
console.log(JSON.stringify(devices, null, 2))
```

And check if the front Camera actually is listed in there?

fobos531, 2023-09-29:
Hmm,

so - that method does indeed recognize the camera.

However, it seems that on this specific device the front camera is a **telephoto** camera. I suppose that on the vast majority of them it is actually a wide angle camera?

So if I do this:

```
  const cameraDevice = useCameraDevice('front', {
    physicalDevices: ['telephoto-camera'],
  });
```

Then the camera device is finally correctly picked up.

But if I just do this:

```
  const cameraDevice = useCameraDevice('front');
```
I expected that the camera device would get inferred automatically in this case and that VisionCamera would just fetch whatever physical device was available. Is this something that's missing from the API currently?

Also, I'm not sure if the below issue is unrelated or not, but listing it just for visibility:

Is there a recommended API usage for switching between the front/back camera in the viewfinder (i.e. NOT while recording, but **before** that)?

I experimented with the following approach: 

`const cameraDevice = useCameraDevice(cameraDirection)` where `cameraDirection` is a `"back" | "front"` and then utilizing `useState` to switch between the camera devices. While this works perfectly on iOS, it seems to cause crashes on Android, see attached logcat: https://gist.github.com/fobos531/8f9acfdfbc32a7b0c471918fe40f30c7

The application crashes at 
```
java.lang.IllegalArgumentException: CaptureRequest contains unconfigured Input/Output Surface!
                                                                                                    	at android.hardware.camera2.CaptureRequest.convertSurfaceToStreamId(CaptureRequest.java:738)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:1087)
```

Please, let me know if there's anything else I can do to help!

mrousavy, 2023-09-29:
Regarding point 4:

1. This is where the session gets re-built if you switch the device: https://github.com/mrousavy/react-native-vision-camera/blob/9bad0bb99eaf0ad88e6d045a7d46f585d510d815/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L511-L561
2. And this is where I update the capture request: https://github.com/mrousavy/react-native-vision-camera/blob/9bad0bb99eaf0ad88e6d045a7d46f585d510d815/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L563-L581

As you can see it's both under a mutex, so I wonder where this is coming from. Are you taking a photo or recording a video while flipping the camera or does this just happen in the preview?

fobos531, 2023-10-02:
Hey Marc,

It seems like https://github.com/mrousavy/react-native-vision-camera/commit/edd73619c3bde3f9c4eebb954e5c539f9f3db8da resolved the issue. As the initially reported issue seems to be fixed, I'll close this issue.

For the rest of the issues that we discussed (mainly point 4), these are slightly offtopic in the context of this issue, so I'll likely open a separate issue for them to continue the discussion there when I catch some time to do so. 

But to answer your question straight away,

> As you can see it's both under a mutex, so I wonder where this is coming from. Are you taking a photo or recording a video while flipping the camera or does this just happen in the preview?

I haven't tested it while taking a photo/recording a video, but it does indeed happen in the preview.

[image]

P.S. Let me know if you still need me to check this or not.

closed 2023-10-02 by fobos531

mrousavy, 2023-10-02:
Thanks for reporting back!

Yes please open the issues when you have some time so we can discuss a fix there!

And no, I don't need you to test the focal length breakpoint now, thanks. :)

fobos531, 2023-10-17:
Hey @mrousavy ,

just wanted to touch on the `CaptureRequest contains unconfigured Input/Output Surface!` that was talked about later in this ticket. 

I tried 3.4.0 and it seems to have fixed all related issues. I can know switch between front/back camera in the viewfinder/preview without any issues. Also, pre < 3.4.0 had some serious issues on certain Samsung devices like the Galaxy S20+ where the app would crash instantly on launch (with no useful logcat info at all) - also happy to report that with 3.4.0 such issues are also completely gone and I have yet to see a crash of this nature on Samsung devices.

Not sure if placebo, but the overall camera experience is super fast now on 5-year-old Android flagships.

Great work on the 3.4.0 release, many thanks! 🥇 🥇 🥇

mrousavy, 2023-10-18:
Awesome that is really nice to hear!!! Thanks for letting me know @fobos531 :)