margelo/react-native-vision-camera #1885: Does the video format affect taking a photo❓

stevengoldberg, 2023-09-29:
### Question

I recently updated to expo SDK49 and vision-camera v 3.1.0 from 2.15 and I'm confused by the behavior of the camera. I'm trying to take a 4:3 photo in maximum quality while also minimizing delay. I'm testing with an iPhone 14 Pro.

This is how I'm selecting a device:

```
 const physicalDevices =
        cameraPosition === 'back'
            ? [
                  'ultra-wide-angle-camera',
                  'wide-angle-camera',
                  'telephoto-camera',
              ]
            : ['wide-angle-camera']

    const device = useCameraDevice(cameraPosition, { physicalDevices })

    const format = useCameraFormat(device, [
        {
            photoAspectRatio: 4 / 3,
        },
        { photoResolution: 'max' },
        { videoAspectRatio: 4 / 3 },
        { videoResolution: 'max' },
    ])
```
And this is how I'm taking the photo:
```
camera.current
            .takePhoto({
                qualityPrioritization: 'quality',
                enableAutoStabilization: true,
            })
```
When I take a photo like this, there's a noticeable delay between pressing the button and the load of the resulting image, but the photo looks good. The image is very sharp, even if the camera is moving as I take it. The iOS photos app reports it as 12MP, 3024x4032, 3.8MB, taken with the Telephoto Camera.

Now, if I change the format filter to this:
```
    const format = useCameraFormat(device, [
        {
            photoAspectRatio: 4 / 3,
        },
        { photoResolution: 'max' },
        { videoAspectRatio: 4 / 3 },
        { videoResolution: { height: 1944, width: 2592 } },
    ])
```
Taking the photo results in a much smaller delay, but the image quality is worse — it's very blurry, like image stabilization is not being used. My phone says this photo is taken by the same camera, same resolution, but has a smaller filesize: 1.8 MB.

Is this intended behavior? I find it surprising that the video settings would affect taking a photo.

### VisionCamera Version

3.1.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-09-30:
Hey!

No, photo resolution should not affect video resolution or vice versa.

What you're seeing is technically a bug, so it would've been great if you filled out the bug template as it asks you a few more relevant questions, but now I'm just gonna ask them myself:

1. Can you please share the formats that are being chosen (`console.log(JSON.stringify(format))`), so both the good one and the low quality one
2. Is that iOS or Android?

stevengoldberg, 2023-09-30:
Sure — this is an iPhone 14 Pro running iOS 16.6.1.

Fast result / blurry photo:
```
const format = useCameraFormat(device, [
        {
            photoAspectRatio: 4 / 3,
        },
        { photoResolution: 'max' },
    ])

// {"minFps":60,"maxFps":60,"videoHeight":144,"supportsVideoHDR":false,"photoHeight":3024,"maxZoom":189,"pixelFormats":["yuv","yuv","rgb","yuv","unknown","yuv","unknown","rgb","unknown"],"videoStabilizationModes":["auto","off"],"supportsPhotoHDR":false,"autoFocusSystem":"phase-detection","photoWidth":4032,"minISO":57,"maxISO":5472,"supportsDepthCapture":false,"videoWidth":192,"fieldOfView":98.91507720947266}
```

Slower result / sharp photo:
```
const format = useCameraFormat(device, [
        {
            photoAspectRatio: 4 / 3,
        },
        { photoResolution: 'max' },
        { videoAspectRatio: 4 / 3 },
        { videoResolution: 'max' },
    ])

// {"supportsDepthCapture":false,"maxISO":12768,"minFps":30,"fieldOfView":101.3733901977539,"maxFps":30,"maxZoom":189,"videoStabilizationModes":["auto","off"],"pixelFormats":["yuv","yuv","rgb","yuv","unknown","yuv","unknown","rgb","unknown"],"supportsPhotoHDR":false,"autoFocusSystem":"phase-detection","minISO":57,"videoHeight":3024,"supportsVideoHDR":false,"photoHeight":3024,"videoWidth":4032,"photoWidth":4032}
```
The rest of the code is the same in both instances —

```
 <Camera
    ref={cameraRef}
    device={device}
    enableHighQualityPhotos
    enableAutoStabilization
    isActive={isActive}
    photo
    hdr
    format={format}
    zoom={device.neutralZoom}
/>

// ...

camera.current.takePhoto({
    qualityPrioritization: 'quality',
    enableAutoStabilization: true,
})
```

mrousavy, 2023-09-30:
Interesting!

1. Does the preview look like it has lower resolution in the first case?
2. Does the resulting photo actually have the dimensions that it claims in the format?
3. If ISO is the case, I need to investigate that more - I want to expose those props anyways.

stevengoldberg, 2023-09-30:
Yes, the preview looks like it matches the video resolution in the format, so it’s very visibly pixelated in the first example above when video resolution isn’t explicitly chosen. But the camera starts up very quickly!

And yes, the resulting photo always has the chosen resolution. The only difference I can see in the EXIF data seems to be ISO and exposure time.

mrousavy, 2023-09-30:
Awesome okay so looks like there's no issue with VisionCamera and everything works as expected!

I guess you want your video resolution to be somewhat close to the screen resolution at least.

As for ISO; I'll maybe do a PR in the coming days or weeks to implement explicit support for controlling that, but this is unrelated to your question as you could just filter for high ISO right?

Or actually now that I'm writing this I don't think that the filter supports ISO filtering so I'll just quickly add that to the useCameraFormat function and you should be good 👍

closed 2023-10-18 by mrousavy

nica0012, 2023-11-05:
I too am seeing a blurry camera preview, with a similar format.

Camera quality seems to be poor when I am using a different size camera view (square or 4:3).

mrousavy, 2023-11-29:
Yea, but it's way back on my TODO list since no one is sponsoring that. I'll focus on sponsored issues first :)

stevengoldberg, 2023-11-29:
Gotcha, no worries since it’s pretty straightforward to do manually anyway.

mrousavy, 2023-11-29:
@stevengoldberg see this for a WIP: https://github.com/mrousavy/react-native-vision-camera/pull/2234

commbits, 2024-04-13:
videoResolution clearly affects photoResolution: In the implementation below, photos are taken at 1024X768, although photoResolution='max'.
If we only change videoResolution: 'max', then photo resolution is going up to max indeed.
 const format = getCameraFormat(device, [
        { videoStabilizationMode: 'auto' },
        { photoAspectRatio: 4 / 3 },
        { videoAspectRatio: 4 / 3 },
        { lowLightBoost: true },
        { photoResolution: 'max' },
        {
           videoResolution: { width: 1024, height: 768 },
        },
    ])

stevengoldberg, 2024-04-13:
> videoResolution clearly affects photoResolution: In the implementation below, photos are taken at 1024X768, although photoResolution='max'. If we only change videoResolution: 'max', then photo [...]

I don't think video resolution affects photo resolution per se; it's more a side effect of how formats work. There is a finite number of formats with fixed options. `getCameraFormat` tries to select an appropriate format based on your criteria, but there may not be a format that matches all of the options. By changing the options object, you end up with a different format.

So the format on your device with a video resolution of 1024x768 doesn't have the highest photo resolution. When you choose the format with the max video resolution, it has a higher photo resolution as well. You can see this by logging all of the available device formats.

commbits, 2024-04-13:
I see. Thank you very much explaining.

How can we log all available formats?

Best regards,

Al Poullis,
Senior e-Business Consultant
Founder, COMMbits, Inc
5063 North Service Road, Suite 100
Burlington - ON, L7L 5H6
Phone: (416) 752-9513 x 100
Direct: (416) 876-4324
***@***.***
www.commbits.com
----------------------
The information transmitted is intended only for the person or entity to which it is addressed and may contain confidential and/or privileged material. Any review,  retransmission, dissemination or other use of, or taking of any action in reliance  upon, this information by persons or entities other than the intended recipient is  prohibited. If you received this in error, please contact the sender and delete the material from any computer.

stevengoldberg, 2024-04-13:
> I see. Thank you very much explaining. How can we log all available formats? 

See here: https://www.react-native-vision-camera.com/docs/guides/formats#choosing-custom-formats
https://www.react-native-vision-camera.com/docs/api/interfaces/CameraDevice#formats

mrousavy, 2024-04-14:
Exactly @stevengoldberg 👍🙌