margelo/react-native-vision-camera #1890: ✨ Implement Flash for Android

mrousavy, 2023-09-30:
### What feature or enhancement are you suggesting?

Flash (photo/video) or Torch currently does not work on Android because of the Camera2 rewrite. It's quite tricky to add flash capture, as the exposure has to be handled manually in Camera2.

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

none

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

LiamBateman, 2023-10-10:
I'm happy to take a look at this as this seems like a critical item for this package. Is there any starting point on this implementation?

jthure, 2023-10-10:
> I'm happy to take a look at this as this seems like a critical item for this package. Is there any starting point on this implementation?

Take a look at my comment here: https://github.com/mrousavy/react-native-vision-camera/issues/1726#issuecomment-1697352602. I would this is related to implementing the pre-capture sequence.

mrousavy, 2023-10-25:
Thanks to everyone who already sponsored towards this goal in the polar pool linked here - we're at $190 right now. $2,810 left to go 😅 

As a teaser I just implemented very basic flash capture which uses an OS feature - classic Android; this does not work on every phone. It's a short torch burst which does not wait for AE/AF exposure to be settled.

A proper implementation requires a fully custom precapture sequence that enables the torch, then waits for AE/AF to adjust, then lock AE/AF to that value, turn off the torch again, then capture with a single flash burst, and then turn the torch off again, and return to auto-AE/AF. This is quite complex, that's why this feature request is marked at $3,000.

mrousavy, 2024-02-14:
Update: We're far off from the $3,000, but **I decided to implement Flash anyways** - PR is up, I'm testing everything now!!! 🥳🎉🚀 

[**feat: Flash with AE Pre-capture trigger for Android (#2558)**](https://github.com/mrousavy/react-native-vision-camera/pull/2558)

(check out the code changes, there's a ton of stuff required for this lol)

closed 2024-02-14 by mrousavy

mrousavy, 2024-02-14:
I just released [VisionCamera 3.9.0-beta.4](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.4), which contains flash support! (will continue testing more, then release 3.9.0 as stable)

With that I'd say this feature request is now finished - thanks to everyone who donated to the money pool!!! ❤️