margelo/react-native-vision-camera #1891: ✨ Implement Orientation

mrousavy, 2023-09-30:
### What feature or enhancement are you suggesting?

Currently, there is no direct orientation support for VisionCamera.

This feature request contains of multiple parts:

- Support `orientation` prop: When this is passed, the Camera will rotate itself to the target orientation
- Take Camera Sensor orientation (often 90deg) into consideration for relative orientation
- Save correct orientation in captured photos (relative to sensor and relative to device/`orientation` prop)
- Save correct orientation in recorded videos (relative to sensor and relative to device/`orientation` prop)
- Pass correct orientation in `Frame` in Frame Processor (relative to sensor and relative to device/`orientation` prop)
- Documentation on how to properly handle orientation (see iOS Camera app, the camera itself should not rotate, only controls should rotate)

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

none

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

valn1, 2023-10-02:
what should the orientation prop do in this case?
i imagine the default "none" to just do nothing, "auto" to read the device's orientation and adjust accordingly and "portrait/landscape" to use that aspect ratio.
but i wonder how it should respond to that prop changing while recording a video

mrousavy, 2023-10-03:
Yea good point. I think those should be the options:

- `"portrait"` | `"portraitUpsideDown"`: Locked at portrait orientation.
- `"landscapeLeft"` | `"landscapeRight"`: Locked at landscape orientation.
- `"device"`: Use phone orientation and always rotate when phone rotates. This is not really recommended as it requires Camera reconfiguration.
- `undefined` -> `"portrait"`

valn1, 2023-10-03:
cool, should that not be changeable mid recording?

mrousavy, 2023-10-03:
No, a recording is started at a specific orientation and will always stay at that orientation. Orientation in a recording is just an EXIF flag. it has a fixed width/height.

rodrigobertolottizinsp, 2023-10-03:
There is a way to lock the camera to portrait mode regardless of the device orientation? Thanks!

valn1, 2023-10-03:
not as of right now. i fixed that in another app of mine that runs the v2 by using react-native-orientation-locker and doing a patch on the ios part of react-native-vision-camera, but i dont think it works for pictures or video as i used it only for barcode scanning

tiste, 2023-10-04:
I really needed to implement orientation quickly, so here is a small hack to make it work on both landscape right and left (as it works on portrait).

```ts
<View
  style={{
    transform: [
      isLandscapeRight ? { rotate: "-90deg" } : { rotate: "90deg" },
    ],
    width: layout.height,
    height: layout.width,
    position: "absolute",
    left: layout.width / 2 - layout.height / 2,
    top: layout.height / 2 - layout.width / 2,
  }}
>
  <Camera
    enableHighQualityPhotos={true}
    orientation={
      isLandscapeRight ? "landscape-right" : "landscape-left"
    }
    device={device}
    isActive={true}
    ref={cameraRef}
    style={{
      width: layout.height,
      height: layout.width,
    }}
    video={true}
    photo={true}
  />
</View>
```

mrousavy, 2023-10-04:
> not as of right now

@valn1 I know, I'm talking about this proposal here.

micm0, 2023-10-04:
> I really needed to implement orientation quickly, so here is a small hack to make it work on both landscape right and left (as it works on portrait).
>
> ```ts
> <View
>   style={{
>     transform: [ [...]

This won't work in Android I believe because the orientation prop is not supported, and doesn't do anything then.

erickcrus, 2023-10-27:
Im using `@oguzhnatly/react-native-image-manipulator` package to handle image rotation after take a picture when necessary. This is not ideal but works.

bglgwyng, 2023-11-01:
Actually, I haven't used `orientation` props since RNVC V2, as its change caused freeze on the view.
So, I gave up utilizing `orientation` months ago and set it `portrait`.
Also I implemented my frame processors to work well with explicitly provided `orientation` arguments, like

```tsx
<Camera orientation="portrait" />

useFrameProcessor((frame) => {
  'worklet';
  scanFaces(frame, { orientation  })
});
```

I'm not entirely satisfied with this solution, but it works well, allowing the frame processors to adapt to any device orientation. 
While the non-functional `orientation` props can be worked around, the real issue lies in the incorrect `frame.orientation`.
I'm really looking forward to the full-featured `orientation` props in RNVC V3. But before that, I hope a simple fix for `frame.orientation` can be merged.

Here I've created a branch to share the PoC of the fix.
https://github.com/bglgwyng/react-native-vision-camera/tree/orientation-in-frame
I thought the fix would be straightforwrad first, but now I'm unsure why the current `isMirrored` implementation is considered wrong as noted in the comment.

bglgwyng, 2023-11-04:
`frame.image` remains unmirrored while the preview is mirrored on Galaxy S21.
From my understanding, `frame.image` is expected to be mirrored, since `mirrorMode` is configured as `MIRROR_MODE_AUTO`. I set `mirrorMode` to `MIRROR_MODE_NONE` and observed that the preview changed to unmirrored while `frame.image` didn't get affected.
Does `imageReader` not respect it? Or is it another device-specific bug?
t would be beneficial if others with different devices could report similar behavior.

mrousavy, 2023-11-07:
> From my understanding, `frame.image` is expected to be mirrored, since `mirrorMode` is configured as `MIRROR_MODE_AUTO`

Not quite - actually mirroring an image is a computationally expensive task. Instead, the Camera uses special flags to tell the receiver (Preview View, Photo File .jpg, Video File, OpenGL pipeline, etc...) how to interpret the image.
Only in rare cases the image is actually mirrored, but this will cause a latency.

So instead, we read the image in it's natural **sensor orientation**, then apply any orientations and mirroring handling ourselves once we save it for end user display (Preview View: view transform matrix, Photo File: exif flag, Video File: exif flag, OpenGL pipeline: well, actually doing a mirror here)

Same story on iOS, except that we curretly actually mirror/orientate the image - which we can also skip for better efficiency (see my idea at https://github.com/mrousavy/react-native-vision-camera/issues/2046)

mrousavy, 2023-11-09:
> The statement 'frame.image is expected to be mirrored' doesn't convey a proposal but reflects my prediction. 

my man is speaking in cursive 😂 
[image]

Well to be honest, we don't know if it is mirrored or not. Some devices might mirror it on the HAL level (Hardware Abstraction Layer), some won't. So I guess the ideal solution is to set `MIRROR_MODE_NONE`, and then handle mirroring on our end.

ibmmt, 2023-11-09:
const photo = await cameraRef.current.takePhoto();
  let rotation=0;
  if ( isFrontCamera && (photo.orientation == 'landscape-left' || photo.orientation == 'landscape-right')) {
      rotation = 90;
    }
    const photoWithOrientation = await manipulateAsync(photo.path, [{ rotate: rotation }], {
      compress: 1,
      format: SaveFormat.JPEG,
    });

Temporary solution.

jazib10218, 2023-11-14:
> I really needed to implement orientation quickly, so here is a small hack to make it work on both landscape right and left (as it works on portrait).
>
> ```ts
> <View
>   style={{
>     transform: [ [...]

Now camera is showing a zoomed in preview, how did you handle that ?

mrousavy, 2024-01-03:
Related: https://stackoverflow.com/questions/48406497/

vanenshi, 2024-01-30:
This issue is causing problems with Google MLKit Face detection.
here is the library I am trying to build [vanenshi/vision-camera-face-detector](https://github.com/vanenshi/vision-camera-face-detector)

Technically MLKit needs the correct orientation to be able to detect the face, so when I manually set it to 90 deg, it works in portrait mode

drLeonymous, 2024-03-05:
Hi, all! 
I'm not a Java/Kotlin friend but I did some research and here is a patch that works on the emulator Pixel_2_XL and Samsung Note10 and some other old Samsung or Huawei devices. The main 

```
--- a/package/android/src/main/java/com/mrousavy/camera/CameraView+TakePhoto.kt
+++ b/package/android/src/main/java/com/mrousavy/camera/CameraView+TakePhoto.kt
@@ -14,6 +14,7 @@ import com.facebook.react.bridge.WritableMap
 import com.mrousavy.camera.core.CameraSession
 import com.mrousavy.camera.core.InsufficientStorageError
 import com.mrousavy.camera.types.Flash
+import com.mrousavy.camera.types.Orientation
 import com.mrousavy.camera.types.QualityPrioritization
 import com.mrousavy.camera.utils.*
 import java.io.File
@@ -46,7 +47,7 @@ suspend fun CameraView.takePhoto(optionsMap: ReadableMap): WritableMap {
     enableShutterSound,
     enableAutoStabilization,
     enablePrecapture,
-    orientation
+    cameraSession.orientation
  
+private val matrix = Matrix()
 private fun writePhotoToFile(photo: CameraSession.CapturedPhoto, file: File) {
   val byteBuffer = photo.image.planes[0].buffer
-  if (photo.isMirrored) {
     val imageBytes = ByteArray(byteBuffer.remaining()).apply { byteBuffer.get(this) }
     val bitmap = BitmapFactory.decodeByteArray(imageBytes, 0, imageBytes.size)
-    val matrix = Matrix()
-    matrix.preScale(-1f, 1f)
-    val processedBitmap =
-      Bitmap.createBitmap(bitmap, 0, 0, bitmap.width, bitmap.height, matrix, false)
+    matrix.reset()
+    if(photo.isMirrored) {
+      matrix.preScale(-1f, 1f)
+    }
+    matrix.postRotate(90f)
+    val processedBitmap = Bitmap.createBitmap(bitmap, 0, 0, bitmap.width, bitmap.height, matrix, false)
     FileOutputStream(file).use { stream ->
       processedBitmap.compress(Bitmap.CompressFormat.JPEG, 100, stream)
     }
-  } else {
-    val channel = FileOutputStream(file).channel
-    channel.write(byteBuffer)
-    channel.close()
-  }
 }
 
 private suspend fun savePhotoToFile(
diff --git a/package/android/src/main/java/com/mrousavy/camera/types/Orientation.kt b/package/android/src/main/java/com/mrousavy/camera/types/Orientation.kt
index 64dc5bcf..86275519 100644
--- a/package/android/src/main/java/com/mrousavy/camera/types/Orientation.kt
+++ b/package/android/src/main/java/com/mrousavy/camera/types/Orientation.kt
@@ -3,17 +3,17 @@ package com.mrousavy.camera.types
 import com.mrousavy.camera.core.CameraDeviceDetails
 
 enum class Orientation(override val unionValue: String) : JSUnionValue {
-  PORTRAIT("portrait"),
   LANDSCAPE_RIGHT("landscape-right"),
-  PORTRAIT_UPSIDE_DOWN("portrait-upside-down"),
-  LANDSCAPE_LEFT("landscape-left");
+  PORTRAIT("portrait"),
+  LANDSCAPE_LEFT("landscape-left"),
+  PORTRAIT_UPSIDE_DOWN("portrait-upside-down");
 
   fun toDegrees(): Int =
     when (this) {
-      PORTRAIT -> 0
-      LANDSCAPE_LEFT -> 90
-      PORTRAIT_UPSIDE_DOWN -> 180
-      LANDSCAPE_RIGHT -> 270
+        LANDSCAPE_LEFT -> 0
+        PORTRAIT -> 90
+        LANDSCAPE_RIGHT -> 180
+        PORTRAIT_UPSIDE_DOWN -> 270
     }
 
   fun toSensorRelativeOrientation(deviceDetails: CameraDeviceDetails): Orientation {
@@ -26,7 +26,7 @@ enum class Orientation(override val unionValue: String) : JSUnionValue {
     }
 
     // Rotate sensor rotation by target rotation
-    val newRotationDegrees = (deviceDetails.sensorOrientation.toDegrees() + rotationDegrees + 360) % 360
+    val newRotationDegrees = (deviceDetails.sensorOrientation.toDegrees() + rotationDegrees + 270) % 360
 
     return fromRotationDegrees(newRotationDegrees)
   }
@@ -34,19 +34,19 @@ enum class Orientation(override val unionValue: String) : JSUnionValue {
   companion object : JSUnionValue.Companion<Orientation> {
     override fun fromUnionValue(unionValue: String?): Orientation =
       when (unionValue) {
+        "landscape-left" -> LANDSCAPE_LEFT
         "portrait" -> PORTRAIT
         "landscape-right" -> LANDSCAPE_RIGHT
         "portrait-upside-down" -> PORTRAIT_UPSIDE_DOWN
-        "landscape-left" -> LANDSCAPE_LEFT
         else -> PORTRAIT
       }
 
     fun fromRotationDegrees(rotationDegrees: Int): Orientation =
       when (rotationDegrees) {
-        in 45..135 -> LANDSCAPE_LEFT
-        in 135..225 -> PORTRAIT_UPSIDE_DOWN
-        in 225..315 -> LANDSCAPE_RIGHT
-        else -> PORTRAIT
+          in 45..135 -> PORTRAIT
+          in 135..225 -> LANDSCAPE_RIGHT
+          in 225..315 -> PORTRAIT_UPSIDE_DOWN
+          else -> LANDSCAPE_LEFT
       }
   }
 }

```

matheusqfql, 2024-04-01:
This is not working for me on Ipad with camera V3

drLeonymous, 2024-04-01:
Use v4 beta 10 and rotate 90 deg the image when you save it in the server

matheusqfql, 2024-04-01:
> Use v4 beta 10 and rotate 90 deg the image when you save it on the backend server

The problem is that the live visualisation is 90 deg inverted on iPad landscape mode. I'm just using the barcode scanner (I'm not saving any image)

simon-assistiq, 2024-04-10:
> The problem is that the live visualisation is 90 deg inverted on iPad landscape mode. I'm just using the barcode scanner (I'm not saving any image)

Having the same issue as @matheusqfql. I'm a bit confused; is there no way to ensure the preview output of the camera on the screen is not rotated and inverted? I'm migrating from 2.13.2 where the preview output was in the correct orientation, so I'm just wondering whether I'm just missing something new or some change was introduced. I don't need it to be able to respond to orientation changes, just want it to be always displayed in landscape on the iPad.

JesseLawler, 2024-04-25:
Does anyone have a good solution for this when applying to *video*? 

Rotating an image is no big deal; I'm unsure of any reasonable solution for video...

pvedula7, 2024-04-25:
> I'm migrating from 2.13.2 where the preview output was in the correct orientation, so I'm just wondering whether I'm just missing something new or some change was introduced. I don't need it to be [...]

Facing something somewhat similar where we are recording a video in standard portrait mode, but the frame.orientation always prints landscape-left. When attempting to overlay things onto the video with coordinates from frame processor, it's always off presumably due to some frame difference. Worked fine in V2 for us.

mrousavy, 2024-04-29:
btw., I wrote an explanation on why rotation/orientation isn't as straight forward as it sounds here: https://github.com/mrousavy/react-native-vision-camera/pull/2807#issuecomment-2083079356

mars-lan, 2024-05-19:
After reverting back to CameraX in 4.x, this doesn't seem to be an issue anymore.

mrousavy, 2024-05-21:
This still is an issue - Frame Processors don't pass you a transformation matrix, and the preview view orientation cannot be frozen while outputs should rotate - that's how most Camera apps work.

Also, on iOS this still isn't implemented. For now, only portrait works perfectly.

luisreiskeys, 2024-05-31:
I Just made a module to handle the device orientation
https://www.npmjs.com/package/rn-device-orientation
the usage is pretty simple, and this module get de device orientation, not the interface orientation, so even if the interface is locked on portrait, it will work

to solve on vision camera a jus use like this
[image]

mrousavy, 2024-06-04:
just hustled this one out... 👀  https://github.com/mrousavy/react-native-vision-camera/pull/2932

mrousavy, 2024-06-04:
So I took a quick stab at implementing Orientation the past two days.
I spent a lot of time researching and playing around with it, and I got a really good understanding of how Orientation should be handled in Camera apps.

- Preview works well, it can rotate to landscape (left + right) and portrait (up + down). No problem. With screen lock, it can even stay the same while the outputs rotate (better visually for the user)
- Photos works well, it can capture photos in all orientations (landscape + portrait) and saves it accordingly. Even if preview is locked.
- Frame Processors works well, I added documentation so users know how to work with the `orientation` property.
- Video works well on Android, and on iOS it works only when you are not flipping the device in-between.

Here's what's problematic; when you flip the Camera while recording a video, the incoming video is upside down.

This is because the selfie camera is mirrored alongside the vertical axis (`isVideoMirrored`), when in reality it should be mirrored alongside the horizontal axis (because it is in the native sensor orientation). Unfortunately Apple does not have an API for mirroring alongside the horizontal axis...
So I don't really know how to proceed here, unless I add mirror + orientation back to the video stream (which comes at a performance cost.)

mrousavy, 2024-06-04:
Ok I figured something out. I can rotate/mirror only for selfie cameras, and keep back cameras efficient. This sounds a bit hacky, but I think this will work well and is much more performant than always rotating. I think this should work on all iPhones.
I'd appreciate any feedback on https://github.com/mrousavy/react-native-vision-camera/pull/2932!

mrousavy, 2024-06-06:
Done! Mergegd #2932, Orientation is now implemented in VisionCamera!! 🤩 

Thanks to all my sponsors who donated to this sponsorpool, without y'all this wouldn't have been possible 🙏

closed 2024-06-06 by mrousavy

laxminarayana-capgemini, 2024-06-07:
Hi @mrousavy, we trying to record the video in Landscape, but the output video is in Portrait. Whatever the orientation we are recording it is in Portrait only. We are running react-native-vision-camera in Android application, but it is not working. Hereby I have attached a sample code. Kindly help us with a solution.

<Camera
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={isAppForeground}
            video={true}
            audio={true}
            ref={camera}
            format={format}
            exposure={0}
            zoom={0}
            orientation='landscape-left'
          />

sugaith, 2024-06-14:
for iOS: 
- Before this update, iOS wouldn't respond to the device's rotation. 
  - and if we lock the orientation before opening the camera, it was getting the wrong rotation
- after this update,  Now for iOS is REALLY GOOD!

For Android:
- Before this update, we had to lock the desired orientation before opening the camera, or else it would also not respond
- now it responds to the rotation, but is getting it 90o wrong.

If I simply switch the camera from front to back again, it works.

mrousavy, 2024-06-15:
Yep I collected a bit of feedback and I can fix it soon

ch3njust1n, 2024-07-17:
I'm getting an issue when I record in portrait, the video is saved in portrait, but the metadata indicates from the width and height that the video is landscape. I need the metadata to be correct for my application. I verified that when I open the video in Google Photos or upload to Instagram or Messenger that the orientation and metadata are correct. But when I view the video with RNFS, ImagePicker or FFprobeKit immediately after recording, the width is greater than the height, but video plays back as portrait.

spsaucier, 2024-11-06:
I get something similar to @ch3njust1n - it saves the portrait photo & previews it correctly, but then when it is resized & uploaded, the resulting photo shows up in an Image component as rotated 90 degrees clockwise.

omerts, 2025-01-13:
Seems like orientation in exif data is always 6 disregarding true orientation