margelo/react-native-vision-camera #1894: Detect if the picture taken is too dark (at night for example)

e-nouri, 2023-09-30:
### Question

Is there an efficient way to detect if the images taken by the camera are too dark ?
To give a bit of context, I am using AWS Rekognition for face indexing. I am wandering if it is possible to do "image too dark" efficiently on device rather than sending the pictures to the backend.

Thank you for this amazing library and your amazing work. I appreciate all the help.

### What I tried

Ambient light sensor, but not available on iOS.

### VisionCamera Version

2.15.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-01:
Hey thanks!

This could achieved in a Frame Processor Plugin that detects the brightness of the image in realtime. Or after taking the picture by using native code. The Camera itself currently does not expose that, I wonder if that would be possible to get in both iOS and Android... 🤔

mrousavy, 2023-10-01:
Here's how you can read those values on the native platform:

* [`AVCaptureDevice.iso`](https://developer.apple.com/documentation/avfoundation/avcapturedevice/1624649-iso)
* [`CaptureRequest.SENSOR_EXPOSURE_TIME`](https://developer.android.com/reference/android/hardware/camera2/CaptureRequest#SENSOR_EXPOSURE_TIME)

..on Android this will introduce additional overhead since we need to add that capture callback/listener to the repeating request. 🤔

e-nouri, 2023-10-02:
Thank you Mark, i'll check those links out.

mrousavy, 2023-10-05:
It won't use a lot of memory, but it's gonna be slow because it loops over every single pixel.
It might make more sense to only do this for like 30 sample pixels in the image (e.g. in the center of the frame; aka middle of the array).

Or sponsor me so I can develop a native feature for this (ISO/exposure value callbacks)

mrousavy, 2023-10-16:
I'll close this issue in favor of the feature-request to track this; https://github.com/mrousavy/react-native-vision-camera/issues/2011

I won't be working on this for now though. If someone pays me to, I'll work on it

closed 2023-10-16 by mrousavy

medhruv27, 2024-09-17:
> @e-nouri this is the solution that I found at the moment with `frameProcessor` and `toArrayBuffer` functions, I'm searching an optimized way:
> 
> ```
> const onLightDetected = Worklets.createRunInJsFn((luminanceState: boolean) =>
>     setIsLightDetect(luminanceState),
> );
> 
> const frameProcessor = useFrameProcessor(frame => {
>     'worklet';
>     runAtTargetFps(1, () => {
>       /* Luminance control */
>       try {
>         const data = frame.toArrayBuffer();
>         // console.log(`Found frame ${frame.width} x ${frame.height}`);
>         let luminanceSum = 0;
>         // Get luminance with data(pixel) average
>         for (let i = 0; i < data.length; i++) {
>           luminanceSum += data[i];
>         }
>         const result = luminanceSum / data.length;
>         const threshold = 110;
>         console.log('luminance', result);
>         void onLightDetected(result > threshold);
>       } catch (e) {
>         console.log('error detectLuminance');
>         void onLightDetected(true);
>       } 
> })
> ```
> 
> My only fear is that can use a lot of memory

Thank you @manolo-battista this helped a lot but your code needed some correction now. I converted the the byes array buffer to unit8array that will allows to interact with the binary data in the ArrayBuffer as an array of bytes (values from 0 to 255).
This makes it easier to work with the frame’s raw binary data (like extracting pixel values ).
and it doesn't take time it did calculate the sum of all pixel's luminance in ms for my device. I used 110 as the brightness threshold. 

 ```
const onLightDetected = Worklets.createRunOnJS((luminanceState) => {
  setIsBrightnessFeedbackVisible(!luminanceState);
});

const frameProcessor = useSkiaFrameProcessor(frame => {
  'worklet';
  runAtTargetFps(1, () => {
    const buffer = frame.toArrayBuffer();
    const data = new Uint8Array(buffer);
    console.log('pixelcount', data.length);
    let luminanceSum = 0;
    const sampleSize =data.length; 
    for (let i = 0; i < sampleSize; i++) {
      luminanceSum += data[i];
    }
    const averageBrightness = luminanceSum / sampleSize;
    console.log('avgbrightness', averageBrightness);
    const luminanceState = averageBrightness > BRIGHTNESS_THRESHOLD;
    void onLightDetected(luminanceState);
});
  const faces = detectFaces(frame);
  frame.render();
});
```

mrousavy, 2024-09-18:
Smart! Nice solution 👍