margelo/react-native-vision-camera #1897: 🐛 rgb doesn't work with ver 3.2.2

rodgomesc, 2023-10-01:
### What's happening?

app crashes while try to use ` pixelFormat="rgb"` in camera component

### Reproduceable Code

```tsx
import React from 'react';
import { StyleSheet } from 'react-native';
import {
  Camera,
  useCameraDevice,
  useFrameProcessor,
} from 'react-native-vision-camera';
import { detectFaceLandmarkers } from 'react-native-mylib-vision';

export default () => {
  const [hasPermission, setHasPermission] = React.useState(false);
  const front = useCameraDevice('front');

    const frameProcessor = useFrameProcessor(
    (frame) => {
      'worklet';
      const result = detectFaceLandmarkers(frame);
      console.log(result)
    },
    []
  );

  React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'granted');
    })();
  }, []);

  if (!front || !hasPermission) {
    return null;
  }

  return (
    <>
      <Camera
        style={{ flex: 1 }}
        device={front}
        pixelFormat="rgb"
        enableFpsGraph={true}
        frameProcessor={frameProcessor}
        isActive={true}
        onError={(error) => console.log({ error })}
      />
    </>
  );
};
```

### Relevant log output

```shell
023-10-01 18:02:51.342  9199-9199  Looper                  com.gamersafervisionexample          W  PerfMonitor doFrame : time=499ms vsyncFrame=2267124 latency=0ms procState=2
2023-10-01 18:02:51.354  9199-9310  CameraDevices           com.gamersafervisionexample          I  Camera #1: Unavailable!
2023-10-01 18:02:51.428  9199-9199  VisionCameraProxy       com.gamersafervisionexample          D  Finding view 15...
2023-10-01 18:02:51.428  9199-9199  VisionCameraProxy       com.gamersafervisionexample          D  Found view 15!
2023-10-01 18:02:51.428  9199-9199  VideoPipeline           com.gamersafervisionexample          I  Setting 1920 x 1440 RecordingSession Output...
2023-10-01 18:02:51.428  9199-9199  VideoPipeline           com.gamersafervisionexample          I  Setting 1920 x 1440 FrameProcessor Output...
2023-10-01 18:02:51.810  9199-9330  BpBinder                com.gamersafervisionexample          W  PerfMonitor binderTransact :  time=470ms interface=android.hardware.ICameraService code=4
2023-10-01 18:02:51.812  9199-9330  CameraSession           com.gamersafervisionexample          I  Camera became un-available: 1
2023-10-01 18:02:51.812  9199-9330  CameraManager           com.gamersafervisionexample          I  Camera 1: Opened!
2023-10-01 18:02:51.817  9199-9330  CreateCaptureSession    com.gamersafervisionexample          I  Camera 1: Creating Capture Session #1000... Hardware Level: 3} | Outputs: [PREVIEW (1920 x 1080), VIDEO (1920 x 1440 in format #RGB)]
2023-10-01 18:02:51.818  9199-9330  CreateCaptureSession    com.gamersafervisionexample          I  Using new API (>=28)
2023-10-01 18:02:51.823  9199-9330  CameraDevice-JV-1       com.gamersafervisionexample          W  Stream configuration failed due to: createSurfaceFromGbp:1549: Camera 1: No supported stream configurations with format 0x1 defined, failed to create output stream
2023-10-01 18:02:51.825  9199-9330  CameraCaptureSession    com.gamersafervisionexample          E  Session 0: Failed to create capture session; configuration failed
2023-10-01 18:02:51.826  9199-9330  CreateCaptureSession    com.gamersafervisionexample          E  Camera 1: Failed to configure Capture Session #1000!
2023-10-01 18:02:51.879  9199-9330  AndroidRuntime          com.gamersafervisionexample          E  FATAL EXCEPTION: mrousavy/VisionCamera.main
                                                                                                    Process: com.gamersafervisionexample, PID: 9199
                                                                                                    com.mrousavy.camera.CameraSessionCannotBeConfiguredError: [session/cannot-create-session] Failed to create a Camera Session for Camera 1! Outputs: [PREVIEW (1920 x 1080), VIDEO (1920 x 1440 in format #RGB)]
                                                                                                    	at com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt$createCaptureSession$2$callback$1.onConfigureFailed(CameraDevice+createCaptureSession.kt:45)
                                                                                                    	at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1$CallbackProxies$SessionStateCallbackProxy(CallbackProxies.java:64)
                                                                                                    	at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(Unknown Source:4)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loop(Looper.java:236)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
                                                                                                    	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@a340673, mrousavy/VisionCamera.main]
2023-10-01 18:02:51.901  9199-9330  Process                 com.gamersafervisionexample          I  Sending signal. PID: 9199 SIG: 9
---------------------------- PROCESS ENDED (9199) for package com.gamersafervisionexample ----------------------------
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

mi 9t pro

### VisionCamera Version

3.2.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-02:
Yes, in the latest version RGB no longer works (i mentioned that in the changelog).

On Android you can only use PRIVATE and YUV for streaming video, RGB is not natively supported by the Cameras, so it has to either be converted on the CPU or through an OpenGL pipeline - both is not recommended. 

Your models should use the more efficient format YUV instead.

closed 2023-10-02 by mrousavy

rodgomesc, 2023-10-02:
Well, it's sad to hear that we are no longer supporting RGB. Quick note: I read the changelog before opening this issue, and it doesn't explicitly mention that RGB support is being dropped

mrousavy, 2023-10-02:
Oh sorry then my bad I might want to explicitly mention that then, thanks!

titanium-cranium, 2024-01-02:
@rodgomesc @mrousavy I managed to get around this in my android frameProcessor with this:
```
        val yuvToRgbConverter = YuvToRgbConverter(ReactNativeFirebaseApp.getApplicationContext())
        val rgbBitmap = Bitmap.createBitmap(mediaImage.width, mediaImage.height, Bitmap.Config.ARGB_8888)
        yuvToRgbConverter.yuvToRgb(mediaImage, rgbBitmap)
```

Still getting inference on a 4K image (3840x2160) in ~45ms.  Not as good as ios (yet) currently running RNVC v2 at 13ms per frame.

jslok, 2024-01-16:
@titanium-cranium can you please show the code for `yuvToRgb` or where is is coming from? Thank you

mrousavy, 2024-01-16:
@jslok just google "YuvToRgbConverter". First result: https://github.com/xizhang/camerax-gpuimage/blob/master/app/src/main/java/com/appinmotion/gpuimage/YuvToRgbConverter.kt