margelo/react-native-vision-camera #1898: 🔧

igperez-ar, 2023-10-02:
### How were you trying to build the app?

I tried to build my app with react-native-vision-camera using the `yarn android` command, and it failed.
I can´t also build the example project but for a different errors. 

### Full build logs

```tsx
> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core not found, Frame Processors disabled!
Checking the license for package CMake 3.18.1 in /Users/user/Library/Android/sdk/licenses
License for package CMake 3.18.1 accepted.
Preparing "Install CMake 3.18.1 (revision: 3.18.1)".
"Install CMake 3.18.1 (revision: 3.18.1)" ready.
Installing CMake 3.18.1 in /Users/user/Library/Android/sdk/cmake/3.18.1
"Install CMake 3.18.1 (revision: 3.18.1)" complete.
"Install CMake 3.18.1 (revision: 3.18.1)" finished.

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a]
C/C++: CMake Warning at /Users/user/Library/Android/sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
C/C++:   An old version of CMake is being used that cannot automatically detect
C/C++:   compiler attributes.  Compiler identification is being bypassed.  Some
C/C++:   values may be wrong or missing.  Update to CMake 3.19 or newer to use
C/C++:   CMake's built-in compiler identification.
C/C++: Call Stack (most recent call first):
C/C++:   /Users/user/Library/Android/sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
C/C++:   .cxx/Debug/5z1p2x5s/arm64-v8a/CMakeFiles/3.18.1-g262b901/CMakeSystem.cmake:6 (include)
C/C++:   CMakeLists.txt:1 (project)
C/C++: ninja: Entering directory `/Users/user/app/node_modules/react-native-vision-camera/android/.cxx/Debug/5z1p2x5s/arm64-v8a'
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JFrame.cpp:65:10: error: 'AHardwareBuffer_fromHardwareBuffer' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer_jni.h:48:18: note: 'AHardwareBuffer_fromHardwareBuffer' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:91:7: error: 'AHardwareBuffer_describe' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:387:6: note: 'AHardwareBuffer_describe' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:112:7: error: 'AHardwareBuffer_lock' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:442:5: note: 'AHardwareBuffer_lock' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:119:7: error: 'AHardwareBuffer_unlock' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:464:5: note: 'AHardwareBuffer_unlock' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:120:7: error: 'AHardwareBuffer_release' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:379:6: note: 'AHardwareBuffer_release' has been explicitly marked unavailable here

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/Users/user/app/node_modules/react-native-vision-camera/android/.cxx/Debug/5z1p2x5s/arm64-v8a'
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JFrame.cpp:65:10: error: 'AHardwareBuffer_fromHardwareBuffer' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer_jni.h:48:18: note: 'AHardwareBuffer_fromHardwareBuffer' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:91:7: error: 'AHardwareBuffer_describe' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:387:6: note: 'AHardwareBuffer_describe' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:112:7: error: 'AHardwareBuffer_lock' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:442:5: note: 'AHardwareBuffer_lock' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:119:7: error: 'AHardwareBuffer_unlock' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:464:5: note: 'AHardwareBuffer_unlock' has been explicitly marked unavailable here
C/C++: ../../../../src/main/cpp/frameprocessor/FrameHostObject.cpp:120:7: error: 'AHardwareBuffer_release' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:379:6: note: 'AHardwareBuffer_release' has been explicitly marked unavailable here

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.5.1/userguide/command_line_interface.html#sec:command_line_warnings
691 actionable tasks: 136 executed, 555 up-to-date

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/Users/user/app/node_modules/react-native-vision-camera/android/.cxx/Debug/5z1p2x5s/arm64-v8a'
  [1/5] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o
  /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android23 --gcc-toolchain=/Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS=false -DVisionCamera_EXPORTS -I../../../../../cpp -I../../../../src/main/cpp -I../../../../src/main/cpp/frameprocessor -I../../../../src/main/cpp/frameprocessor/java-bindings -I/Users/user/app/node_modules/react-native/ReactCommon -I/Users/user/app/node_modules/react-native/ReactCommon/callinvoker -I/Users/user/app/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -isystem /Users/user/.gradle/caches/transforms-3/8bcf80ee2a1bfa604895db953412a7ec/transformed/jetified-react-android-0.71.1-debug/prefab/modules/jsi/include -isystem /Users/user/.gradle/caches/transforms-3/8bcf80ee2a1bfa604895db953412a7ec/transformed/jetified-react-android-0.71.1-debug/prefab/modules/reactnativejni/include -isystem /Users/user/.gradle/caches/transforms-3/2e68e23f8328304be19dadb03640c2bb/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -fno-limit-debug-info  -fPIC -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o -c ../../../../src/main/cpp/frameprocessor/java-bindings/JFrame.cpp
[... 237 lines cut ...]
```

[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fobos531, 2023-10-02:
Hey @igperez-ar,

based on your build logs, it seems that your `minSdkVersion` is set to a value lower than 26. Recent versions of Vision Camera require at least API 26, so that might be the cause of your issue. I'm building my app with `minSdkVersion` set to 26 and it works just fine.

mrousavy, 2023-10-02:
Exactly what @fobos531 said (thanks!) 👍

closed 2023-10-02 by mrousavy

hamdij0maa, 2023-10-02:
same problem

mrousavy, 2023-10-02:
wdym same problem? the comment above literally tells you the solution.

hamdij0maa, 2023-10-02:
i'll recheck maybe something else causing the build failure :D thank you

pepperav, 2023-10-03:
> Hey @igperez-ar,
> 
> based on your build logs, it seems that your `minSdkVersion` is set to a value lower than 26. Recent versions of Vision Camera require at least API 26, so that might be the cause of your issue. I'm building my app with `minSdkVersion` set to 26 and it works just fine.

what is exactly the file and the line to edit?

I've tried `android/build.gradle` line 6 but without any success

[image]

I got always same error:
[image]

mrousavy, 2023-10-03:
Did you check in both `app/build.gradle` and `build.gradle`?

pepperav, 2023-10-03:
Yes I did

`build.gradle`
[image]

`app/build.gradle`
[image]

Any suggestion? What can I try more?
Tnx

mrousavy, 2023-10-03:
just set it to 26 directly, I don#t know otherwise - maybe you have a caching issue clear yo cache

pepperav, 2023-10-03:
Nothing, I did a lot of tries but I get always same error

what I did:
- delete all folders (_android, ios, node_modules, .expo_) before every build
- delete _package-lock.json_
- added `26` directly at line 6 col 25 of `/android/build.gradle`
- updated Android Studio with _Android Studio Giraffe | 2022.3.1 Patch 2_
- applied all of this [guide lines here](https://www.react-native-vision-camera.com/docs/guides/troubleshooting#build-issues-1)

[image]

No idea left 😅

hamdij0maa, 2023-10-03:
just upgrade to the last version ,
than : npx clean-rn
yarn install 
./gradlew clean 
yarn android

mrousavy, 2023-10-03:
I actually made a PR that makes it possible to use VisionCamera on minSdkVersion 25 or lower, see this: https://github.com/mrousavy/react-native-vision-camera/pull/1910

Hope that works :)

pepperav, 2023-10-03:
> I actually made a PR that makes it possible to use VisionCamera on minSdkVersion 25 or lower, see this: #1910
> 
> Hope that works :)

Tnx man! I appreciate it.

So what is the procedure now for this pr? I'm pretty new with this stuff

haianh230797, 2023-10-25:
> I actually made a PR that makes it possible to use VisionCamera on minSdkVersion 25 or lower, see this: #1910
> 
> Hope that works :)

Many thanks, it's relly work <3

harya72, 2025-08-31:
@mrousavy first of all great work bro for such a useful library. I am having issues using skia frameProcessor, i am getting the error: **Frame Processor Error: Exception in HostFunction: HardwareBuffers are only supported on Android API 26 or higher! Set your minSdk to 26 (or higher) and try again., js engine: VisionCamera** although i have set my minSdk to 26 in my build.gradle and also hard coded the same to the node modules of vision camera because first i was getting the same error for getNativeBuffer so i hardcoded the value so it was fixed but still getting the above error. Please help or guide me to fix the problem. Below is the dependencies version i am using:
```
dependencies": {
    "expo": "~53.0.22",
    "expo-status-bar": "~2.2.3",
    "react": "19.0.0",
    "react-native": "0.79.6",
    "react-native-vision-camera": "^4.7.1",
    "react-native-worklets-core": "^1.6.2",
    "@shopify/react-native-skia": "v2.0.0-next.4",
    "react-native-reanimated": "~3.17.4"
  }, 
```

Andromeda606, 2025-09-05:
When I downloaded version 4.7.2, it started giving this error. In the APK's AndroidManifest file, minSdk is clearly 26

mrousavy, 2025-09-06:
minSdk needs to be set in `build.gradle`.

PR #1910 (2023-10-03): fix: Make HardwareBuffers compile optionally

## What

Makes `AHardwareBuffer` C++ compilation optional.

- If the user's app specified `minSdkVersion` of 26 or higher, the feature is available and everything works as normal.
- If the user's app specified `minSdkVersion` of 25 or lower, HardwareBuffers are simply not compiled/linked in C++ and `Frame.toArrayBuffer()` will not work.

## Changes

## Tested on

## Related issues

- Fixes 1905
- Fixes #1899 
- Fixes #1898