margelo/react-native-vision-camera #1899: App does not build on Android🐛

Romick2005, 2023-10-02:
### What's happening?

After updating to 3.2.2 got Android build error, but no error when building on 3.1.0.
`'AHardwareBuffer_fromHardwareBuffer' is unavailable: introduced in Android 26`

### Reproduceable Code

```tsx
Code from example
```

### Relevant log output

```shell
> Task :react-native-vision-camera:compileDebugKotlin
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraDevicesManager.kt: (88, 19): Parameter 'eventName' is never used
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraDevicesManager.kt: (91, 23): Parameter 'count' is never used
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (7, 44): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (37, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (37, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt: (64, 5): Unreachable code
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/RecordingSession.kt: (51, 96): 'constructor MediaRecorder()' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getOutputSizes.kt: (21, 36): 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java
w: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraDevice+createCaptureSession.kt: (82, 12): 'createCaptureSessionByOutputConfigurations((Mutable)List<OutputConfiguration!>!, CameraCaptureSession.StateCallback!, Handler?): Unit' is deprecated. Deprecated in Java

> Task :react-native-webview:compileDebugJavaWithJavac

> Task :app:compileDebugJavaWithJavac

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/.cxx/Debug/3f1p2q1o/arm64-v8a'
C/C++: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp:65:10: error: 'AHardwareBuffer_fromHardwareBuffer' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer_jni.h:48:18: note: 'AHardwareBuffer_fromHardwareBuffer' has been explicitly marked unavailable here
C/C++: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/FrameHostObject.cpp:91:7: error: 'AHardwareBuffer_describe' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:387:6: note: 'AHardwareBuffer_describe' has been explicitly marked unavailable here
C/C++: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/FrameHostObject.cpp:112:7: error: 'AHardwareBuffer_lock' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:442:5: note: 'AHardwareBuffer_lock' has been explicitly marked unavailable here
C/C++: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/FrameHostObject.cpp:119:7: error: 'AHardwareBuffer_unlock' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:464:5: note: 'AHardwareBuffer_unlock' has been explicitly marked unavailable here
C/C++: /Users/user/repo/app/na-mobile/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/FrameHostObject.cpp:120:7: error: 'AHardwareBuffer_release' is unavailable: introduced in Android 26
C/C++: /Users/user/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/android/hardware_buffer.h:379:6: note: 'AHardwareBuffer_release' has been explicitly marked unavailable here
615 actionable tasks: 610 executed, 5 up-to-date
```

### Camera Device

```json
Not important.
```

### Device

Samsung S21+

### VisionCamera Version

3.2.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Romick2005, 2023-10-02:
Regression [commit](https://github.com/mrousavy/react-native-vision-camera/commit/cf1952d8f7032feac4ca8013a0943cb40aa0a482).

peternoordijk, 2023-10-02:
You will have to change the `minSdkVersion` in `android/build.gradle` to 26

igperez-ar, 2023-10-02:
Yes, it's like @peternoordijk says, check this https://github.com/mrousavy/react-native-vision-camera/issues/1898

Romick2005, 2023-10-02:
@peternoordijk, @igperez-ar Isn't it a breaking change that should be mention in release [3.2.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.2.0)?
I guess that should be on in readme file, that minSdkVersion was changed from 21 to 26. Besides that it is possible to write code to support 21 and 26 version? Especially in this case using smth like `if (Build.VERSION.SDK_INT < Build.VERSION_CODES.O) ...`.

peternoordijk, 2023-10-02:
@Romick2005 I have the feeling you can expect quite a lot more breaking changes like these since version 3 has been released only a few weeks ago. However I noticed the minSdkVersion has already been updated to 26 in the docs.

Also, I think @igperez-ar and me are just trying out this library as well, just like you. You should ask @mrousavy about compatibility for older SDK versions (or create a PR yourself, obviously)

BenQoder, 2023-10-02:
This happens also when building with expo

mrousavy, 2023-10-03:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/1898.

mrousavy, 2023-10-03:
@Romick2005 

> Isn't it a breaking change that should be mention in release [3.2.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.2.0)?

Well I don't consider it breaking if I just increase the min sdk version, I did a Twitter poll a while ago and I think 80%+ of users are on minSdk 26 or higher and don't even need any lower SDKs.
I can explicitly mention it as well though.

> I guess that should be in readme file, that minSdkVersion was changed from 21 to 26

It is in the documentation, first page, second sentence. Has been there since V3.0.0 (more than a month) afaik.

> Besides that it is possible to write code to support 21 and 26 version? Especially in this case using smth like if `(Build.VERSION.SDK_INT < Build.VERSION_CODES.O) ...`

Nope, not possible. This is the NDK (C++), you cannot do that in C++. It has to be linked against a specific version, that version is not available in this context.

What I can do is make this optional, if the user has `minSdkVersion` 25 or lower, I need to completely disable the `.toArrayBuffer()` functionality. Even if the app is running on a phone on SDK 33, the `minSdkVersion` is what counts (because it links the C++).

closed 2023-10-03 by mrousavy

mrousavy, 2023-10-03:
@peternoordijk 

> You will have to change the minSdkVersion in android/build.gradle to 26

Exactly, thanks for helping other people out Peter (& @igperez-ar), makes maintenance much easier for me 😅 

> I have the feeling you can expect quite a lot more breaking changes like this since version 3 has been released only a few weeks ago.

Yes, VisionCamera V3 has a ton of new features, completely rewritten Android codebase, new devices API, rewritten Frame Processor implementation, and much more. It is a much more robust yet flexible foundation. There are obviously some breaking changes with that. But I think from a user perspective those are minor, and the buildscript got much simpler and more robust as well.

mrousavy, 2023-10-03:
@Romick2005 you _could_ try [disabling Frame Processors](https://www.react-native-vision-camera.com/docs/guides/frame-processors#disabling-frame-processors) if you don't need them, that might no longer use HardwareBuffers then.

How many % of your users are actually on Android 7 (SDK 25) or lower?

mrousavy, 2023-10-03:
I actually made a PR that makes it possible to use VisionCamera on minSdkVersion 25 or lower, see this: https://github.com/mrousavy/react-native-vision-camera/pull/1910

Hope that works :)

PR #1910 (2023-10-03): fix: Make HardwareBuffers compile optionally

## What

Makes `AHardwareBuffer` C++ compilation optional.

- If the user's app specified `minSdkVersion` of 26 or higher, the feature is available and everything works as normal.
- If the user's app specified `minSdkVersion` of 25 or lower, HardwareBuffers are simply not compiled/linked in C++ and `Frame.toArrayBuffer()` will not work.

## Changes

## Tested on

## Related issues

- Fixes 1905
- Fixes #1899 
- Fixes #1898