margelo/react-native-vision-camera #1903: 🐛 App closes unexpectedly after recording a video

igperez-ar, 2023-10-02:
### What's happening?

Seconds after recording a video, the app quits unexpectedly with no visible errors.

### Reproduceable Code

```tsx
import React, { useRef, useState } from 'react';
import { StyleSheet, View, TouchableOpacity } from 'react-native';
import { Camera, useCameraDevice } from 'react-native-vision-camera';

const CameraScreen = () => {
  const device = useCameraDevice('front');
  const camera = useRef<Camera>(null);
  const [isRecording, setIsRecording] = useState(false);

  const recordVideo = () => {
    camera.current?.startRecording({
      flash: 'off',
      onRecordingFinished: () => console.log('finish'),
      onRecordingError: (e) => console.log(e),
      fileType: 'mp4',
    });
  };

  const onPressTakeVideo = async () => {
    if (!isRecording) {
      recordVideo();
    } else {
      await camera.current?.stopRecording();
    }
    setIsRecording(!isRecording);
  };

  if (!device) {
    return <View />;
  }

  return (
    <View style={{ flex: 1 }}>
      <Camera
        ref={camera}
        style={StyleSheet.absoluteFill}
        enableZoomGesture={false}
        video
        orientation="portrait"
        device={device}
        isActive
      />
      <TouchableOpacity
        style={{
          height: 50,
          width: '100%',
          backgroundColor: 'green',
          bottom: 0,
          position: 'absolute',
        }}
        onPress={onPressTakeVideo}
      />
    </View>
  );
};

export default CameraScreen;
```

### Relevant log output

```shell
2023-10-02 09:58:27.002  9634-9634  VideoPipeline           app.debug                            I  Rendering to RecordingSession..
[... 70 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Motorola G8 Plus - Android 10

### VisionCamera Version

3.1.0 - 3.2.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-03:
Huh, that's weird! The only error I can read in your logs is that some `release()` method is not called, but I am calling it here: https://github.com/mrousavy/react-native-vision-camera/blob/554b7416b115d6d4f88c2ac749e0155c1e9494e4/package/android/src/main/java/com/mrousavy/camera/core/RecordingSession.kt#L103

Maybe I need to release something else? 🤔

mrousavy, 2023-10-03:
Ah - could you maybe try swapping those two lines and try again? https://github.com/mrousavy/react-native-vision-camera/blob/554b7416b115d6d4f88c2ac749e0155c1e9494e4/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L302-L303

closed 2023-10-16 by mrousavy