margelo/react-native-vision-camera #1905: ❓ Android minSdkVersion 21 or 26? Documentation is somewhat misleading?

gtaylor44, 2023-10-03:
### Question

It looks like there's a discrepancy in the documentation.

Is the minimum Android SDK version 21 or 26? 

[image]

### VisionCamera Version

Latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-03:
Hey my bad yea everywhere else it also says 26, so 26 is what you need. fixed that in the docs

closed 2023-10-03 by mrousavy

mrousavy, 2023-10-03:
I actually made a PR that makes it possible to use VisionCamera on minSdkVersion 25 or lower, see this: https://github.com/mrousavy/react-native-vision-camera/pull/1910

Hope that works :)