margelo/react-native-vision-camera #1906: ❓ MaskView doesn't work with the camera

danielchan303, 2023-10-03:
### Question

I was using react-native-camera before. As it was deprecated, I tried to use this library. As I want the user to capture a picture with a card within the center I use the MaskedView (@react-native-masked-view/masked-view) to black out the other region a bit. , but it doesn't seem to work with the react-native-vision-camera (3.2.2). Any ideas about how to fix?

```
<MaskedView
  style={{flex: 1}}
  maskElement={
    <View
      style={styles.overlay}
      onLayout={event => {
        setCameraHeight(event.nativeEvent.layout.height);
        setCameraWidth(event.nativeEvent.layout.width);
      }}>
      <View
        style={styles.imageFrame}
        onLayout={event => {
          setFrameHeight(event.nativeEvent.layout.height);
          setFrameWidth(event.nativeEvent.layout.width);
        }}
      />
    </View>
  }>
  {device ? (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  ) : null}
</MaskedView>
```

### VisionCamera Version

3.2.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-03:
Hey!

This should've been a bug report instead of a question, as that would've given me more detail. Anyways

1. What platform? This is highly platform specific, I believe this will work on iOS, but won't work on Android.
2. Why do you want to use MaskedView?

Here's something you can try:

1. Replace `SurfaceView` with [`TextureView`](https://developer.android.com/reference/android/view/TextureView#setSurfaceTextureListener(android.view.TextureView.SurfaceTextureListener)) here: https://github.com/mrousavy/react-native-vision-camera/blob/554b7416b115d6d4f88c2ac749e0155c1e9494e4/package/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt#L19
2. Use `setSurfaceTextureListener` instead of the two holder methods here: https://github.com/mrousavy/react-native-vision-camera/blob/554b7416b115d6d4f88c2ac749e0155c1e9494e4/package/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt#L23-L38
3. See if that works.

TextureView is not really recommended by Android, as it is less memory efficient and slower, and does not fully support HDR.
[image]

If that works for you, I _could_ add that as an option to the Camera

mrousavy, 2023-10-03:
[image]

But yea, if that works and that's what you want then this should be a feature request instead of a Question

danielchan303, 2023-10-03:
> 2\. setSurfaceTextureListener

Excuse me, can you show me the content of the PreviewView.kt after modification. I don't know how to use setSurfaceTextureListener instead of the two holder methods that you specified.

I dont't create a bug report as your package is fine by itself, it just doesn't work together with another package.

1. What platform? This is highly platform specific, I believe this will work on iOS, but won't work on Android.
Yes, it's only a problem on android.

2. Why do you want to use MaskedView?
As I want to do something like this.

[image]

mrousavy, 2023-10-03:
I see, well why not just place a black view over the Camera and cut a hole through that (using MaskedView) instead of doing it on the Camera directly as a workaround? Does that work?

danielchan303, 2023-10-03:
No, the camera is supposed to be nested inside the MaskedView. It just doesn't work that way.

valn1, 2023-10-03:
@danielchan303 you can simply place an overlay view on top of the camera view with position absolute
in my case, i needed a square hole with rounded corners and this worked for me(i use styled components, but you can simply translate the css to stylesheet)

```
//styles.ts
export const ScannerOverlay = styled.View`
    position: absolute;
    bottom: 0;
    top: 0;
    left: 0;
    right: 0;
`;

const { width, height } = Dimensions.get("window");
const side = width * 0.6; // size of the hole
const halfSide = side / 2;

export const Square = styled.View`
    aspect-ratio: 1;
    width: ${width * 3}px;
    border: ${width * 1.5 - halfSide}px solid #0005;
    border-radius: ${width * 1.5 - halfSide + 10}px;
    left: -${width * 1.5 - side - (width / 2 - side)}px;
    top: -${width * 1.5 - side - (height / 2 - side)}px;
`;

export const Ruler = styled.View`
    aspect-ratio: 1;
    width: ${side}px;
    position: absolute;
    border: 5px solid ${({ theme }) => theme.COLORS.PRIMARY};
    left: ${width / 2 - halfSide}px;
    top: ${height / 2 - halfSide}px;
    border-radius: 10px;
`;

//Scanner.tsx
return (
        <Screen>
            {hasPermission && device && (
                <>
                    <Camera style={StyleSheet.absoluteFill} device={device} isActive={true} />
                    <ScannerOverlay>
                        <Square />
                        <Ruler />
                    </ScannerOverlay>
                </>
            )}
        </Screen>
    );
```
this is just an example tho, you should use better code than this

closed 2023-10-03 by mrousavy

danielchan303, 2023-10-04:
```
      <View style={{flex: 1}}>
        {device ? (
          <Camera
            style={{...StyleSheet.absoluteFillObject, zIndex: -100}}
            device={device}
            isActive={true}
          />
        ) : null}
        <View
          style={{
            ...StyleSheet.absoluteFillObject,
            backgroundColor: 'red',
            overflow: 'hidden',
          }}></View>
      </View>
```
```
const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
  overlay: {
    // backgroundColor: 'rgba(0,0,0,0.01)',
    backgroundColor: 'transparent',
    flex: 1,
    justifyContent: 'center',
    alignItems: 'center',
  },
  imageFrame: {
    width: layout.width - 16 * 2,
    height: undefined,
    aspectRatio: 335 / 189,
    backgroundColor: 'black',
    borderRadius: 10,
  },
  bottomBar: {
    flexDirection: 'row',
    justifyContent: 'space-between',
    alignItems: 'center',
    paddingTop: 24,
    paddingHorizontal: 24,
  },
  fromGalleryButton: {
    backgroundColor: MyColors.white,
    height: 48,
    aspectRatio: 1,
    borderRadius: 48 / 2,
    justifyContent: 'center',
    alignItems: 'center',
  },
  captureButton: {
    width: 56,
    height: 56,
    borderRadius: 56 / 2,
    backgroundColor: MyColors.hex8E7D61,
    alignSelf: 'center',
    justifyContent: 'center',
    alignItems: 'center',
  },
  captureButtonInnerRing: {
    width: 52,
    height: 52,
    borderRadius: 52 / 2,
    backgroundColor: MyColors.hex8E7D61,
    borderWidth: 2,
    borderColor: MyColors.white,
  },
  titleContainer: {
    position: 'absolute',
    left: 0,
    width: '100%',
    alignItems: 'center',
  },
});

```

I don't know why, but the camera just doesn't seems to follow the rules of the view.
[image]

mrousavy, 2023-10-04:
Try giving the Camera overflow: hidden maybe?

erwanlpfr, 2023-11-10:
Hello ! 

For those that need a workaround, I ended by using `react-native-svg`.

```typescript
const CameraSvgMask = (props: React.PropsWithChildren<Props>) => {
  const [svgLayout, setSvgLayout] = useState<LayoutRectangle>();

  const PresetMask = presets[props.mask.mask];

  return (
    <View
      style={StyleSheet.absoluteFill}
      onLayout={(event) => setSvgLayout(event.nativeEvent.layout)}
    >
      {props.children}
      <Svg pointerEvents="none">
        <G>
          <Defs>
            <Mask id="clip">
              <Rect
                fill="white"
                width="100%"
                height="100%"
              />
              {svgLayout && (
                <PresetMask parentLayout={svgLayout} />
              )}
            </Mask>
          </Defs>
          <Rect
            width="100%"
            height="100%"
            mask="url(#clip)"
            fillOpacity={props.mask?.opacity ?? 0.4}
            fill={props.mask?.color ?? Color.BLACK_LIGHT}
          />
        </G>
      </Svg>
    </View>
  );
};
```

And an example of a mask, like a typical driving license : 
```typescript
const DriveLicense = (props: MaskPresetProps) => {
  const cardWidth = props.parentLayout.width - PADDING_SIDES * 2;
  const cardHeight = cardWidth / 1.59; // Ratio of ID card

  return (
    <Rect
      width={cardWidth}
      height={cardHeight}
      fill="black"
      x={props.parentLayout.width / 2 - cardWidth / 2}
      y={props.parentLayout.height / 2 - cardHeight / 2 + HEIGHT_HEADER / 2}
      rx={10}
      ry={10}
    />
  );
};
```

Pretty smooth, works well in Android and iOS.

rodgomesc, 2023-11-11:
just to complement i'm using skia and works smooth as well

à(s) 07:32: