margelo/react-native-vision-camera #1913: 🔧 V2.16.1 and react-native 0.72.4 android Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'🔧

Skizu, 2023-10-03:
### How were you trying to build the app?

I tried to build my app with react-native-vision-camera using the android studio and the `yarn react-native run-android --variant=devdebug --active-arch-only --appIdSuffix 'dev'`, and it failed.

```gradle
# gradle.properties snipper

# Use this property to specify which architecture you want to build.
# You can also override it from the CLI using
# ./gradlew <task> -PreactNativeArchitectures=x86_64
reactNativeArchitectures=armeabi-v7a,arm64-v8a

# Use this property to enable support to the new architecture.
# This will allow you to use TurboModules and the Fabric render in
# your application. You should enable this flag either if you want
# to write custom TurboModules/Fabric components OR use libraries that
# are providing them.
newArchEnabled=false

# Use this property to enable or disable the Hermes JS engine.
# If set to false, you will be using JSC instead.
hermesEnabled=false
```

### Full build logs

```tsx
Executing tasks: [:app:assembleDevRelease, :app:assembleDevReleaseUnitTest] in project /home/runner/git/app/android

Starting Gradle Daemon...
Gradle Daemon started in 1 s 472 ms
> Task :gradle-plugin:compileKotlin UP-TO-DATE
> Task :gradle-plugin:compileJava NO-SOURCE
> Task :gradle-plugin:pluginDescriptors UP-TO-DATE
> Task :gradle-plugin:processResources UP-TO-DATE
> Task :gradle-plugin:classes UP-TO-DATE
> Task :gradle-plugin:jar UP-TO-DATE
> Task :gradle-plugin:inspectClassesForKotlinIC UP-TO-DATE

> Configure project :app
AGPBI: {"kind":"warning","text":"The option setting 'android.enableDexingArtifactTransform.desugaring=false' is deprecated.\nThe current default is 'true'.\nIt will be removed in version 8.0 of the Android Gradle plugin.","sources":[{}]}

> Configure project :react-native-reanimated
Android gradle plugin: 7.4.2
Gradle: 8.0.1

> Configure project :react-native-video
useExoplayerIMA:false

> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: /home/runner/git/app/node_modules
VisionCamera: Building with JSC...
VisionCamera: Frame Processors are enabled! Building C++ part...

The Kotlin Gradle plugin was loaded multiple times in different subprojects, which is not supported and may break the build. 
This might happen in subprojects that apply the Kotlin plugins with the Gradle 'plugins { ... }' DSL if they specify explicit versions, even if the versions are equal.
Please add the Kotlin plugin to the common parent project or the root project, then remove the versions in the subprojects.
If the parent project does not need the plugin, add 'apply false' to the plugin line.
See: https://docs.gradle.org/current/userguide/plugins.html#sec:subprojects_plugins_dsl
The Kotlin plugin was loaded in the following projects: ':react-native-vision-camera', ':react-native-webview'

> Task :app:createDevReleaseVariantModel UP-TO-DATE
> Task :react-native-fs:preBuild UP-TO-DATE
> Task :react-native-fs:preReleaseBuild UP-TO-DATE
> Task :react-native-fs:compileReleaseAidl NO-SOURCE
> Task :react-native-fs:mergeReleaseJniLibFolders UP-TO-DATE
> Task :react-native-fs:mergeReleaseNativeLibs NO-SOURCE
> Task :react-native-fs:stripReleaseDebugSymbols NO-SOURCE
> Task :react-native-fs:copyReleaseJniLibsProjectAndLocalJars UP-TO-DATE
> Task :react-native-fs:compileReleaseRenderscript NO-SOURCE
[... 435 lines cut ...]
```

### VisionCamera Version

2.16.1

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Skizu, 2023-10-03:
It looks like this is a result `reactNativeArchitectures` due to `react-native-vision-camera` hard coding its values throughout the entire `build.gradle`.

lostMindCtrl, 2023-10-05:
Any update on the issue? We are facing the same issue with:
```json
"dependencies": {
  "react": "18.2.0",
  "react-native": "0.72.5",
  "react-native-reanimated": "^3.5.4",
  "react-native-vision-camera": "2",
  ...
}
``` 

This issue has surfaced both in a project that we recently updated to React Native version 0.72.5 and in newly created React Native projects using this version. Specifically, the issue arises when attempting to install both "react-native-vision-camera" and "react-native-reanimated." Interestingly, installing "react-native-vision-camera" in isolation does not result in any errors. However, the moment "react-native-reanimated" (v3) is added to the mix, the described error consistently reoccurs.

We would greatly appreciate any insights or assistance in resolving this persistent issue.

mrousavy, 2023-10-05:
Hey!

Sorry but I don't provide any free support for VisionCamera V2 anymore. I think this can be fixed in the build.gradle file by fixing reactNativeArchitectures, if someone wants to do a PR for this (targeting the `v2` branch) I can merge & release it.

If you absolutely need this fixed, consider paying me for consultancy through my agency [https://margelo.io](margelo.io).

L7ARomeo, 2023-10-06:
[image]

Guys, I did upgrade to Gradle 8.1 from 8.0
I found sometimes it weird error "`Task :react-native-vision-camera:buildCMakeRelWithDebInfo[arm64-v8a] FAILED`" on vision camera and worklets everytime I do gradlew clean or deleting gradle cache even sometimes happen after reinstalling all packages, it's like vision camera can find but won't able to load worklets successfully.
Gradle 8.1 has stable configuration cache

note: only up gradle to 8.1 if the default one is 8.0
[image]

podger, 2023-10-06:
@mrousavy I don't think this is related to v2 but to the lib in general. 

It happens also to me with this setup:

    "react-native-vision-camera": "^3.3.1",
    "react-native-reanimated": "^3.5.4",
    "react-native": "^0.72.5"

Seems happening only to debug. I'm using a Mac with Intel chip.

mrousavy, 2023-10-06:
@podger I'm pretty sure it is related to v2.

In the logs shared by @Skizu it says:

```
> Task :react-native-vision-camera:buildCMakeRelWithDebInfo[armeabi-v7a] FAILED
C/C++: ninja: Entering directory `/home/runner/git/app/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/4o6y3r3t/armeabi-v7a'
C/C++: ld: error: undefined symbol: reanimated::extractWorkletRuntime(facebook::jsi::Runtime&, facebook::jsi::Value const&)
C/C++: ld: error: undefined symbol: reanimated::extractShareableOrThrow(facebook::jsi::Runtime&, facebook::jsi::Value const&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&)
C/C++: ld: error: undefined symbol: typeinfo for reanimated::Shareable
C/C++: ld: error: undefined symbol: typeinfo for reanimated::ShareableWorklet
C/C++: clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
```

VisionCamera V3 no longer uses reanimated, so it is not possible for it to cause a build failure with Reanimated like above.

podger, 2023-10-06:
> @podger I'm pretty sure it is related to v2.
>
> In the logs shared by @Skizu it says:
>
> ```
> C/C++: ninja: Entering directory [...]

Yea, I was confused by the first lines, my error actual involves react-native-worklets-core (0.2.2). 

By the way only in debug. ./gradlew bundleRelease build successfully

```
What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/Users/podger/Projects/beesy-worker-app/node_modules/react-native-vision-camera/android/.cxx/Debug/265u415o/arm64-v8a'
  [1/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o
  [2/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o
  [3/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o
  [4/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o
  [5/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o
  [6/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o
  [7/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  [8/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o
  [9/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o
  [10/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o
  [11/16] Building CXX object CMakeFiles/VisionCamera.dir/Users/podger/Projects/beesy-worker-app/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o
  [12/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o
  [13/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o
  [14/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o
  [15/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o
  [16/16] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/265u415o/obj/arm64-v8a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cxx/Debug/265u415o/obj/arm64-v8a/libVisionCamera.so 
  : && /Users/podger/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android26 --sysroot=/Users/podger/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -g  -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Qunused-arguments -Wl,--no-undefined -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/265u415o/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/Users/podger/Projects/beesy-worker-app/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o  /Users/podger/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/26/liblog.so  -landroid  /Users/podger/.gradle/caches/transforms-3/752a8789e4025c9fb0f6b63808422207/transformed/jetified-react-android-0.72.5-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /Users/podger/.gradle/caches/transforms-3/752a8789e4025c9fb0f6b63808422207/transformed/jetified-react-android-0.72.5-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so  /Users/podger/.gradle/caches/transforms-3/51d799d01a002cecbc2c0a47ca799d93/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  -lGLESv2  -lEGL   -latomic -lm && :
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
  >>> referenced by JFrameProcessor.cpp:37 (/Users/podger/Projects/beesy-worker-app/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:37)
[... cut ...]
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/8.0.1/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 4m 5s
490 actionable tasks: 481 executed, 9 up-to-date
```

natemartins, 2023-10-07:
I having the same issue here

```
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'
```

Related dependencies and versions installed

```json
"react-native-reanimated": "^3.5.4",
"react-native-vision-camera": "^3.3.1",
"react-native-worklets-core": "^0.2.2"
```

mrousavy, 2023-10-07:
Does it work if you clean cache, then immediately do a release build?

As in 

• rm -rf node_modules
• rm -rf android (COMMIT ur changes before!!)
• git reset --hard
• yarn
• cd android
• ./gradlew assembleRelease or whatever

devoren, 2023-10-08:
> I having the same issue here
> 
> ```
> Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'
> ```
> 
> Related dependencies and versions installed
> 
> ```json
> "react-native-reanimated": "^3.5.4",
> "react-native-vision-camera": "^3.3.1",
> "react-native-worklets-core": "^0.2.2"
> ```
> 
> @mrousavy This is not a `v2` problem as you can see from the dependencies outlined above
Yes, i have the same issue
```
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `C:\Programs\...\app\node_modules\react-native-vision-camera\android\.cxx\Debug\2o5s6am4\arm64-v8a'
```

robertyulisman, 2023-10-09:
> > I having the same issue here
> > ```
> > Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Related dependencies and versions installed
> > ```json
> > "react-native-reanimated": "^3.5.4",
> > "react-native-vision-camera": "^3.3.1",
> > "react-native-worklets-core": "^0.2.2"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > @mrousavy This is not a `v2` problem as you can see from the dependencies outlined above
> 
> Yes, i have the same issue
> 
> ```
> Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> > com.android.ide.common.process.ProcessException: ninja: Entering directory `C:\Programs\...\app\node_modules\react-native-vision-camera\android\.cxx\Debug\2o5s6am4\arm64-v8a'
> ```
> 
> **UPD**: After deleting `.gradle` folder, build works now, but i must delete every time when i build :(

its worked but also weird, there is some alternatively ?

mrousavy, 2023-10-09:
Hm, maybe there is a cache issue with react-native-worklets-core. I honestly don't know why this is happening, I would really appreciate it if you guys could report with more information and try potential solutions so we can fix this upstream.

mrousavy, 2023-10-16:
Maybe some kind of clean command in react-native-worklets-core.

Does it work if you uninstall react-native-worklets-core?

closed 2023-10-16 by mrousavy

katerynakostikova, 2023-10-23:
@mrousavy I don't have react-native-worklets-core in my package.json
Here is my versions of reanimated and vision camera:
`"react-native-reanimated": "^3.5.4",
	"react-native-vision-camera": "^2.16.2",`

pasquale95, 2023-11-09:
This issue seems the same as #1814.

DaAitch, 2023-11-24:
Deleting `android/.gradle` finally made it 🎉, thank you.

```
"react-native": "0.72.6",
"react-native-reanimated": "^3.5.4",
"react-native-vision-camera": "2.16.5",
```

mateuschines, 2023-12-08:
This error is still happening, and there is no solution, because I want to read qrcode with the vision-camera library with the options from the latest version but it doesn't work

"react-native": 0.72.6
"react-native-vision-camera": 3.6.14

Olalexy1, 2023-12-10:
"dependencies": {
    "expo": "~49.0.18",
    "react": "18.2.0",
    "react-native-reanimated": "~3.3.0",
    "react-native-vision-camera": "^3.6.14",
    "react-native-worklets-core": "^0.2.4",
  }
  
Has anyone successfully addressed this issue? I'm currently facing the same problem, and it's preventing me from building on Android.

andresfelipe3112, 2024-01-01:
> > > Tengo el mismo problema aquí.
> > > ```
> > > Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Dependencias relacionadas y versiones instaladas
> > > ```json
> > > "react-native-reanimated": "^3.5.4",
> > > "react-native-vision-camera": "^3.3.1",
> > > "react-native-worklets-core": "^0.2.2"
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > @mrousavyEsto no es un `v2`problema como puede ver en las dependencias descritas anteriormente.
> > 
> > 
> > Si, tengo el mismo problema
> > ```
> > Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> > > com.android.ide.common.process.ProcessException: ninja: Entering directory `C:\Programs\...\app\node_modules\react-native-vision-camera\android\.cxx\Debug\2o5s6am4\arm64-v8a'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > **UPD** : Después de eliminar `.gradle`la carpeta, la compilación funciona ahora, pero debo eliminarla cada vez que compilo :(
> 
> Ha funcionado pero también es extraño, ¿hay alguna alternativa?

donde esta la carpeta .grade

Paolo-Gambardella, 2024-01-11:
> Deleting `android/.gradle` finally made it 🎉, thank you.

It fixed my problem

mrousavy, 2024-01-18:
> Worked for me too. Annoying though, will likely remove this library soon.

Or just... upgrade to V3 where this is fixed? lol

mrousavy, 2024-01-23:
Well you're commenting under a V2 issue with a Reanimated build error - V3 does not use Reanimated anymore.

The Worklet error is caused by a cache miss, and can be resolved by deleting android build cache. Then it builds fine.

jeevan8888, 2024-04-19:
I also get encountered such an error.

mostasim, 2024-06-12:
``> Deleting `android/.gradle` finally made it 🎉, thank you.
> 
> ```
> "react-native": "0.72.6",
> "react-native-reanimated": "^3.5.4",
> "react-native-vision-camera": "2.16.5",
> ```

Deleting `.gradle` folder in `android` directory solved the build issue for me.
Here are the versions I have used :
```
"react-native-reanimated": "3.11.0",
"react-native-vision-camera": "4.0.0",
"react-native-worklets-core": "1.1.1",
```

mrousavy, 2024-06-28:
Next step would be to fix the cache/build issue. I don't know where it's coming from, but you can take a look at [react-native-worklets-core's `build.gradle`](https://github.com/margelo/react-native-worklets-core/blob/main/android/build.gradle) - PRs are much appreciated!

MalekZishan, 2024-08-12:
distributionUrl=https\://services.gradle.org/distributions/gradle-8.6-all.zip  to change distributionUrl=https\://services.gradle.org/distributions/gradle-8.5-all.zip

NoahSimajji, 2024-08-16:
Any other way to fix this issue rather than deleting .gradle folder file each time i boot up my app?

Malcom-Mullen, 2024-08-19:
1 -  ./gradlew clean (in android directory)
2 -  delete .gradle file in android folder)
3 -  yarn start --reset-cache
4 -  yarn run android

nppull, 2024-08-30:
delete `android/.gradle` work for me
but when i rebuild with android studio the error comes back

shyrynaz, 2024-09-11:
> delete `android/.gradle` work for me but when i rebuild with android studio the error comes back

this also worked for me

boiboif, 2024-09-26:
Is there any way to permanently solve this problem? It's so annoying to have to cache every time before building