margelo/react-native-vision-camera #1914: Is the fix for crash in iOS 15 pro avaliable for v2 / 2.x version ❓

vaishnav-197, 2023-10-04:
### Question

When i try to update from 2.13.0 to 3.2.2
I am encountering an error 

`Exception in thread "main" java.lang.IllegalArgumentException: Only schema_version 1 is supported. ReactAndroid uses version 2.
         at com.google.prefab.api.Package.<init>(Package.kt:47)
         at com.google.prefab.cli.Cli$packages$2.invoke(Cli.kt:124)
         at com.google.prefab.cli.Cli$packages$2.invoke(Cli.kt:95)
         at kotlin.SynchronizedLazyImpl.getValue(LazyJVM.kt:74)
         at com.google.prefab.cli.Cli.getPackages(Cli.kt)
         at com.google.prefab.cli.Cli.validate(Cli.kt:172)
         at com.google.prefab.cli.Cli.run(Cli.kt:189)
         at com.github.ajalt.clikt.parsers.Parser.parse(Parser.kt:168)
         at com.github.ajalt.clikt.parsers.Parser.parse(Parser.kt:16)
         at com.github.ajalt.clikt.core.CliktCommand.parse(CliktCommand.kt:258)
         at com.github.ajalt.clikt.core.CliktCommand.parse$default(CliktCommand.kt:255)
         at com.github.ajalt.clikt.core.CliktCommand.main(CliktCommand.kt:273)
         at com.github.ajalt.clikt.core.CliktCommand.main(CliktCommand.kt:298)
         at com.google.prefab.cli.AppKt.main(App.kt:28)`

RN version: 0.67.5
react-native-vision-camera: 3.2.2
kotlin: 1.5.31
ndkVersion: 23.1.7779620

### What I tried

i tried upgrading kotlin version to 1.7 , however the issue still remains the same

### VisionCamera Version

3.2.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

powerpivar, 2023-10-04:
no, there is no release with this fix but you can use a branch as the source `ssh://git@github.com:mrousavy/react-native-vision-camera.git#v2`, keep in mind that it can be risky because of new changes.

mrousavy, 2023-10-04:
There is some issue with prefabs, I don't know how you achieved this but clear your cache. You also created a question instead of a build error bug report, why?