margelo/react-native-vision-camera #1917: 🔧iOS Build - Frame Processor issue

pgpankajcapgemini, 2023-10-04:
### How were you trying to build the app?

I am stuck after build an example app with iOS using frame processor to scan QR code.

### Full build logs

```tsx
FrameProcessorBindings: Finished installing bindings.
Thread Performance Checker: Thread running at QOS_CLASS_USER_INTERACTIVE waiting on a lower QoS thread running at QOS_CLASS_DEFAULT. Investigate ways to avoid priority inversions
PID: 10224, TID: 26604477
Backtrace
=================================================================
3   scanner                             0x0000000100992c3b -[SRRunLoopThread runLoop] + 43
4   scanner                             0x000000010098cd47 +[NSRunLoop(SRWebSocket) SR_networkRunLoop] + 71
5   scanner                             0x0000000100991572 -[SRProxyConnect _openConnection] + 82
6   scanner                             0x000000010099094c -[SRProxyConnect _configureProxy] + 988
7   scanner                             0x00000001009900ab -[SRProxyConnect openNetworkStreamWithCompletion:] + 91
8   scanner                             0x00000001009949cb -[SRWebSocket open] + 731
9   scanner                             0x000000010068ce8d -[RCTReconnectingWebSocket start] + 173
10  scanner                             0x000000010067b50b -[RCTPackagerConnection init] + 507
11  scanner                             0x000000010067b2f4 __49+[RCTPackagerConnection sharedPackagerConnection]_block_invoke + 36
12  libdispatch.dylib                   0x000000010e7227ec _dispatch_client_callout + 8
13  libdispatch.dylib                   0x000000010e723e85 _dispatch_once_callout + 66
14  scanner                             0x000000010067b29e +[RCTPackagerConnection sharedPackagerConnection] + 78
15  scanner                             0x000000010071612a -[RCTDevSettings initialize] + 170
16  scanner                             0x00000001006645c0 -[RCTModuleData _initializeModule] + 112
17  scanner                             0x0000000100663e3d -[RCTModuleData setUpInstanceAndBridge:] + 2589
18  scanner                             0x0000000100665dc0 -[RCTModuleData instance] + 1408
19  scanner                             0x000000010060ac2c -[RCTCxxBridge moduleForName:lazilyLoadIfNecessary:] + 812
20  scanner                             0x0000000100670417 -[RCTModuleRegistry moduleForName:lazilyLoadIfNecessary:] + 135
21  scanner                             0x000000010067037e -[RCTModuleRegistry moduleForName:] + 46
22  scanner                             0x0000000100728523 -[RCTPerfMonitor devMenuItem] + 99
23  scanner                             0x00000001007283e2 -[RCTPerfMonitor initialize] + 114
24  scanner                             0x00000001006645c0 -[RCTModuleData _initializeModule] + 112
25  scanner                             0x0000000100663e3d -[RCTModuleData setUpInstanceAndBridge:] + 2589
26  scanner                             0x0000000100666087 __25-[RCTModuleData instance]_block_invoke + 39
27  scanner                             0x00000001006d2d68 RCTUnsafeExecuteOnMainQueueSync + 56
28  scanner                             0x0000000100665c02 -[RCTModuleData instance] + 962
29  scanner                             0x0000000100610525 __49-[RCTCxxBridge _prepareModulesWithDispatchGroup:]_block_invoke + 181
30  libdispatch.dylib                   0x000000010e72154f _dispatch_call_block_and_release + 12
31  libdispatch.dylib                   0x000000010e7227ec _dispatch_client_callout + 8
32  libdispatch.dylib                   0x000000010e7337e9 _dispatch_main_queue_drain + 1725
33  libdispatch.dylib                   0x000000010e73311e _dispatch_main_queue_callback_4CF + 31
34  CoreFoundation                      0x00000001111f06cc __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
35  CoreFoundation                      0x00000001111eafbe __CFRunLoopRun + 2429
36  CoreFoundation                      0x00000001111ea264 CFRunLoopRunSpecific + 560
37  GraphicsServices                    0x000000011b47e24e GSEventRunModal + 139
38  UIKitCore                           0x0000000129bd77bf -[UIApplication _run] + 994
39  UIKitCore                           0x0000000129bdc5de UIApplicationMain + 123
40  scanner                             0x00000001001e5788 main + 104
```

### Project dependencies

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.5",
    "react-native-reanimated": "3.5.4",
    "react-native-vision-camera": "2.16.1",
    "vision-camera-code-scanner": "0.2.0"
  },
```

### VisionCamera Version

2.16.1

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

pgpankajcapgemini, 2023-10-05:
I am closing this ticket because I got new update by this library with latest version 3.3.0 which is working fine.

closed 2023-10-05 by pgpankajcapgemini

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.