margelo/react-native-vision-camera #1919: 🐛 crash when enabling

Skizu, 2023-10-04:
### What's happening?

Crash when starting local debugger in browser.

**NOTE** `gradle.properties` excerpt
```
# gradle.properties snipper

# Use this property to specify which architecture you want to build.
# You can also override it from the CLI using
# ./gradlew <task> -PreactNativeArchitectures=x86_64
reactNativeArchitectures=armeabi-v7a,arm64-v8a

# Use this property to enable support to the new architecture.
# This will allow you to use TurboModules and the Fabric render in
# your application. You should enable this flag either if you want
# to write custom TurboModules/Fabric components OR use libraries that
# are providing them.
newArchEnabled=false

# Use this property to enable or disable the Hermes JS engine.
# If set to false, you will be using JSC instead.
hermesEnabled=false
```

**NOTE** Frame Processors are enable but not `react-native-worklets-core` due to v2. Unable to use V3 due to numerous build issues on Android and QR Code scanner in documentation is **not** compatible.

### Reproduceable Code

```tsx
Example app.
```

### Relevant log output

```shell
11:31:02.095 dragonforce.de    A  java_vm_ext.cc:577] JNI DETECTED ERROR IN APPLICATION: java_object == null
                                  java_vm_ext.cc:577]     in call to GetObjectClass
                                  java_vm_ext.cc:577]     from com.facebook.react.turbomodule.core.CallInvokerHolderImpl com.facebook.react.bridge.CatalystInstanceImpl.getJSCallInvokerHolder()
```

### Camera Device

```json
Unable to see due to crash.
```

### Device

Samsung S10-S22, Samsung XCover Pro V3-6, Pixel 3a-6

### VisionCamera Version

2.16.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

trcoffman, 2023-10-04:
Are you using the old react-native chrome debugging where the JS runtime runs in the browser instead of on the device? That's where you enable debugging from the react-native in-app developer menu. This runs the app's JS bundle inside of chrome's V8 runtime and the react-native message bridge communicates events between native and the browser over the network.

You can't do that workflow anymore once you start using any libraries that use JSI. This library uses JSI to synchronously call native code from JS. This library literally can't work when using chrome debugging, because the JS has to be running on the same CPU as the native code. All code paths that involve JSI will crash when running your app with chrome debugging.

If you adopt any JSI libraries, then if you want to be able to use a debugger you have to switch to Hermes instead of using JavaScriptCore. Hermes allows you to connect a debugger over the network to the Hermes runtime running on the device. Unfortunately, debugging with Hermes isn't as good of an experience as the old Chrome debugging experience, especially around Profiling and inspecting memory usage and stuff like that. For just stepping through code with breakpoints it's fine but all the other stuff like network & memory & cpu profiling is not as good. I use Flipper to debug our react-native Hermes apps and it's OK but it's not as good as chrome debugging.

You have to choose: Use JSI libraries and have great performance and powerful capabilities but sacrifice great developer debugging experience, or avoid all JSI libraries and preserve the ability to use chrome debugging.

mrousavy, 2023-10-05:
Exactly what @trcoffman said. Thanks Tyler! ❤️

closed 2023-10-05 by mrousavy