margelo/react-native-vision-camera #1921: 🐛

valn1, 2023-10-04:
### What's happening?

when using the newly developed qrcode scanner, the camera view starts skipping frames after some seconds of running time
the scanner still works fine, but the view lags overtime and also, the fps prop does not change how often the scanner scans 

### Reproduceable Code

```tsx
import { useEffect } from "react";
import { StyleSheet } from "react-native";
import { Camera, useCameraDevice, useCameraPermission, useCodeScanner } from "react-native-vision-camera";

const Scanner = (props: NavigationProps) => {
    const { hasPermission, requestPermission } = useCameraPermission();
    const device = useCameraDevice("back");

    const codeScanner = useCodeScanner({
        codeTypes: ["qr"],
        onCodeScanned: (codes) => {
            console.log(codes);
        },
    });

    useEffect(() => {
        if (!hasPermission) {
            requestPermission().then(console.log).catch(console.warn);
        }
    }, []);

    return (
        <View style={StyleSheet.absoluteFill}>
            {hasPermission && device &&
                    <Camera style={StyleSheet.absoluteFill} device={device} isActive={true} codeScanner={codeScanner} fps={1} />
            )}
        </Screen>
    );
};

export default Scanner;
```

### Relevant log output

```shell
10-04 14:14:41.414  2917  2917 E pageboostd: Received HALT command code 2
10-04 14:14:41.446 16279 16279 E ssioncontroller: Not starting debugger since process cannot load the jdwp agent.
10-04 14:14:41.549  1215  6542 E CameraService: CameraService::connect evicting conflicting client for camera ID 0
10-04 14:14:41.549  1215  6542 E CameraDeviceClient: notifyError: pid=13634, errorCode=0, errorStreamId=-1, frameNumber=0
10-04 14:14:41.549  1215  6542 E CameraDeviceClient: notifyError: pid=13634, skipClientNotification=false
10-04 14:14:41.549  1215  6542 E CameraDeviceClient: Disconnect from CameraDeviceClient
10-04 14:14:41.559  1215  6542 E Camera2-FrameProcessorBase: removeListener: 
10-04 14:14:41.614  2917  2917 E pageboostd: Received HALT command code 2
10-04 14:14:41.652  1051 10753 E libcsc  : csc_deinit:: unsupported csc_hw_type
10-04 14:14:41.652  1051 10753 E libcsc  : csc_deinit:: unsupported csc_hw_type
10-04 14:14:41.652  1051 10753 E libcsc  : csc_deinit:: unsupported csc_hw_type
10-04 14:14:41.652  1051 10753 E ExynosCameraPlugIn: [CAM(0)][BayerGDCPL]-(destroy[143]):stop() was skipped!!(2)
10-04 14:14:41.665  1026  1700 E Sensors : Sensor : Meta event
10-04 14:14:41.692  1051 10753 E libcsc  : csc_deinit:: unsupported csc_hw_type
10-04 14:14:41.692  1051 10753 E libcsc  : csc_deinit:: unsupported csc_hw_type
10-04 14:14:41.692  1051 10753 E libcsc  : csc_deinit:: unsupported csc_hw_type
10-04 14:14:41.712  2640  2640 E SecHAL  : nfc_hal_write: [Cmd]Send EVT data_sending_flag = true, return size=4!!
10-04 14:14:41.723  1051 10753 E ExynosCamera: Build Date is (Jul 21 2023) (20:21:37) Process 64 bit
10-04 14:14:41.725  1051 10753 E FMQ     : grantorIdx must be less than 3
10-04 14:14:41.725  1051 10753 E FMQ     : grantorIdx must be less than 3
10-04 14:14:41.727  1215  6542 E Camera2-FrameProcessorBase: FrameProcessorBase: created
10-04 14:14:41.727  1215  6542 E Camera2-FrameProcessorBase: registerListener: Registering listener for frame id range 0 - 2147483647
10-04 14:14:41.730 13634 13740 E CameraView: invokeOnError(...):
10-04 14:14:41.731  1215  3785 E CameraDeviceClient: Disconnect from CameraDeviceClient
10-04 14:14:41.739  2640  3942 E SecHAL  : nfc_data_callback: [Cmd]Received a response packet : data_sending_flag = false!!
10-04 14:14:41.739  2454  3939 E libnfc_nci: [ERROR:nfc_ncif.cc(1462)] nfc_ncif_proc_deactivate - evt_data.deactivate.reason 0, nfc_cb.deact_reason = 0
[... 405 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung s20 plus

### VisionCamera Version

3.3.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

valn1, 2023-10-04:
https://drive.google.com/file/d/1Y_-oep6_d8V9sHG0OZc2wiwsHmJPRJjU/view?usp=drive_link
this is a video of the bug happening

jason-digiex, 2023-10-05:
+1 Hope this resolve soon.

mrousavy, 2023-10-05:
Hi @valn1, thanks for the bug report! I'll try to take a look at this soon.

> +1 Hope this resolve soon.

@jason-digiex if you want this to be resolved soon consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to support me developing this.

jmunckhof, 2023-10-05:
+1 Have the same issue

BrechtBD, 2023-10-06:
Same issue here!

mgcrea, 2023-10-06:
it just so happens that I was working on a custom frame processor to scan barcodes before @mrousavy bundled it to the core of this library! Litteray worked on it and released it a couple of days before it "landed" officially here 😅!

Anyway I do not encounter frame skipping at all while using it (either on iOS or Android) so it is technically doable.

I've also added basic barcode highlighting that works well and that was relatively painful to implement 😅!

You can check a working demo on GitHub [@mgcrea/vision-camera-barcode-scanner](https://github.com/mgcrea/vision-camera-barcode-scanner).

This work may not have much future now that it is a core feature but maybe there will be some corner cases or experiments that will be easier to achieve using an external frame processor.

Thanks again @mrousavy for your awesome work on this library and for sharing it with all of us!

mrousavy, 2023-10-06:
Hey @mgcrea!

> Thanks again @mrousavy for your awesome work on this library and for sharing it with all of us!

Thanks for the kind words!

> I've also added basic barcode highlighting that works well and that was relatively painful to implement 😅 You can check a working demo on GitHub [@mgcrea/vision-camera-barcode-scanner](https://github.com/mgcrea/vision-camera-barcode-scanner).

Amazing, nice work!

> This work may not have much future now that it is a core feature but maybe there will be some corner cases or experiments that will be easier to achieve using an external frame processor.

Definitely, maybe some different QR/Barcode types, some special native parsing, or some other edge cases like synchronously being able to get/draw the Frame is definitely better off in an external Frame Processor Plugin, they are just insanely powerful and capable of this.

The integrated CodeScanner here is a simple implementation that should cover most basic use-cases. Anything more advanced should be a Frame Processor :)

hengkx, 2023-10-07:
I  use 3.3.1. 
Android cannot recognize any QR codes

mrousavy, 2023-10-10:
What happens if you switch the Queues here?

https://github.com/mrousavy/react-native-vision-camera/blob/a498144fd5786c6dba1d78730d362e0f675cbaf7/package/android/src/main/java/com/mrousavy/camera/core/outputs/CameraOutputs.kt#L201

Either `codeScannerQueue` or `videoQueue`? 

Can someone try this?

mrousavy, 2023-10-10:
Maybe this fixes it: https://github.com/mrousavy/react-native-vision-camera/pull/1971

closed 2023-10-10 by mrousavy