margelo/react-native-vision-camera #1923: 🐛 Error Domain=AVFoundationErrorDomain Code=-11803 "Cannot Record" UserInfo={AVErrorRecordingFailureDomainKey=3, NSLocalizedDescription=Cannot Record, NSLocalizedRecoverySuggestion=Try recording again.}

vaishnav-197, 2023-10-05:
### What's happening?

This is happening for us in production when the user tries to take the photo. Occurring in iPhone 13 ( 17.0.2 )
This is occurring randomly and not in every iPhone 13s. Sentry logs also shows that its been occurring in iPhone 12,14 Pro with iOs 16.6 and 16.5.1 being predominant

React Native : 0.67.5
react-native-vision-camera: 2.13.0

Also attached the logs found

### Reproduceable Code

```tsx
const camera = useRef(null);
  const devices = useCameraDevices('back');

  <Camera ref={camera} device={device} style={styles.preview} isActive photo />
```

### Relevant log output

```shell
[native code] in construct
node_modules/@babel/runtime/helpers/construct.js in <global> at line 16:21

node_modules/@babel/runtime/helpers/wrapNativeSuper.js in SyntheticError at line 17:14

[native code] in construct
node_modules/react-native-vision-camera/src/Camera.tsx in <global> at line 496:3

node_modules/react-native-vision-camera/src/CameraError.ts in <global> at line 133:5

[native code] in construct
node_modules/react-native-vision-camera/src/Camera.tsx in <global> at line 496:3

node_modules/react-native-vision-camera/src/CameraError.ts in <global> at line 151:32

node_modules/react-native-vision-camera/src/CameraError.ts in <global> at line 196:18

node_modules/react-native-vision-camera/src/Camera.tsx in IconToolbarAndroid at line 124:13

node_modules/@babel/runtime/helpers/regeneratorRuntime.js in <global> at line 44:17

node_modules/@babel/runtime/helpers/regeneratorRuntime.js in <global> at line 125:22

node_modules/@babel/runtime/helpers/regeneratorRuntime.js in <global> at line 44:17

node_modules/@babel/runtime/helpers/regeneratorRuntime.js in <global> at line 75:20

node_modules/@babel/runtime/helpers/regeneratorRuntime.js in <global> at line 82:11

node_modules/promise/setimmediate/core.js in minimal_1.configureScope$argument_0 at line 37:12

node_modules/promise/setimmediate/core.js in <global> at line 123:15

node_modules/react-native/Libraries/Core/Timers/JSTimers.js in <global> at line 248:18

node_modules/react-native/Libraries/Core/Timers/JSTimers.js in <global> at line 112:7

node_modules/react-native/Libraries/Core/Timers/JSTimers.js in <global> at line 166:5

node_modules/react-native/Libraries/Core/Timers/JSTimers.js in <global> at line 418:12

node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js in <global> at line 393:12

node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js in <global> at line 135:12

node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js in <global> at line 370:9

node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js in <global> at line 134:10
```

### Camera Device

```json
device:
devices: ["wide-angle-camera"]
hasFlash: true
hasTorch: false
id: "com.apple.avfoundation.avcapturedevice.built-in_video:1"
isMultiCam: false
maxZoom: 128.875
minZoom: 1
name: "Front Camera"
neutralZoom: 1
position: "front"
supportsDepthCapture: false
supportsFocus: false
supportsLowLightBoost: false
supportsParallelVideoProcessing: true
supportsRawCapture: false
```

### Device

iPhone 13

### VisionCamera Version

2.13.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-05:
Hi! Is there any way you could drop the native logs (symbolicated) here as well?

mrousavy, 2023-10-05:
For Sentry: [Debug Symbols](https://docs.sentry.io/platforms/react-native/upload-debug/?original_referrer=https%3A%2F%2Fwww.google.com%2F) ([symbolicate](https://docs.sentry.io/platforms/native/debug-information/))

mrousavy, 2023-10-16:
Hey! Two weeks no reply, I'm closing this for now.

If you have more information, please post them here and I'll gladly re-open the issue.

mrousavy, 2023-10-16:
As per [this StackOverflow post](https://stackoverflow.com/questions/24176738/error-domain-avfoundationerrordomain-code-11814-cannot-record): 

> Having your device restrict the camera access under "Settings > General > Restrictions" will also give you this error.

Could it be that the user disabled Camera access? In that case, `useCameraPermission()` would probably return `"restricted"`. Do you check for that?

mrousavy, 2023-10-16:
Also, [searching for this error on StackOverflow](https://stackoverflow.com/search?q=%22-11803%22+Cannot+Record) could be of help. 

Please let me know if you find anything so I can fix it upstream in VisionCamera!

closed 2023-10-16 by mrousavy

spsaucier, 2024-07-02:
I'm getting this on iOS on Vision 4.3.2 up to 900 times per day with about 10k daily active users. It looks like 

`Error Domain=AVFoundationErrorDomain Code=-11803 "Cannot Record" UserInfo={AVErrorRecordingFailureDomainKey=3, NSLocalizedDescription=Cannot Record, NSLocalizedRecoverySuggestion=Try recording again.}`

I am using `useCameraPermission`'s `hasPermission` and `requestPermission` functions.

```
      if (!hasCameraPermission && Platform.OS === 'ios') {
        const newPermission = await requestCameraPermission();
        if (!newPermission) {
          Toast.show({
            description: t('permissions.camera.weNeedPermissions'),
            placement: 'top',
          });
        }
      }
```

mrousavy, 2024-07-03:
That's not enough info for me, and also what does "cannot record" have to do with permissions?

I'd need an issue with logs and your code snippet usage of the Camera component + startRecording

spsaucier, 2024-07-03:
Apologies, I was referring to your comment on this issue originally:

> Could it be that the user disabled Camera access? In that case, useCameraPermission() would probably return "restricted". Do you check for that?

mrousavy, 2024-07-03:
Ah I see - well hm then I don't know, I haven't seen this before. You'd need to create an issue with logs and repro

rsainiWin, 2024-07-25:
I also faced the same @mrousavy 
 [image]

felipejh, 2024-11-18:
I have the same issue here in a simple app. And I also have seen this issue being reported at Apple Developer's forum.

yoann84, 2025-02-24:
Same issue

leomiller12, 2025-04-14:
Same issue. Someone found a fix?

leomiller12, 2025-04-14:
@mrousavy reopen this issue. I do have this error on some iPhone in production, can't catch it locally tho.

gontovnik, 2025-04-30:
@mrousavy I am having the same issue. Any ideas on how to resolve this?

edritech93, 2025-08-08:
same issue for me

lucksp, 2025-08-25:
I am seeing this on production today:

- iPhone 16 Pro
```
    "expo": "~52.0.46",
    "react-native": "0.76.9",
    "react-native-vision-camera": "^4.6.4",
```

ashishsarin-fareye, 2026-01-12:
@mrousavy we are still facing this issue on prod app. This is happening randomly for iOS users.