margelo/react-native-vision-camera #1925: 🐛 Changing camera device causes a crash if code scanner is being used

hsjoberg, 2023-10-05:
### What's happening?

If the code scanner is being used via `useCodeScanner()`, then an insta-crash will happen on Android if you switch camera device (with `useCameraDevice()`).
At least this is the case on Android. I will take a look at iOS later today.

Logging output is from an Android 14 emulator. The error happens on a real device (Samsung S22 Ultra) too.

If it's of any help, the stack trace crash goes specifically to these lines of code (react-native-vision-camera@3.3.0):

[com.mrousavy.camera.core.CameraSession.updateRepeatingRequest(CameraSession.kt:581)](https://github.com/mrousavy/react-native-vision-camera/blob/3ed74e67941af626e3a5bd6a15d515aa9f3569a9/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L581)
[com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:559)](
https://github.com/mrousavy/react-native-vision-camera/blob/3ed74e67941af626e3a5bd6a15d515aa9f3569a9/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L559)
com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:49) (just a trace to the class itself)
[com.mrousavy.camera.core.CameraSession$configureFormat$1.invokeSuspend(CameraSession.kt:203)](https://github.com/mrousavy/react-native-vision-camera/blob/3ed74e67941af626e3a5bd6a15d515aa9f3569a9/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L203)

### Reproduceable Code

```tsx
// Run this code to observe the error

import React, { useState, useEffect } from "react";
import { Button, StyleSheet, Text, View } from "react-native";
import { Camera, useCodeScanner, CameraPosition, useCameraDevice, useCameraPermission } from "react-native-vision-camera";

export default function SendCamera() {
  const [cameraType, setCameraType] = useState<CameraPosition>("back");
  const device = useCameraDevice(cameraType);
  const { hasPermission, requestPermission } = useCameraPermission();
  const codeScanner = useCodeScanner({
    codeTypes: ["qr"],
    onCodeScanned: (codes) => {}
  });

  useEffect(() => {
    if (hasPermission === false) {
      console.log("Does not have camera permission");
      requestPermission().catch(() => {});
    }
  }, [requestPermission, hasPermission]);

  const onCameraSwitchClick = () => {
    setCameraType(
      cameraType === "front"
        ? "back"
        : "front"
    );
  };

  if (!hasPermission) {
    return (<Text>Camera permission was denied</Text>);
  }

  if (!device) {
    return (<></>);
  }

  return (
    <>
      <Camera
        style={StyleSheet.absoluteFill}
        codeScanner={codeScanner}
        device={device}
        isActive={true}
      />
      <View style={{ marginTop: 100 }}>
        <Button title="flip" onPress={onCameraSwitchClick} />
      </View>
    </>
  );
};
```

### Relevant log output

```shell
10-05 10:58:39.158  8996  8996 I CameraSession: Setting Format (fps: null | videoStabilization: null | hdr: null | lowLightBoost: null)...
10-05 10:58:39.160  8996  8996 I CameraOutputs: Preparing Outputs for Camera 10...
10-05 10:58:39.161  8996  8996 I CameraOutputs: Adding native preview view output.
10-05 10:58:39.180  8996  8996 I CameraOutputs: Adding 1280x720 code scanner output. (Code Types: [256])
10-05 10:58:39.181  8996  8996 I CameraOutputs: Prepared 2 Outputs for Camera 10!
10-05 10:58:39.182  8996  8996 I CameraSession: Setting isActive: true (isRunning: true)
10-05 10:58:39.183  8996  9160 I CameraSession: Starting Camera Session...
10-05 10:58:39.184  8996  9160 I CameraSession: Adding preview output surface PREVIEW..
10-05 10:58:39.184  8996  9160 I CameraSession: Adding code scanner output surface VIDEO
10-05 10:58:39.184  8996  9160 I CameraSession: Camera Session initialized! Starting repeating request..
10-05 10:58:39.189  8996 10177 D TransportRuntime.SQLiteEventStore: Storing event with priority=DEFAULT, name=FIREBASE_ML_SDK for destination cct
10-05 10:58:39.192  8996  9160 D CameraSession: Setting Repeating Request..
10-05 10:58:39.193  8996  8996 I PreviewView: Surface created! Surface(name=null)/@0xc9aeca7
10-05 10:58:39.193  8996  8996 I CameraView: Configuring Camera Device...
10-05 10:58:39.193  8996  8996 I CameraSession: Configuring Session for Camera 1...
10-05 10:58:39.194  8996  9160 E Camera-SurfaceUtils-JNI: SurfaceUtils_nativeGetSurfaceId: Could not retrieve native Surface from surface.
10-05 10:58:39.199  8996  8996 I CameraOutputs: Preparing Outputs for Camera 1...
10-05 10:58:39.199  8996  8996 I CameraOutputs: Adding native preview view output.
10-05 10:58:39.204  8996  8996 I CameraOutputs: Adding 1280x720 code scanner output. (Code Types: [256])
10-05 10:58:39.204  8996  8996 I CameraOutputs: Prepared 2 Outputs for Camera 1!
10-05 10:58:39.204  8996  8996 I SurfaceOutput: Closing BarcodeScanner..
10-05 10:58:39.204  8996  8996 I SurfaceOutput: Closing 1280x720 VIDEO ImageReader..
10-05 10:58:39.205  8996  8996 I PreviewView: Surface resized! Surface(name=null)/@0xc9aeca7 (1920 x 1080 in format #4)
10-05 10:58:39.226  8996  9184 E BufferQueueProducer: [SurfaceView[com.blixtwallet.debug/com.blixtwallet.MainActivity]#13(BLAST Consumer)13](id:23240000000d,api:4,p:416,c:8996) queueBuffer: BufferQueue has been abandoned
10-05 10:58:39.226   416  2268 E Surface : queueBuffer: error queuing buffer, -19
10-05 10:58:39.226   416  2268 E Camera3-OutputStream: returnBufferCheckedLocked: Stream 0: Error queueing buffer to native window: No such device (-19)
--------- beginning of crash
10-05 10:58:39.263  8996  9160 E AndroidRuntime: FATAL EXCEPTION: mrousavy/VisionCamera.main
10-05 10:58:39.263  8996  9160 E AndroidRuntime: Process: com.blixtwallet.debug, PID: 8996
10-05 10:58:39.263  8996  9160 E AndroidRuntime: java.lang.IllegalArgumentException: CaptureRequest contains unconfigured Input/Output Surface!
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.hardware.camera2.CaptureRequest.convertSurfaceToStreamId(CaptureRequest.java:765)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:1274)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest(CameraDeviceImpl.java:1330)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:313)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession.updateRepeatingRequest(CameraSession.kt:581)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:559)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:49)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at com.mrousavy.camera.core.CameraSession$configureFormat$1.invokeSuspend(CameraSession.kt:203)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:958)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:205)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:294)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:67)
10-05 10:58:39.263  8996  9160 E AndroidRuntime: 	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@401273e, mrousavy/VisionCamera.main]
10-05 10:58:39.285   552  2215 W ActivityTaskManager:   Force finishing activity com.blixtwallet.debug/com.blixtwallet.MainActivity
```

### Camera Device

```json
N/A. Could provide if really needed. But the bug appears in Android emulators too
```

### Device

Android 14 emulator

### VisionCamera Version

3.3.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hsjoberg, 2023-10-05:
I wasn't sure if it's actually a dupe of #1773. The Javascript code here is different, though the stack trace seems to be the same.
Feel free to close.

mrousavy, 2023-10-06:
Yea duplicate, but can you please post your reproduction steps (flipping Cam in code scanner) over in #1773? Thanks!

closed 2023-10-06 by mrousavy