margelo/react-native-vision-camera #1931: ✨ Basic web support

nandorojo, 2023-10-05:
### What feature or enhancement are you suggesting?

Could be cool...

### What Platforms whould this feature/enhancement affect?

iOS

### Alternatives/Workarounds

Use `expo-camera` for now or a JS-based camera.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-10-06:
Yea web support probably wouldn't be that difficult, I think I could do it in a week or a month max.

But I don't have any apps running in react-native-web, so I don't have a need for that. If anyone wants to pay me for that, I'm open to get it into VisionCamera main :)

mfkrause, 2024-02-24:
@mrousavy Are you still open to implementing this or have there been changes to VisionCamera that would act as a blocker to web support in the meantime? Depending on how long this would take with the current feature set (including frame processors & plugins, which would be our use case for Vision Camera on the web), and, by extension, how much money we're looking at, we might be open to sponsoring this.

mrousavy, 2024-02-24:
@mfkrause I think I'd be open to implement this if I can somewhat rely on somebody (eg you) to test the thing on web and contribute fixes if something breaks.
Let's discuss in detail, can you message us through my agency's website? https://margelo.io

mfkrause, 2024-02-24:
@mrousavy Gotcha. Whether or not we'd be able to fund this and actively contribute fixes would depend on a customer project that's currently in the concept phase. We're still working out if react-native-web would work for this project or if we're going to implement web support through React.js. Once that has been sorted out (which should hopefully happen sometime next week), I'll shoot you a message.

mfkrause, 2024-03-25:
@mrousavy I've sent a request through the contact form a few weeks ago. You guys had a chance to look at it, or didn't it come through?

mrousavy, 2024-03-25:
Hey @mfkrause - I didn't receive anything from your name in that inbox 🤔, maybe try directly reaching out to us through hello (at) margelo (dot) io?

verofy-simon-denny, 2024-07-18:
Hi @mrousavy,

Appreciate all the work you do on this. I just wondered if you were in the process of implementing web support and if you had any sort of time frames as to when this may be ready if you were?

Many Thanks, Simon.

Erotemic, 2025-04-14:
How do you even develop without web support? I'd like to use this package, but as someone new to javascript I don't see how to even get start trying to learn how to use this.