margelo/react-native-vision-camera #1936: How to read a barcode by pushing a button with "QR/Barcode Scanning"❓

Productivix, 2023-10-05:
### Question

I want to ask the camera to read the barcode :

 I read mainly the code of page : https://www.react-native-vision-camera.com/docs/guides/code-scanning 

### What I tried

I try to start the scan by a button that starts handleBarCodeScanned_via_vision() but it starts to take a picture and it should have scan !

but the 
```
const ScanCBVision = ()=>{
const [barcodeDecoding, setBarcodeDecoding] = useState(false);
  const device = useCameraDevice('back'); 
  const camera = useRef(null)

  const handleBarCodeScanned_via_vision = async () => {
    setBarcodes([]);
    setBarcodeDecoding(true);
    try {
      if (camera.current == null) throw new Error('Camera ref is null!');
  
    const photo = await camera.current.takePhoto({
        qualityPrioritization: 'balanced',
        flash: 'on',
        enableShutterSound: false,
        enableAutoStabilization: true,
      }); 
    } catch (error) {
      console.log('TryCatch de handleBarCodeScanned_via_vision', error.message);
    }
    setBarcodeDecoding(false);
  };

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      console.log(`Scanned ${codes.length} codes!`);
      setBarcodes(codes);
    },
  });
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      console.log(`Scanned ${codes.length} codes!`);
      setBarcodes(codes);
    },
  });

  if (device == null) {
    return <ActivityIndicator />;
  }

  return (
    <View
      style={[
        portrait ? CommonPrxStyles.containerRow : CommonPrxStyles.container,
        {flex: 1},
      ]}>
      <View
        style={[
          CommonPrxStyles.subContainer,
          {flex: 2, margin: 3, justifyContent: 'space-between'},
        ]}>
        {!barcodeDecoding ? (
          <TouchableWithoutFeedback
            style={style.subContainerScan}
            onPress={() => {
              console.log('appui sur bouton');
              handleBarCodeScanned_via_vision();
            }}>
            <Text style={[CommonPrxStyles.button, CommonPrxStyles.title]}>
              Scan: {!barcodeDecoding && barcodes.length}
            </Text>
          </TouchableWithoutFeedback>
        ) : (
          <ActivityIndicator />
        )}
        <ScrollView>
          {barcodes.map((label, idx) => (
            <Text key={idx}>
              {label.value} - {label.police}
            </Text>
          ))}
        </ScrollView>
      </View>
      <View style={CommonPrxStyles.subContainer}>
        <Camera
          ref={camera}
          //style={style.scanSmallWindow}
          style={StyleSheet.absoluteFill}
          //style={style.scanSmallWindow}
          device={device}
          isActive={true}
          //torch={'on'}
          //photo={true}
          enableZoomGesture={true}
          lowLightBoost={true}
          //zoom={5}
          onInitialized={console.log('camera ok')}
          codeScanner={codeScanner}
          //frameProcessor={frameProcessor}
          //frameProcessorFps={5}
        />
      </View>
    </View>
  );
};

const style = StyleSheet.create({
  subContainerScan: {
    flex: 1,
    padding: 1,
    margin: 2,
    borderBottomWidth: 2,
    borderColor: 'green',
    borderWidth: 2,
  },
  scanSmallWindow: {
    //flex: 1,
    minHeight: 60,
    minWidth: 60,
  },
});

export default ScanCBVision;
```

What should I do instead ?

### VisionCamera Version

3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-06:
WHAT?

You're calling `takePhoto()`, it takes a photo.

closed 2023-10-06 by mrousavy

Productivix, 2023-10-06:
Very sympatic.. and you close for me ...It is not solved sorry.

Productivix, 2023-10-06:
Yes I know that I take photo, but I want to get barcode : the question is how, it is not documented.