margelo/react-native-vision-camera #1937: 🐛[v2.16.2] Frame Processors are not enabled

pasquale95, 2023-10-05:
### What's happening?

I updated the `react-native-vision-camera` from `2.15.4` to `2.16.2` to overcome the `load+` problem on iOS, but after the update the camera stops working as soon as I open it on both iOS and Android.
I also tried to use the version `2.16.1` and the issue still persists.
The 2.16.2 version should not require `react-native-worklets-core`, right? This should be mandatory from v3.

### Reproduceable Code

```tsx
const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.QR_CODE], {
    checkInverted: true,
  });
useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);
return device != null && hasPermission ? (
    <Camera
      style={styles.scanner}
      device={device}
      isActive={isFocused}
      frameProcessor={frameProcessor}
      frameProcessorFps={5}
    />
  ) : null;
```

### Relevant log output

```shell
ERROR  frame-processor/unavailable: Frame Processors are not enabled. See https://react-native-vision-camera.com/docs/guides/troubleshooting, js engine: hermes
```

### Camera Device

```json
{
   "maxZoom":8,
   "minZoom":1,
   "supportsLowLightBoost":false,
   "supportsDepthCapture":false,
   "neutralZoom":1,
   "supportsFocus":true,
   "supportsRawCapture":false,
   "hasFlash":true,
   "name":"back (0)",
   "supportsParallelVideoProcessing":false,
   "isMultiCam":false,
   "hasTorch":true,
   "devices":[
      "wide-angle-camera"
   ],
   "position":"back",
   "id":"0"
}
```

### Device

SM-A336B - Android 13

### VisionCamera Version

2.16.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

2sem, 2023-10-06:
Xcode 14.3.1

[image]

ombogdanJoinToIt, 2023-10-06:
same issue

fedpre, 2023-10-06:
I think I have a similar issue where after upgrading to get rid of the Xcode 15 problem, the frame processor doesn't recognize my custom frame processor anymore. I cannot share code because it is proprietary of the company I am developing for, but I have created a bodyPose for iOS frame processor and now when I try to activate it it says that it doesn't find it. I have added it to the `globals` plugins and it was working before the change.

podger, 2023-10-06:
Since version 2.16.0  react-native-reanimanted 3.5+ is needed for the Frame Processors to be enabled.

ongdetui, 2023-10-06:
same issue. frame-processor/unavailable: Frame Processors are not enabled. using reanimanted v3 ?

Danushka96, 2023-10-06:
> 3.5

Works for me

Danushka96, 2023-10-06:
> Since version 2.16.0 react-native-reanimanted 3.5+ is needed for the Frame Processors to be enabled.

Works for me

pasquale95, 2023-10-06:
> Since version 2.16.0 react-native-reanimanted 3.5+ is needed for the Frame Processors to be enabled.

If I switch to the latest `react-native-reanimated` 3.5.4 I get this error coming from `react-native-vision-camera`:
```
  CMake Error at CMakeLists.txt:208 (find_package):
    Could not find a package configuration file provided by
    "react-native-reanimated" with any of the following names:
  
      react-native-reanimatedConfig.cmake
      react-native-reanimated-config.cmake
  
    Add the installation prefix of "react-native-reanimated" to
    CMAKE_PREFIX_PATH or set "react-native-reanimated_DIR" to a directory
    containing one of the above files.  If "react-native-reanimated" provides a
    separate development package or SDK, be sure it has been installed.
```

mrousavy, 2023-10-06:
Reanimated 3 is required for VisionCamera 2.16.

I really recommend you guys to upgrade to VisionCamera V3

fedpre, 2023-10-06:
@mrousavy that's actually what I have suggested to my company and gave me the green light. I will updated several other stuff including to VisionCamera V3, as it seems that the frameprocessor issue that I had were resolved in the latest releases. Excited to see it in action.

mrousavy, 2023-10-06:
& V3 now has a QR/Barcode scanner :)

fedpre, 2023-10-06:
I know! We use the old qr scanner but I want to get rid of it and use the one from the library as we use a QR code as some sort of config for the app

I wish I would be more knowledgable to help you with the orientation issue. I will try to take a look at it because you do a fantastic job at keeping up with this awesome library and I want to help

danieloprado, 2023-10-06:
I've try to use version 3 with vision-camera-ocr, but when I set pixelFormat to yuv app crashe with camera error 4 on my Samsung device.

2sem, 2023-10-07:
I had to just cherry-pick the fix into 2.15.

edritech93, 2023-10-11:
same issue for me, iOS 17 and Xcode 15

IvanByRool, 2023-10-13:
hey @mrousavy, thanks for the great work on this.

I have the same issue here, with 
reanimated : 3.5.4
vision camera : 3.3.1

And still getting "Frame processors are not enabled!"

Don't know if it's related, but trying to fix another issue updating, I followed your example and added
"#if __has_include(<VisionCamera/FrameProcessorPlugin.h>)" on top of my FrameProcessor.m file.

Any idea ? thx
Ivan

mrousavy, 2023-10-14:
Hey - hm that's weird, which react-native-worklets-core version do you have? @IvanByRool

IvanByRool, 2023-10-15:
Hey @mrousavy, 

"react-native-worklets-core": "^0.2.2",
"react-native-reanimated": "^3.5.4",

Still happening, 
I've tried the tuto as described here : https://react-native-vision-camera.com/docs/guides/frame-processors-plugins-ios
Also tried the example (which is a little bit different) here : https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/ios/Frame%20Processor%20Plugins/Example%20Swift%20Plugin/ExampleSwiftFrameProcessor.m

Both without success.
Thanks for your help,
Ivan

IvanByRool, 2023-10-16:
Don't know if it's a pre-requisite, but the problem solved itself after updating Xcode from 14.3 to 15.
If it helps someone,

mrousavy, 2023-10-16:
Awesome!

Weird issue though.

closed 2023-10-16 by mrousavy

rach-123, 2023-10-17:
I am also facing the same issue after upgrading to xcode 15. For some reason i cannot upgrade reanimated and vision-camera to v3. Could you please provide some solution for v2.

ombogdanJoinToIt, 2023-10-19:
same problem with  xcode 15

mrousavy, 2023-10-19:
Sorry, I only provide such support for V3. In V3 everything works afaik

thongquach, 2023-10-30:
> If I switch to the latest `react-native-reanimated` 3.5.4 I get this error coming from `react-native-vision-camera`:
>
> ```
>   CMake Error at CMakeLists.txt:208 (find_package):
>     Could not find a [...]

Hi @pasquale95, I encountered the exactt same rror on Android. Did you manage to fix it yet?

mssoliveira, 2023-11-08:
> > > Desde a versão 2.16.0, o react-native-reanimated 3.5+ é necessário para que os processadores de quadros sejam habilitados.
> > 
> > 
> > Se eu mudar para o `react-native-reanimated`3.5.4 mais recente, recebo este erro de `react-native-vision-camera`:
> > ```
> >   CMake Error at CMakeLists.txt:208 (find_package):
> >     Could not find a package configuration file provided by
> >     "react-native-reanimated" with any of the following names:
> >   
> >       react-native-reanimatedConfig.cmake
> >       react-native-reanimated-config.cmake
> >   
> >     Add the installation prefix of "react-native-reanimated" to
> >     CMAKE_PREFIX_PATH or set "react-native-reanimated_DIR" to a directory
> >     containing one of the above files.  If "react-native-reanimated" provides a
> >     separate development package or SDK, be sure it has been installed.
> > ```
> 
> Oi@pasquale95, encontrei exatamente o mesmo erro no Android. Você já conseguiu consertar?

Apparently the solution will be go to v3 as I'm having the same problem but I've already tried implementing v3 and I'm running into a build problem.

erickcrus, 2024-09-27:
> `system/frame-processors-unavailable: Frame Processors are not available, react-native-worklets-core is not installed!, js engine: hermes ` getting this error, even tough `react-native-worklets-core` is installed.

same here

aslamanver, 2024-10-24:
Any solution ?

utkarshatre, 2025-05-12:
hey @mrousavy I am on  "react-native-vision-camera": "^2.16.1" with  "react-native": "0.69.4", and     "react-native-reanimated": "^2.12.0", i am getting error  of  FrameProcessor are not enable while using frameProcessor(as it available property in my project) in <Camera/>.
i enable hermer engine and allow VisionCamera_enableFrameProcessor=true.
And i already use <Camera /> in many modules, but i want to use this for qr scanning, and it will be conflict with outher modules if i update vision-camera
i tried with updating reaniated to 3x and 3.X but getting error during build of  Task :react-native-reanimated:prepareBoost FAILED