margelo/react-native-vision-camera #1938: Camera.focus seems to have no effect 🐛

stevengoldberg, 2023-10-05:
### What's happening?

Calling the Camera.focus method doesn't throw an error, but it has no apparent effect. I'm not sure how to verify this other than visually — here is a video where I tap to focus in the iOS Camera app (clearly discernible effect) and then in my app with vision-camera (no discernible effect).

https://imgur.com/a/W6bqUcL

### Reproduceable Code

```tsx
export default function CameraPage() {
    const camera = useRef(null)
    const [cameraPosition, setCameraPosition] = useState('back')
    const [isCameraInitialized, setIsCameraInitialized] = useState(false)

    const physicalDevices =
        cameraPosition === 'back'
            ? [
                  'ultra-wide-angle-camera',
                  'wide-angle-camera',
                  'telephoto-camera',
              ]
            : ['wide-angle-camera']

    const device = useCameraDevice(cameraPosition, { physicalDevices })

    const format = useCameraFormat(device, [
        {
            photoAspectRatio: 4 / 3,
        },
        { photoResolution: 'max' },
        { videoAspectRatio: 4 / 3 },
        { videoResolution: 'max' },
    ])
    
    if (device == null) {
        return <LoadingView />
    }

    return (
            <CameraView
                device={device}
                cameraRef={camera}
                cameraPosition={cameraPosition}
                format={format}
            />
}

const CameraView = ({
    device,
    cameraRef,
    cameraPosition,
    format,
}) => {
    const [focusIconPosition, setFocusIconPosition] = useState({ x: 0, y: 0 })
    
    const handleTap = (position) => {
        setFocusIconPosition(position)
        cameraRef.current?.focus(position)
    }

    const tapGesture = Gesture.Tap().onStart((e) => {
        runOnJS(handleTap)({ x: e.x, y: e.y })
        focusIconOpacity.value = withTiming(1, { duration: 300 }, () => {
            focusIconOpacity.value = withDelay(1000, withTiming(0))
        })
    })

    const focusIconStyle = useAnimatedStyle(() => ({
        opacity: focusIconOpacity.value,
    }))

    return (
        <GestureDetector gesture={tapGesture}>
                    <Camera
                        style={{ flex: 1, width: windowWidth }}
                        ref={cameraRef}
                        device={device}
                        photo
                        hdr
                        enableZoomGesture
                        orientation="portrait"
                        format={format}
                        zoom={device.neutralZoom}
                    />
                </GestureDetector>
                <Animated.View
                    style={[
                        {
                            top: focusIconPosition.y - FOCUS_ICON_SIZE / 2,
                            left: focusIconPosition.x - FOCUS_ICON_SIZE / 2,
                        },
                        styles.focus,
                        focusIconStyle,
                    ]}
                >
                    <Icon
                        as={MaterialCommunityIcons}
                        name="focus-field"
                        size={16}
                    />
                </Animated.View>
        )
}

export default CameraView
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "hasTorch": true,
  "minZoom": 1,
  "hasFlash": true,
  "name": "Back Triple Camera",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
  "maxZoom": 189,
  "supportsRawCapture": false,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "neutralZoom": 2,
  "supportsLowLightBoost": false,
  "sensorOrientation": "portrait",
  "isMultiCam": true,
  "hardwareLevel": "full",
  "position": "back"
}
```

### Device

iPhone 14 Pro

### VisionCamera Version

3.2.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-06:
Hm, maybe this change is required: https://github.com/mrousavy/react-native-vision-camera/pull/1541

closed 2023-10-06 by mrousavy

Joanna-Philips, 2024-01-08:
@stevengoldberg Would you be able to share the code for the tap-to-focus functionality? :)
I am trying it with react-native-gesture-handler and have had no luck.

stevengoldberg, 2024-01-08:
> @stevengoldberg Would you be able to share the code for the tap-to-focus functionality? :) I am trying it with react-native-gesture-handler and have had no luck.

Sure — there isn't much to it!

```
const [focusIconPosition, setFocusIconPosition] = useState({ x: 0, y: 0 })
    const focusIconOpacity = useSharedValue(0)

    const handleTap = (position) => {
        setFocusIconPosition(position)
        cameraRef.current?.focus(position)
    }

  const tapGesture = Gesture.Tap().onStart((e) => {
        runOnJS(handleTap)({ x: e.x, y: e.y })
        focusIconOpacity.value = withTiming(1, { duration: 300 }, () => {
            focusIconOpacity.value = withDelay(1000, withTiming(0))
        })
    })
```

mrousavy, 2024-01-09:
Btw: I just added an example on how to focus using RNGH to the docs.

Joanna-Philips, 2024-01-09:
Thank you @stevengoldberg! I'll have to try it out. 😁

Joanna-Philips, 2024-01-09:
Awesome sauce @mrousavy! Thank you so much!

Dhinakarthi, 2025-02-28:
It will be helpfull

import React, { useRef } from 'react';
import { View, TouchableOpacity } from 'react-native';
import { Camera, useCameraDevice } from 'react-native-vision-camera';

export default function CameraScreen() {
  const device = useCameraDevice('back');
  const cameraRef = useRef<Camera>(null);

  const handleFocus = async (event: any) => {
    if (cameraRef.current) {
      const { locationX, locationY } = event.nativeEvent;
      await cameraRef.current.focus({
        x: locationX,
        y: locationY,
      });
    }
  };

  if (device == null) return <View />;

  return (
    <TouchableOpacity
      activeOpacity={1}
      onPress={handleFocus}
      style={{ flex: 1 }}
    >
      <Camera ref={cameraRef} style={{ flex: 1 }} device={device} isActive={true} />
    </TouchableOpacity>
  );
}

PR #1943 (2023-10-06): fix: Fix `focus()` on iOS

## What

Fixes `focus()` not working on iOS since it stayed in continuous auto-focus mode.

Changes are heavily inspired by https://github.com/mrousavy/react-native-vision-camera/pull/1541 from @levipro

## Changes

## Tested on

## Related issues

- Fixes #1938