margelo/react-native-vision-camera #1941: ninja: error: manifest 'build.ninja' still dirty after 100 tries 🔧

osaidfaisal12, 2023-10-06:
### How were you trying to build the app?

I tried to build my app using "npx react-native run-android" but failed. It is working fine if I uninstall worklets core library and not use frame processor. But when I use frame processor and worklets core then this issue occurs. 

### Full build logs

```tsx
info Starting JS server...
info Installing the app...

> Configure project :react-native-reanimated
Android gradle plugin: 7.4.2
Gradle: 8.0.1
WARNING:Software Components will not be created automatically for Maven publishing from Android Gradle Plugin 8.0. To opt-in to the future behavior, set the Gradle property android.disableAutomaticComponentCreation=true in the `gradle.properties` file or use the new publishing DSL.

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core found, Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[armeabi-v7a]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-vision-camera:buildCMakeDebug[armeabi-v7a]
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
C/C++: VisionCamera: Frame Processors enabled!
[... 652 lines cut ...]
```

### VisionCamera Version

3.3.1

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-06:
huh, that's weird. maybe it was the deleteCmakeCache change

mrousavy, 2023-10-06:
Does this fix your issue? https://github.com/mrousavy/react-native-vision-camera/pull/1945

james-cohen, 2023-10-06:
Exactly the same issue for me, unfortunately that patch doesn't fix it.

I previously had an identical issue when trying to build with the react-native-vision-camera package alone. The issue was due to the total file name being longer than 160 chars (I think this is a Windows limit) - the issue resolved itself when I moved the project to a parent folder. Unfortunately the explicit error message isn't shown even with verbose logging, but it is acting the same as the similar problem.

mrousavy, 2023-10-06:
ohh, so maybe this is just a Windows error. I don't know, I don't have a windows pc then. If you find a fix please let me know & submit a PR!

osaidfaisal12, 2023-10-06:
> Does this fix your issue? #1945

The error disappear. but the app is crashing and it is showing "app keeps stopping"  and the below description.

com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
	at com.mrousavy.camera.core.VideoPipeline.init$lambda$0(VideoPipeline.kt:107)
	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
	at android.media.ImageReader$1.run(ImageReader.java:837)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:211)
	at android.os.Looper.loop(Looper.java:300)
	at android.os.HandlerThread.run(HandlerThread.java:67)

james-cohen, 2023-10-06:
> ohh, so maybe this is just a Windows error. I don't know, I don't have a windows pc then. If you find a fix please let me know & submit a PR!

Yes have just built and installed the app on Mac without a problem, so seems to be a Windows specific issue

osaidfaisal12, 2023-10-06:
> @osaidfaisal12 did you install react-native-worklets-core accordingly? Including the babel plugin, then restart metro with clean cache?
> 
> If yes, try upgrading to latest Reanimated.

Tried everything even download reanimated and latest version of worklets and vision camera but still app is crashing.

info Opening the app on Android...
info JS server already running.
info Installing the app...
> Task :gradle-plugin:compileKotlin UP-TO-DATE
> Task :gradle-plugin:compileJava NO-SOURCE
> Task :gradle-plugin:pluginDescriptors UP-TO-DATE
> Task :gradle-plugin:processResources UP-TO-DATE
> Task :gradle-plugin:classes UP-TO-DATE
> Task :gradle-plugin:jar UP-TO-DATE
> Task :gradle-plugin:inspectClassesForKotlinIC UP-TO-DATE

> Configure project :react-native-reanimated
Android gradle plugin: 7.4.2
Gradle: 8.0.1
WARNING:Software Components will not be created automatically for Maven publishing 
from Android Gradle Plugin 8.0. To opt-in to the future behavior, set the Gradle property android.disableAutomaticComponentCreation=true in the `gradle.properties` file or use the new publishing DSL.

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core found, Frame Processors enabled!

> Task :app:buildCodegenCLI SKIPPED
> Task :app:generateCodegenSchemaFromJavaScript SKIPPED
> Task :app:generateCodegenArtifactsFromSchema SKIPPED
> Task :app:generatePackageList
> Task :app:preBuild
> Task :app:preDebugBuild
> Task :react-native-reanimated:assertLatestReactNativeWithNewArchitectureTask SKIPPED
> Task :react-native-reanimated:assertMinimalReactNativeVersionTask SKIPPED        
> Task :react-native-reanimated:assertNoMultipleInstances
> Task :react-native-reanimated:prepareHeadersForPrefab UP-TO-DATE
> Task :react-native-reanimated:preBuild
> Task :react-native-reanimated:preDebugBuild
> Task :react-native-reanimated:compileDebugAidl NO-SOURCE
> Task :react-native-vision-camera:preBuild UP-TO-DATE
> Task :react-native-vision-camera:preDebugBuild UP-TO-DATE
> Task :react-native-worklets-core:prepareHeaders UP-TO-DATE
> Task :react-native-worklets-core:preBuild UP-TO-DATE
> Task :react-native-worklets-core:preDebugBuild UP-TO-DATE
> Task :react-native-worklets-core:compileDebugAidl NO-SOURCE
> Task :react-native-vision-camera:compileDebugAidl NO-SOURCE
> Task :app:compileDebugAidl NO-SOURCE
> Task :react-native-reanimated:packageDebugRenderscript NO-SOURCE
> Task :react-native-vision-camera:packageDebugRenderscript NO-SOURCE
> Task :react-native-worklets-core:packageDebugRenderscript NO-SOURCE
> Task :app:compileDebugRenderscript NO-SOURCE
> Task :app:generateDebugBuildConfig UP-TO-DATE
> Task :app:javaPreCompileDebug UP-TO-DATE
> Task :react-native-reanimated:writeDebugAarMetadata UP-TO-DATE
> Task :react-native-vision-camera:writeDebugAarMetadata UP-TO-DATE
> Task :react-native-worklets-core:writeDebugAarMetadata UP-TO-DATE
> Task :app:checkDebugAarMetadata UP-TO-DATE
> Task :app:generateDebugResValues UP-TO-DATE
> Task :react-native-reanimated:compileDebugRenderscript NO-SOURCE
> Task :react-native-reanimated:generateDebugResValues UP-TO-DATE
> Task :react-native-reanimated:generateDebugResources UP-TO-DATE
> Task :react-native-reanimated:packageDebugResources UP-TO-DATE
> Task :react-native-vision-camera:compileDebugRenderscript NO-SOURCE
> Task :react-native-vision-camera:generateDebugResValues UP-TO-DATE
> Task :react-native-vision-camera:generateDebugResources UP-TO-DATE
> Task :react-native-vision-camera:packageDebugResources UP-TO-DATE
> Task :react-native-worklets-core:compileDebugRenderscript NO-SOURCE
> Task :react-native-worklets-core:generateDebugResValues UP-TO-DATE
> Task :react-native-worklets-core:generateDebugResources UP-TO-DATE
> Task :react-native-worklets-core:packageDebugResources UP-TO-DATE
[... 124 lines cut ...]
> Task :react-native-vision-camera:copyDebugJniLibsProjectOnly UP-TO-DATE
> Task :app:mergeDebugNativeLibs UP-TO-DATE
> Task :app:stripDebugDebugSymbols UP-TO-DATE
> Task :app:validateSigningDebug UP-TO-DATE
> Task :app:writeDebugAppMetadata UP-TO-DATE
> Task :app:writeDebugSigningConfigVersions UP-TO-DATE
> Task :app:packageDebug UP-TO-DATE
> Task :app:createDebugApkListingFileRedirect UP-TO-DATE

> Task :app:installDebug
Installing APK 'app-debug.apk' on '21061119AG - 12' for :app:debug
Installed on 1 device.

BUILD SUCCESSFUL in 55s
126 actionable tasks: 30 executed, 96 up-to-date
info Connecting to the development server...
info Starting the app on "003454670508"...
Starting: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] cmp=com.campro/.MainActivity }
 BUNDLE  ./index.js

 LOG  Running "campro" with {"rootTag":11}
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  granted
 BUNDLE  ./index.js

osaidfaisal12, 2023-10-06:
@mrousavy 
on my node terminal after "npx react-native run-android" it is showing "Frame Processor threw an error: Unexpected token ')'". Also now the app is not crashing and "Frame Processor threw an error: Unexpected token ')'" this error is running in loop. 
In package/android/build.gradle on lin 63 I am having error "unable to resolve class GradleException" and on line 22 "unable to resolve class Gradle".

zjkettell, 2023-10-10:
I believe this is just an issue on Windows, I was running into it as well while making Android builds, until I swapped to my Macbook and suddenly had no build issues.

james-cohen, 2023-10-12:
Yep also confirmed the problem goes away when you move the project to a directory with a shorter absolute file path. I've spent an unnecessary amount of time trying to find other workarounds involving CMake and ninja configuration and none of it made any difference. Ultimately it seems like an issue with ninja and Windows rather than anything specific to this project

mrousavy, 2023-10-12:
Yea this sucks. Windows paths are weird and very wanky when used with such cross-platform build systems. I think we can close this as this is not related to my library

closed 2023-10-12 by mrousavy

saquibhayat26, 2023-12-31:
> Yep also confirmed the problem goes away when you move the project to a directory with a shorter absolute file path. I've spent an unnecessary amount of time trying to find other workarounds [...]

thanks @james-cohen , this worked for me

GaneshNawghare, 2024-01-16:
Just a weird problem with Windows.
I made it work just by moving the project folder to a shorter path.
From
C:\Users\This PC\OneDrive\Desktop\Personal\AppName
to
C:\Project\AppName
@GnApollo this is worked for me. thanku so much

alidf92, 2024-02-10:
> Exactly the same issue for me, unfortunately that patch doesn't fix it.
> 
> I previously had an identical issue when trying to build with the react-native-vision-camera package alone. The issue was due to the path length of the target folder being longer than 160 chars (I think this is a Windows limit) - the issue resolved itself when I moved the project to a parent folder. Unfortunately the explicit error message isn't shown even with verbose logging.

Thank you

JmePaz, 2024-02-18:
For those who try a shorter path directory of project and still does not work.

I suffer the same experiences and even try moving in a shorter directory won't work.

What fixes my problems in Windows is syncing the time. 
[image]

Go to Date & Time > Additional Settings> Sync Now.

Even though you set the time zone automatically, still try the Sync Now button (chances are there might be delay in time)

I found the solution here: 
https://github.com/ninja-build/ninja/issues/1599

mlhtnc, 2024-02-29:
> Exactly the same issue for me, unfortunately that patch doesn't fix it.
>
> I previously had an identical issue when trying to build with the react-native-vision-camera package alone. The issue was due [...]

Thank you, I did not expect this to solve my problem but it did!

tpwatson, 2024-03-14:
> Thank you, I did not expect this to solve my problem but it did!
 Solved this issue for me too 🙏👍

xianguoGou, 2024-03-14:
> Just a weird problem with Windows. I made it work just by moving the project folder to a shorter path. From **C:\Users\This PC\OneDrive\Desktop\Personal\AppName** to **C:\Project\AppName**

has been solved, thank you

Rc85, 2024-03-22:
After a little digging, I found [this](https://github.com/ninja-build/ninja/issues/1900#issuecomment-1828258152). For someone like me who is building projects in a mono repo, I cannot just simply move directories. I'm pretty much already at the root of the drive (F:/Projects) but my RN projects live inside directories like client/mobile/<app_name> that uses imports from ./_shared and such.

Hopefully the patch in the link can be incorporated into this library.

RaguRam1991, 2024-06-25:
> Just a weird problem with Windows. I made it work just by moving the project folder to a shorter path. From **C:\Users\This PC\OneDrive\Desktop\Personal\AppName** to **C:\Project\AppName**

it worked for me.

Ali-Raza-Gill, 2024-06-25:
i also get this error , but i didn't resolved it 

> Task :react-native-reanimated:buildCMakeRelWithDebInfo[armeabi-v7a] FAILED
C/C++: ninja: error: manifest 'build.ninja' still dirty after 100 tries

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-reanimated:buildCMakeRelWithDebInfo[armeabi-v7a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `C:\Users\Admin\Desktop\Traveloacco UserProfile\travelacco-mobile-rn\node_modules\react-native-reanimated\android\.cxx\RelWithDebInfo\434gs6t3\armeabi-v7a'
  [0/2] Re-checking globbed directories...
  [1/2] Re-running CMake...
  -- Configuring done
  -- Generating done
  C++ build system [build] failed while executing:
      @echo off
      "C:\\Users\\Admin\\AppData\\Local\\Android\\Sdk\\cmake\\3.22.1\\bin\\ninja.exe" ^
        -C ^
        "C:\\Users\\Admin\\Desktop\\Traveloacco UserProfile\\travelacco-mobile-rn\\node_modules\\react-native-reanimated\\android\\.cxx\\RelWithDebInfo\\434gs6t3\\armeabi-v7a" ^
        reanimated
    from C:\Users\Admin\Desktop\Traveloacco UserProfile\travelacco-mobile-rn\node_modules\react-native-reanimated\android
  ninja: error: manifest 'build.ninja' still dirty after 100 tries

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 3m 38s

GnApollo, 2024-06-25:
C:\Users\Admin\Desktop\Traveloacco UserProfile\travelacco-mobile-rn

**Traveloacco UserProfile** contains space.

try to move your project  **travelacco-mobile-rn** in Desktop.

zanotta, 2025-03-12:
2025 and yes: it was a problem with Windows path, just move the project to something close to C:

AleVeron, 2025-07-01:
> Just a weird problem with Windows. I made it work just by moving the project folder to a shorter path. From **C:\Users\This PC\OneDrive\Desktop\Personal\AppName** to **C:\Project\AppName**

Amazing, i cant believe that error i just lost 5 hours of my life fixing it