margelo/react-native-vision-camera #1946: Drivers License data should be returned as a parsed object and not raw✨

NTKrieger, 2023-10-06:
### What feature or enhancement are you suggesting?

When you scan a barcode (in North America this is PDF417), it should return an object containing the various data fields (firstName, lastName, DOB, etc)

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I could write a parser for the raw data in the JSX layer but the parsing is already supported by the MLKit and should be implemented at the native layer.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

NTKrieger, 2023-10-06:
Also, I can't release until this is done so I'm working on it right now.

mrousavy, 2023-10-06:
Hey!

Sorry, no, this is not something VisionCamera should do in my opinion. It detects barcodes and decodes them into a string, from there on you can do whatever you want with that.

If I were to add all that kinds of data we'd end up with a really complex logic for all of that including tons of optional fields, special types, platform specific fields, etc. It's just not a generic approach anymore.

On a side note; I don't think you get the parsed data on iOS either, you'd have to start from the string value then parse it yourself.

JavaScript is most flexible for this, so I think parsing in JS makes more sense. If there's a bug in the parser, I'd have to fix it on the native side and you'd have to wait for a native change/update, whereas JS can be fixed instantly & over the air.

closed 2023-10-06 by mrousavy

mrousavy, 2023-10-06:
https://github.com/PeculiarVentures/js-zxing-pdf417

> I want to take advantage of the MLKit parser and not write one at all.

Yes, but iOS does not use MLKit.

NTKrieger, 2023-10-06:
It could, but I understand now why you don't want to implement in that way.

mrousavy, 2023-10-06:
yea thanks. I just don't want to implement very specific things, especially in this case we have a really cool stable generic solution, drilling into such specifics as drivers license will just make things more complex, both with types, providing support, fixing specific bugs in my native code, etc.

It's really easy to just get the string as it works today, and run it through some JS library that decodes it 😄