margelo/react-native-vision-camera #1947: 🐛 v3: frame processor not working with  pixelFormat yuv

danieloprado, 2023-10-06:
### What's happening?

I want to use the frameProcessor with `pixelFormat='yuv'` but the app crashes.

Bundle output:
```
 BUNDLE  ./index.js 

 LOG  Running "TracersApp" with {"rootTag":11}
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 ERROR  [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR]
```

### Reproduceable Code

```tsx
import { memo } from 'react';
import { StyleSheet } from 'react-native';
import { Camera, useCameraDevice, useFrameProcessor } from 'react-native-vision-camera';

import { useIsFocused } from '@react-navigation/native';

const CameraVision = () => {
  const backCamera = useCameraDevice('back');
  const isFocused = useIsFocused();

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    console.log('Frame format: ' + frame.pixelFormat);
  }, []);

  if (!backCamera) return;

  return (
    <Camera
      style={styles.camera}
      frameProcessor={frameProcessor}
      pixelFormat='yuv'
      onError={err => console.error(err)}
      device={backCamera}
      isActive={isFocused}
    />
  );
};

const styles = StyleSheet.create({
  camera: {
    height: '100%',
    width: '100%',
    position: 'absolute'
  }
});

export default memo(CameraVision);
```

### Relevant log output

```shell
2023-10-06 10:35:50.036 31513-31525 ExynosCameraSec         ven...amera.provider@4.0-service_64  E  Build Date is (Sep 15 2023) (05:43:51) Process 64 bit
2023-10-06 10:35:50.047 31513-31912 vendor.sam...service_64 ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(m_readEfsFile[364]):ERR :failed to open sysfs entry
2023-10-06 10:35:50.050 31513-31912 vendor.sam...service_64 ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(m_executeIoctlCommand[237]):failed to ioctl: (22) IS_CAMINFO_IOCTL_COMMAND (14 - Bad address)
2023-10-06 10:35:50.050 31513-31912 vendor.sam...service_64 ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(setCameraEfsData[137]):failed to communicate kernel: (2)
2023-10-06 10:35:50.097  1007-5221  Camera2-Fr...cessorBase cameraserver                         E  FrameProcessorBase: created
2023-10-06 10:35:50.097  1007-5221  Camera2-Fr...cessorBase cameraserver                         E  registerListener: Registering listener for frame id range 0 - 2147483647
2023-10-06 10:35:50.122  2882-2882  SecHAL                  sec...roid.hardware.nfc@1.2-service  E  nfc_hal_write: [Cmd]Send EVT data_sending_flag = true, return size=4!!
2023-10-06 10:35:50.147  2882-3967  SecHAL                  sec...roid.hardware.nfc@1.2-service  E  nfc_data_callback: [Cmd]Received a response packet : data_sending_flag = false!!
2023-10-06 10:35:50.148  2734-3932  libnfc_nci              com.android.nfc                      E  [ERROR:nfc_ncif.cc(1462)] nfc_ncif_proc_deactivate - evt_data.deactivate.reason 0, nfc_cb.deact_reason = 2
2023-10-06 10:35:50.156  2882-2882  SecHAL                  sec...roid.hardware.nfc@1.2-service  E  nfc_hal_write: [Cmd]Send EVT data_sending_flag = true, return size=a!!
[... 177 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung A53 

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-06:
Does it work if you don't set `pixelFormat='yuv'`?

danieloprado, 2023-10-06:
Yes, but MLKit does not work without YUV format.

mrousavy, 2023-10-06:
Does your `format` support `pixelFormats` YUV?

mrousavy, 2023-10-06:
If yes, this is an issue I need to investigate and spend time on. I currently have many other things to do

danieloprado, 2023-10-06:
[image]

Stills the same 😢

mrousavy, 2023-10-09:
It's always Samsung man. I don't know, the stacktrace isn't very precise, I don't have that phone myself so I can't really develop a fix for that.-

mrousavy, 2023-10-16:
Hey - @danieloprado if you want this fixed please try to figure out what the problem here is:

- Try running a native Java/Kotlin Android app that uses **Camera2** to see if that works with YUV MLKit
  - If that works, it's an issue in VisionCamera and needs to be fixed.
- Try running a native Java/Kotlin Android app that uses **CameraX** to see if that works with YUV MLKit
  - If that works, it's an issue with Samsung/Android, and CameraX has a special workaround. Browse the CameraX codebase (or contact their team) to figure out what exactly they are doing.
- Contact Samsung and report this as a bug - maybe they can help you. Say something like "Using an `ImageReader` with `ImageFormat.YUV_420_888` crashes on Samsung A58".

Also, you can try to replace `YUV_420_888` with `NV21` here: https://github.com/mrousavy/react-native-vision-camera/blob/e8ae11e30b4fb4e25a38fa589b27229c18c03171/package/android/src/main/java/com/mrousavy/camera/core/VideoPipeline.kt#L160

Maybe that format is just broken, idk.

danieloprado, 2023-10-17:
I've tested on another Samsung model (m52) and it worked, I'll close the issue for now. Thanks!

mrousavy, 2023-10-17:
Wait why are you closing the issue? Doesn't it still crash on the Samsung A53?

danieloprado, 2023-10-17:
I dont know, I dont have this phone anymore so will not able to reproduce. I'm closing just to keep in history and If anyone have this problem can open again. But If tou want I can open it again to keep track of It.

danieloprado, 2023-10-17:
I have a parent with this phone, I'll generate a release build of the new lib version and try to see If it Stills happens.

edritech93, 2023-10-18:
I have issue on the Samsung A14

meabed, 2023-10-23:
I have the same in Samsun A10, 
- if you remove the frameprocessor propert it works fine ( but no frame processor )
- if you remove the pixel format - it works, but I get PRIVATE image format in the plugin

rodgomesc, 2023-10-30:
can you guys check if this patch solves this issue?

https://github.com/mrousavy/react-native-vision-camera/pull/2108

mgcrea, 2023-12-12:
I'm encountering this as well on a OnePlus 8T as up-to-date as possible (running on Android 13), when trying to use `pixelFormat="yup"`, the device first shows a black rectangle instead of the camera and then reboots (like @kjimenezdev) after a few seconds, I'm using the latest release (3.6.15).

Here is a couple of logcat before the crashes/reboot: https://gist.github.com/mgcrea/677ad9677886c7f67c259dd9b6141238

Happy to help if there is anything I can do.

Looks like this could be the issue:
```
12-12 17:32:48.700  1190  2395 E CamX    : [ERROR][FORMAT ] camximageformatmapper.cpp:1088 ValidateBufferSize() grallocBufferSize 3112960 is less than TotalSize 3133440
12-12 17:32:48.700  1190  2395 E CamX    : [ERROR][MEMMGR ] camximagebuffer.cpp:491 Import() [RealTimeFeatureZSLPreviewRaw_IPE0_OutputPortId9_Video][0xb40000739345de00] Failed in importing Gralloc Buffer type=0, BufferHandle=0xb4000073bcdfa178, offset=0, size=3133440, flags=0x1, deviceCount=1, WxH 1920x1080, format 3
12-12 17:32:48.700  1190  2395 E CamX    : [ERROR][CORE   ] camxnode.cpp:2850 SetupRequestOutputPorts() Import failed
12-12 17:32:48.748  1190  2393 E CamX    : [ERROR][STATS_AWB] camxcawbioutil.cpp:2450 PopulateColorSensorFrontData() NCS ColorSensor front handle is NULL
12-12 17:32:48.748  1190  2393 E CamX    : [ERROR][STATS_AWB] camxcawbioutil.cpp:1766 RetrieveExtensionTriggerInfo() AI fail, aiSceneID: 0x0, aiSceneConfidence: 0x0 ,reqID: 4
```

Samuele-Barbiera, 2023-12-14:
i still have the same problem on my samsung a40

Samuele-Barbiera, 2023-12-14:
> I have the same in Samsun A10,
> 
> * if you remove the frameprocessor propert it works fine ( but no frame processor )
> * if you remove the pixel format - it works, but I get PRIVATE image format in the plugin
> 
> With pixelFormat Native it works fine, but the frameprocessor plugin cannot read the image and convert to bitmap for example.
> 
> It worth mentioning, that it's not a but in Samsung, I am currently using this package in android [natario1/CameraView](https://github.com/natario1/CameraView) with [frame processing](https://natario1.github.io/CameraView/docs/frame-processing) and it works just fine in all android devices
> 
> I am using in natario1/CameraView `ImageFormat.NV21` without issues.

how do you implement it on a react native project using this library? @meabed

danieloprado, 2024-01-15:
Testing again using the lastest release, it's working properly, may you can close this issue.

mrousavy, 2024-01-15:
Great, thanks!

closed 2024-01-15 by mrousavy