margelo/react-native-vision-camera #1949: 🐛  Crash in android when using frameprocessor

nateshmbhat, 2023-10-07:
### What's happening?

i am just using frame processor feature and trying to see the logs and the app is crashing. 

### Reproduceable Code

```tsx
const frontDevice = useCameraDevice('front');
  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    runAtTargetFps(2, () => {
      'worklet';
      console.log('frameProcessor');
    });
  }, []);

 <View
        style={{
          flex: 1,
          width: '100%',
          alignItems: 'center',
          justifyContent: 'center',
          borderColor: 'black',
          borderWidth: 2,
        }}>
          <Camera
            device={frontDevice}
            isActive
            style={StyleSheet.absoluteFill}
            ref={cameraRef}
            frameProcessor={frameProcessor}
            photo={false}
            video={false}
            audio={false}
            enableHighQualityPhotos={false}
            torch="off"
          />
      </View>
```

### Relevant log output

```shell
com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:107)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:837)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:233)
                                                                                                    	at android.os.Looper.loop(Looper.java:344)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Oneplus 7 Android 12

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

AndreaLiboni, 2023-10-09:
I have the same problem with ios

softgenicsShubham, 2023-10-09:
I also have the same issues.

Muriel-Gasparini, 2023-10-10:
same problem here, i just create a new application using the `npx react-native@latest init appName` command and i followed the installation steps according to the [https://react-native-vision-camera.com/docs/guides](url) and i have this error.

but if i try to set an pixel format different to `native` like `yuv` or `rgb` i get a black screen and the app stop crash.

In this case hermes is enabled, but i tried to use disabled and i get a syntax error inside the frameProcessor function.

my phone is a Samsung M31.

there is my dependencies below
```
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.5",
    "react-native-vision-camera": "^3.3.1",
    "react-native-worklets-core": "^0.2.2"
  }

```

chuans, 2023-10-11:
me too,I have the same problem with ios

mbfassnacht, 2023-10-11:
Hey @nateshmbhat  I was having the same issue, I did a rollback to `"react-native-worklets-core": "0.2.0"` instead of `0.2.2` . I do not know what is happening but it is useful as a workaround until the issue is fixed. From `react-native-vision-camera` you can still use the last version.

karel-suchomel-ed, 2023-10-11:
I have the same issue but only on production.
When I downgrade react-native-worklets-core to 0.2.0 the app wont even build.

"dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.5",
    "react-native-vision-camera": "^3.3.1",
    "react-native-worklets-core": "^0.2.2"
  }

FluffyDiscord, 2023-10-12:
This happens for me only when building release, the debug works fine, worklet version does not matter

zzz08900, 2023-10-12:
> This happens for me only when building release, the debug works fine, worklet version does not matter

Same issue. Debug builds works fine and release build crashes.
Are you guys using reanimated? It appears the `invalid empty parentheses '( )'` crashes are related to reanimated in some way.

mrousavy, 2023-10-16:
Anyways, this is not a VisionCamera issue. Either report at RN Worklets or at RN Reanimated. I'll try to fix it in Worklets soon

closed 2023-10-16 by mrousavy

softgenicsShubham, 2023-10-20:
> I think this is a react-native-worklets-core + react-native-reanimated
> compatibility issue. Try:
>
>    1. Try removing react-native-reanimated (or create a new blank app
>    without reanimated) and see if that works (no REA babel plugin)
>    2. Try downgrading (or upgrading) react-native-reanimated and see if
>    that works
>    3. Try downgrading react-native-worklets-core and see if that works

We've tried your suggestions, but unfortunately, the issues persists:
  Removing React Native Reanimated didn't resolve the issue.
  Upgrading/downgrading React Native Reanimated and React Native Worklets
Core didn't work either.

We're actively seeking solutions and will keep you updated, Your insights
are appreciated.