margelo/react-native-vision-camera #1951: 🔧 Example app is giving babel bundling error

nateshmbhat, 2023-10-08:
### How were you trying to build the app?

i tried running the example app. it builds fine on ios but when bundling using metro bundle, i'm getting this error : 

Node found at: /Users/natesh.bhat/.nvm/versions/node/v16.18.1/bin/node

                        ▒▒▓▓▓▓▒▒
                     ▒▓▓▓▒▒░░▒▒▓▓▓▒
                  ▒▓▓▓▓░░░▒▒▒▒░░░▓▓▓▓▒
                 ▓▓▒▒▒▓▓▓▓▓▓▓▓▓▓▓▓▒▒▒▓▓
                 ▓▓░░░░░▒▓▓▓▓▓▓▒░░░░░▓▓
                 ▓▓░░▓▓▒░░░▒▒░░░▒▓▒░░▓▓
                 ▓▓░░▓▓▓▓▓▒▒▒▒▓▓▓▓▒░░▓▓
                 ▓▓░░▓▓▓▓▓▓▓▓▓▓▓▓▓▒░░▓▓
                 ▓▓▒░░▒▒▓▓▓▓▓▓▓▓▒░░░▒▓▓
                  ▒▓▓▓▒░░░▒▓▓▒░░░▒▓▓▓▒
                     ▒▓▓▓▒░░░░▒▓▓▓▒
                        ▒▒▓▓▓▓▒▒

Error while reading cache, falling back to a full crawl:
 Error: Unable to deserialize cloned data due to invalid or unsupported version.
    at deserialize (node:v8:380:7)
    at DiskCacheManager.read (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro-file-map/src/cache/DiskCacheManager.js:53:33)
    at HasteMap.read (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro-file-map/src/index.js:356:39)
    at /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro-file-map/src/index.js:290:36
    at HasteMap.build (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro-file-map/src/index.js:341:9)
    at new DependencyGraph (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/node-haste/DependencyGraph.js:60:8)
    at new Bundler (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/Bundler.js:18:22)
    at new IncrementalBundler (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/IncrementalBundler.js:35:21)
    at new Server (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/Server.js:77:21)
    at runMetro (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/index.flow.js:59:18)
                Welcome to Metro v0.76.7
              Fast - Scalable - Integrated

r - reload the app
d - open developer menu
i - run on iOS
a - run on Android

```

transform[stderr]: Could not resolve "/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/index" in file /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/src/App.tsx.
transform[stderr]: Could not resolve "/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/index" in file /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/src/PermissionsPage.tsx.
transform[stderr]: Could not resolve "/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/index" in file /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/src/CameraPage.tsx.
transform[stderr]: Could not resolve "/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/index" in file /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/src/CameraPage.tsx.
transform[stderr]: Could not resolve "/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/index" in file /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/src/DevicesPage.tsx.
 BUNDLE  ./index.js 

error: Error: Unable to resolve module @babel/runtime/helpers/interopRequireDefault from /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/CameraError.ts: @babel/runtime/helpers/interopRequireDefault could not be found within the project or in these directories:
  ../../node_modules
  ../../../node_modules
  ../../../../../node_modules
> 1 | export type PermissionError = 'permission/microphone-permission-denied' | 'permission/camera-permission-denied'
  2 | export type ParameterError =
  3 |   | 'parameter/invalid-parameter'
  4 |   | 'parameter/unsupported-output'
    at ModuleResolver.resolveDependency (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/node-haste/DependencyGraph/ModuleResolution.js:139:15)
    at DependencyGraph.resolveDependency (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/node-haste/DependencyGraph.js:277:43)
    at Object.resolve (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/lib/transformHelpers.js:169:21)
    at Graph._resolveDependencies (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/DeltaBundler/Graph.js:473:35)
    at Graph._processModule (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/DeltaBundler/Graph.js:261:38)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async Graph._addDependency (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/DeltaBundler/Graph.js:372:20)
    at async Promise.all (index 2)
    at async Graph._processModule (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/DeltaBundler/Graph.js:322:5)
    at async Graph._addDependency (/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/node_modules/metro/src/DeltaBundler/Graph.js:372:20)
transform[stderr]: Could not resolve "/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/index" in file /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/src/frame-processors/ExamplePlugin.ts.
transform[stderr]: Could not resolve "/Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/src/index" in file /Users/natesh.bhat/Desktop/isha/react-native-vision-camera/package/example/src/views/CaptureButton.tsx.
```

[... 121 lines cut ...]
```

### VisionCamera Version

Commit id : 6e72781 (main branch)

### Target platforms

iOS, Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-14:
It builds for me.

IvanByRool, 2023-10-15:
Same here, 
xcode 14.3
iphone 13 / 16.6.1
thx

IvanByRool, 2023-10-15:
To be more precise, 
it's building, but getting this error when the app launches
Tested with xcode 15 & iphone 13 / 16.6.1
Example app (no changes)

closed 2023-10-16 by mrousavy