margelo/react-native-vision-camera #1953: 🐛 Memory leaks when calling runInJs functions from frame processor

bglgwyng, 2023-10-08:
### What's happening?

The function created by `Worklets.createRunInJsFn` causes memory leaks, and the app gets crashed eventually.
 It seems that the arguments passed are not garbage collected properly. With arguments of large size, for example, the buffer of a cropped image, we can observe this behavior more easily.

> I'm not sure if it is VisionCamera issue or react-native-worklets-core issue. As I don't know how to reproduce this behavior only with react-native-worklets-core, I'm posting it here.

### Reproduceable Code

```tsx
const send = useMemo(() => Worklets.createRunInJsFn((x: any) => {}), []);
const frameProcessor = useFrameProcessor(
  (frame) => {
    'worklet';

    // Increasing 10000 to a larger value will show faster memory usage increase
    send(Array(10000).fill(0));
  },
[send]);
```

### Relevant log output

```shell
(lldb) bt
* thread #7, queue = 'mrousavy/VisionCamera.video', stop reason = EXC_RESOURCE (RESOURCE_TYPE_MEMORY: high watermark memory limit exceeded) (limit=2098 MB)
  * frame #0: 0x0000000102345a40 AwesomeProject`std::__1::__shared_weak_count::__shared_weak_count[abi:v160006](this=0x00000002cdfebff0, __refs=0) at shared_ptr.h:178:11
    frame #1: 0x00000001028d7764 AwesomeProject`std::__1::__shared_ptr_emplace<RNWorklet::JsiWrapper, std::__1::allocator<RNWorklet::JsiWrapper>>::__shared_ptr_emplace[abi:v160006]<facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWrapper*&>(this=0x00000002cdfebff0, __a=allocator<RNWorklet::JsiWrapper> @ 0x000000016e0b81cf, __args=0x000000028033a940, __args=0x000000016e0b8418, __args=0x000000016e0b83b0) at shared_ptr.h:274:14
    frame #2: 0x00000001028d7470 AwesomeProject`std::__1::__shared_ptr_emplace<RNWorklet::JsiWrapper, std::__1::allocator<RNWorklet::JsiWrapper>>::__shared_ptr_emplace[abi:v160006]<facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWrapper*&>(this=0x00000002cdfebff0, __a=allocator<RNWorklet::JsiWrapper> @ 0x000000016e0b820f, __args=0x000000028033a940, __args=0x000000016e0b8418, __args=0x000000016e0b83b0) at shared_ptr.h:276:5
    frame #3: 0x00000001028d7368 AwesomeProject`std::__1::shared_ptr<RNWorklet::JsiWrapper> std::__1::allocate_shared[abi:v160006]<RNWorklet::JsiWrapper, std::__1::allocator<RNWorklet::JsiWrapper>, facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWrapper*&, void>(__a=0x000000016e0b82cf, __args=0x000000028033a940, __args=0x000000016e0b8418, __args=0x000000016e0b83b0) at shared_ptr.h:995:55
    frame #4: 0x00000001028d63a8 AwesomeProject`std::__1::shared_ptr<RNWorklet::JsiWrapper> std::__1::make_shared[abi:v160006]<RNWorklet::JsiWrapper, facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWrapper*&, void>(__args=0x000000028033a940, __args=0x000000016e0b8418, __args=0x000000016e0b83b0) at shared_ptr.h:1004:12
    frame #5: 0x00000001028d6094 AwesomeProject`RNWorklet::JsiWrapper::wrap(runtime=0x000000028033a940, value=0x000000016e0b8418, parent=0x00000002801d6f58) at WKTJsiWrapper.cpp:36:14
    frame #6: 0x00000001028d9c18 AwesomeProject`RNWorklet::JsiArrayWrapper::setValue(this=0x00000002801d6f58, runtime=0x000000028033a940, value=0x000000016e0b8d18) at WKTJsiArrayWrapper.h:324:11
    frame #7: 0x00000001028d62f0 AwesomeProject`RNWorklet::JsiWrapper::wrap(runtime=0x000000028033a940, value=0x000000016e0b8d18, parent=0x0000000000000000) at WKTJsiWrapper.cpp:54:11
    frame #8: 0x00000001027be714 AwesomeProject`RNWorklet::JsiWrapper::wrap(runtime=0x000000028033a940, value=0x000000016e0b8d18) at WKTJsiWrapper.h:62:12
    frame #9: 0x00000001028a4ec8 AwesomeProject`RNWorklet::ArgumentsWrapper::ArgumentsWrapper(this=0x000000016e0b88a8, runtime=0x000000028033a940, arguments=0x000000016e0b8d18, count=1) at WKTArgumentsWrapper.h:19:23
    frame #10: 0x00000001028a4d34 AwesomeProject`RNWorklet::ArgumentsWrapper::ArgumentsWrapper(this=0x000000016e0b88a8, runtime=0x000000028033a940, arguments=0x000000016e0b8d18, count=1) at WKTArgumentsWrapper.h:16:23
    frame #11: 0x00000001028b7a78 AwesomeProject`RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5::operator()(this=0x0000000283a07728, runtime=0x000000028033a940, thisValue=0x000000016e0b8c98, arguments=0x000000016e0b8d18, count=1) const at WKTJsiWorkletContext.cpp:215:22
    frame #12: 0x00000001028b79ec AwesomeProject`decltype(std::declval<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5&>()(std::declval<facebook::jsi::Runtime&>(), std::declval<facebook::jsi::Value const&>(), std::declval<facebook::jsi::Value const*>(), std::declval<unsigned long>())) std::__1::__invoke[abi:v160006]<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long>(__f=0x0000000283a07728, __args=0x000000028033a940, __args=0x000000016e0b8c98, __args=0x000000016e0b8ad8, __args=0x000000016e0b8ad0) at invoke.h:394:23
    frame #13: 0x00000001028b7974 AwesomeProject`facebook::jsi::Value std::__1::__invoke_void_return_wrapper<facebook::jsi::Value, false>::__call<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long>(__args=0x0000000283a07728, __args=0x000000028033a940, __args=0x000000016e0b8c98, __args=0x000000016e0b8ad8, __args=0x000000016e0b8ad0) at invoke.h:478:16
    frame #14: 0x00000001028b7928 AwesomeProject`std::__1::__function::__alloc_func<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5, std::__1::allocator<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5>, facebook::jsi::Value (facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)>::operator()[abi:v160006](this=0x0000000283a07728, __arg=0x000000028033a940, __arg=0x000000016e0b8c98, __arg=0x000000016e0b8ad8, __arg=0x000000016e0b8ad0) at function.h:185:16
    frame #15: 0x00000001028b6698 AwesomeProject`std::__1::__function::__func<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5, std::__1::allocator<RNWorklet::JsiWorkletContext::createCallInContext(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWorkletContext*)::$_5>, facebook::jsi::Value (facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)>::operator()(this=0x0000000283a07720, __arg=0x000000028033a940, __arg=0x000000016e0b8c98, __arg=0x000000016e0b8ad8, __arg=0x000000016e0b8ad0) at function.h:356:12
    frame #16: 0x0000000102682bc8 AwesomeProject`std::__1::__function::__value_func<facebook::jsi::Value (facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)>::operator()[abi:v160006](this=0x0000000283a076f8, __args=0x000000028033a940, __args=0x000000016e0b8c98, __args=0x000000016e0b8ad8, __args=0x000000016e0b8ad0) const at function.h:510:16
    frame #17: 0x0000000102682b34 AwesomeProject`std::__1::function<facebook::jsi::Value (facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)>::operator()(this=0x0000000283a076f8, __arg=0x000000028033a940, __arg=0x000000016e0b8c98, __arg=0x000000016e0b8d18, __arg=1) const at function.h:1156:12
    frame #18: 0x0000000102894d08 AwesomeProject`RNWorklet::JsiWorkletApi::createRunInJsFn(this=0x0000000283a076f8, runtime=0x000000028033a940, thisValue=0x000000016e0b8c98, arguments=0x000000016e0b8d18, count=1)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const at WKTJsiWorkletApi.h:90:18
    frame #19: 0x0000000102894cbc AwesomeProject`decltype(std::declval<RNWorklet::JsiWorkletApi::createRunInJsFn(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Valu [... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-14:
Interesting, nice catch! We'll try to fix that in RN Worklets soon!

chrfalch, 2023-10-23:
The problem here is that when we allocate data in a JSI object there is no information available for the Hermes garbage collector about the inner allocation - read more here: https://github.com/facebook/hermes/issues/982

I'm unsure how this will be / can be fixed when using Hermes (JSC handles this a bit differently) to be honest.

induRaju, 2024-01-29:
@chrfalch - I am also facing the same issue, If you could explain about your fix it would be really helpful.

mrousavy, 2024-02-27:
according to @bglgwyng this is still a problem. reopening for now (cc @chrfalch )

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

dev-john-nguyen, 2024-10-03:
useFrameProcessor seems to be stable for me, but experiencing frequent crashing with useSkiaFrameProcessor. I posted a discussion about it in react-native-skia https://github.com/Shopify/react-native-skia/discussions/2675 with the error log. I'm not able to yet pinpoint exactly where the issue is, but seems like gc is not properly releasing memory?

"react-native-vision-camera": "4.5.3"
"react-native-worklets-core": "1.3.3"
"react-native": "0.75.4"
"@shopify/react-native-skia": "1.3.13"

kuldip-simform, 2024-12-11:
This is happening with me. After sometime app crashes 

I am using this versions:
"react-native-vision-camera": "^4.6.1",
    "react-native-vision-camera-face-detector": "^1.7.1",
    "react-native-worklets-core": "^1.5.0",
    "vision-camera-resize-plugin": "^3.1.0"

lukachi, 2025-02-24:
Facing same issues, described it there: https://github.com/Shopify/react-native-skia/discussions/2675#discussioncomment-12300270