margelo/react-native-vision-camera #1958: ❓ Crash on using camera multi times

vahidid, 2023-10-09:
### Question

Hi
I'm using react native vision camera for capturing video and scan QR codes.
As I'm using camera in two different parts of my app, one as a selfie camera and other as the back camera, the first try for opening the camera works without any problem, but as soon as I click back and go for the second camera, the app crashes and i keep getting the error of "TOO_MANY_OPEN_CAMERAS". It seems that the first try for opening the camera does not stop when i close it. Can anyone help me with this problem?
and the weird part is that it occurs on some devices in Android. 

### VisionCamera Version

3.3.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

vahidid, 2023-10-10:
I asked ChatGPT to fix my problem. ChatGPT offers me to update the following method in `CameraView.kt`:
```
override fun onDetachedFromWindow() {
    super.onDetachedFromWindow()
    cameraSession.close()
    updateLifecycle()
  }
```

I'm not sure that is a clean way to close camera when you want to use the back and front camera.

Also i opened a PR #1962

mrousavy, 2023-10-10:
Hey, why did you open a question for this? This is clearly a bug report, you even have an error message. 

I think I know what's happening here, but I believe this is by design on Android.
You could fully unmount the `<Camera>` component when navigating away, but yea I can see that this is weird. Maybe your PR is a good solution, I need to think about this

vahidid, 2023-10-10:
> Hey, why did you open a question for this? This is clearly a bug report, you even have an error message.
>
> I think I know what's happening here, but I believe this is by design on Android. You could [...]

Sorry i wasn't sure that is a bug but after a lot of exploring i found that the problem is in native codes.
I'm not sure that is a good solution. I'm not a Kotlin developer.

mrousavy, 2023-10-10:
Maybe this fixes your issue? https://github.com/mrousavy/react-native-vision-camera/pull/1972

myeongjae-kim, 2023-10-11:
> Maybe this fixes your issue? #1972

I encountered the same issue and #1972 have fixed it.

mrousavy, 2023-10-13:
Awesome!

mlecoq, 2023-10-13:
@mrousavy I have tried to patch vision-camera with #1972 but I have another crash

``` 
FATAL EXCEPTION: mrousavy/VisionCamera.main
Process: com.azzapp.azzapp_dev, PID: 7596
android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
	at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:10871)
	at android.view.ViewRootImpl.requestTransparentRegion(ViewRootImpl.java:4455)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:8297)
	at android.view.SurfaceView.onAttachedToWindow(SurfaceView.java:368)
	at android.view.View.dispatchAttachedToWindow(View.java:22016)
	at android.view.ViewGroup.addViewInner(ViewGroup.java:6108)
	at android.view.ViewGroup.addView(ViewGroup.java:5884)
	at android.view.ViewGroup.addView(ViewGroup.java:5824)
	at android.view.ViewGroup.addView(ViewGroup.java:5796)
	at com.mrousavy.camera.CameraView.setupPreviewView(CameraView.kt:156)
```

mlecoq, 2023-10-13:
or 
``` 
FATAL EXCEPTION: mrousavy/VisionCamera.main
Process: com.azzapp.azzapp_dev, PID: 4368
android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): endConfigure:605: Camera 0: Error configuring streams: Broken pipe (-32)
	at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:1440)
	at android.hardware.camera2.impl.ICameraDeviceUserWrapper.endConfigure(ICameraDeviceUserWrapper.java:119)
	at android.hardware.camera2.impl.CameraDeviceImpl.configureStreamsChecked(CameraDeviceImpl.java:484)
	at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSessionInternal(CameraDeviceImpl.java:679)
	at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession(CameraDeviceImpl.java:644)
	at com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt.createCaptureSession(CameraDevice+createCaptureSession.kt:82)
	at com.mrousavy.camera.core.CameraSession.getCaptureSession(CameraSession.kt:438)
	at com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:532)
	at com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:49)
```

fahmiabdulyakub, 2023-10-13:
> Maybe this fixes your issue? #1972

How to use this one ?

myeongjae-kim, 2023-10-16:
> > Maybe this fixes your issue? #1972
> 
> How to use this one ?

@fahmiabdulyakub Try to use https://github.com/ds300/patch-package to patch library locally.

fahmiabdulyakub, 2023-10-16:
> > > Maybe this fixes your issue? #1972
> > 
> > 
> > How to use this one ?
> 
> @fahmiabdulyakub Try to use https://github.com/ds300/patch-package to patch libraries locally.

I've create the patch, but I get this error when I running project
[react-native-vision-camera+3.3.1.patch](https://github.com/mrousavy/react-native-vision-camera/files/12911436/react-native-vision-camera%2B3.3.1.patch)

react-native-vision-camera+3.3.1.patch
```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
index c4699c3..d68a805 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
@@ -13,7 +13,11 @@ import android.view.Surface
 import android.widget.FrameLayout
 import androidx.core.content.ContextCompat
 import com.facebook.react.bridge.ReadableMap
+import com.facebook.react.bridge.UiThreadUtil
+import com.mrousavy.camera.core.CameraPermissionError
+import com.mrousavy.camera.core.CameraQueues
 import com.mrousavy.camera.core.CameraSession
+import com.mrousavy.camera.core.NoCameraDeviceError
 import com.mrousavy.camera.core.PreviewView
 import com.mrousavy.camera.core.outputs.CameraOutputs
 import com.mrousavy.camera.extensions.bigger
@@ -28,8 +32,8 @@ import com.mrousavy.camera.parsers.PixelFormat
 import com.mrousavy.camera.parsers.ResizeMode
 import com.mrousavy.camera.parsers.Torch
 import com.mrousavy.camera.parsers.VideoStabilizationMode
+import kotlin.coroutines.CoroutineContext
 import kotlinx.coroutines.CoroutineScope
-import kotlinx.coroutines.Dispatchers
 import kotlinx.coroutines.launch
 
 //
@@ -42,7 +46,9 @@ import kotlinx.coroutines.launch
 // TODO: takePhoto() return with jsi::Value Image reference for faster capture
 
 @SuppressLint("ClickableViewAccessibility", "ViewConstructor", "MissingPermission")
-class CameraView(context: Context) : FrameLayout(context) {
+class CameraView(context: Context) :
+  FrameLayout(context),
+  CoroutineScope {
   companion object {
     const val TAG = "CameraView"
 
@@ -104,8 +110,11 @@ class CameraView(context: Context) : FrameLayout(context) {
   internal val outputOrientation: Orientation
     get() = orientation ?: inputOrientation
 
+  override val coroutineContext: CoroutineContext = CameraQueues.cameraQueue.coroutineDispatcher
+
   init {
     this.installHierarchyFitter()
+    clipToOutline = true
     setupPreviewView()
     cameraSession = CameraSession(context, cameraManager, { invokeOnInitialized() }, { error -> invokeOnError(error) })
   }
@@ -116,12 +125,12 @@ class CameraView(context: Context) : FrameLayout(context) {
       isMounted = true
       invokeOnViewReady()
     }
-    updateLifecycle()
+    launch { updateLifecycle() }
   }
 
   override fun onDetachedFromWindow() {
[... 327 lines cut ...]
e: file:///Users/ReactNative/klikoov2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:320:50 Unresolved reference: CameraNotReadyError
e: file:///Users/ReactNative/klikoov2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:321:57 Unresolved reference: CameraNotReadyError
e: file:///Users/ReactNative/klikoov2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:345:52 Unresolved reference: CameraNotReadyError
e: file:///Users/ReactNative/klikoov2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:346:45 Unresolved reference: CameraNotReadyError
e: file:///Users/ReactNative/klikoov2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:398:8 Unresolved reference: CameraQueues
e: file:///Users/ReactNative/klikoov2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:426:8 Unresolved reference: CameraQueues
e: file:///Users/ReactNative/klikoov2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:443:50 Unresolved reference: CameraNotReadyError

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 36s
error Failed to install the app.
```

myeongjae-kim, 2023-10-16:
@mrousavy This is a patch I created: [react-native-vision-camera+3.3.1.patch](https://github.com/mrousavy/react-native-vision-camera/files/12911826/react-native-vision-camera%2B3.3.1.patch). Try this one. I diffed it with your patch and It is slightly different with your patch.

fahmiabdulyakub, 2023-10-16:
> @mrousavy This is a patch I created: [react-native-vision-camera+3.3.1.patch](https://github.com/mrousavy/react-native-vision-camera/files/12911826/react-native-vision-camera%2B3.3.1.patch). Try this one. I diffed it with your patch and It is slightly different with your patch.

It's run but I get this crash after using your patch

```
FATAL EXCEPTION: mrousavy/VisionCamera.main
Process: com.pvp.klikoo.original, PID: 29126
android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
	at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:11586)
	at android.view.ViewRootImpl.requestTransparentRegion(ViewRootImpl.java:5042)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.ViewGroup.requestTransparentRegion(ViewGroup.java:7668)
	at android.view.SurfaceView.onAttachedToWindow(SurfaceView.java:349)
	at android.view.View.dispatchAttachedToWindow(View.java:22479)
	at android.view.ViewGroup.addViewInner(ViewGroup.java:5520)
	at android.view.ViewGroup.addView(ViewGroup.java:5296)
	at android.view.ViewGroup.addView(ViewGroup.java:5236)
	at android.view.ViewGroup.addView(ViewGroup.java:5208)
	at com.mrousavy.camera.CameraView.setupPreviewView(CameraView.kt:156)
	at com.mrousavy.camera.CameraView.access$setupPreviewView(CameraView.kt:44)
	at com.mrousavy.camera.CameraView$update$1.invokeSuspend(CameraView.kt:174)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.os.HandlerThread.run(HandlerThread.java:67)
	Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@351dfa6, mrousavy/VisionCamera.main]

```

mrousavy, 2023-10-16:
This has been fixed in the latest commit. I'll release this to npm later today

edritech93, 2023-10-26:
> @mrousavy I have tried to patch vision-camera with #1972 but I have another crash
>
> ```
> FATAL EXCEPTION: mrousavy/VisionCamera.main
> Process: com.azzapp.azzapp_dev, PID: 7596 [...]

sane issue for me, I use Samsung A14

xseignard, 2023-11-09:
And still happening with:
```
react-native 0.72.6
react-native-vision-camera 3.6.4
react-native-worklets-core 0.2.2
```

xseignard, 2023-11-13:
Looking at the code closely:

- First, look at `CameraManager.openCamera`: https://developer.android.com/reference/android/hardware/camera2/CameraManager#openCamera(java.lang.String,%20android.hardware.camera2.CameraDevice.StateCallback,%20android.os.Handler)
- It throws a `CameraAccessException`: https://developer.android.com/reference/android/hardware/camera2/CameraAccessException
- The `CameraManager.openCamera` extension ([here](https://github.com/mrousavy/react-native-vision-camera/blob/main/package/android/src/main/java/com/mrousavy/camera/extensions/CameraManager%2BopenCamera.kt#L45)) derives the thrown error with `CameraDeviceError.fromCameraDeviceError`
- Looking at the code of `CameraDeviceError.fromCameraDeviceError` ([here](https://github.com/mrousavy/react-native-vision-camera/blob/dc89664be2f6f69c1d1417013d64cbd5f7b0f95a/package/android/src/main/java/com/mrousavy/camera/types/CameraDeviceError.kt#L14)) you'll see it compares with `CameraDevice.StateCallback` and not with `CameraAccessException`
- [`ERROR_MAX_CAMERAS_IN_USE`](https://developer.android.com/reference/kotlin/android/hardware/camera2/CameraDevice.StateCallback#error_max_cameras_in_use) has value 2, but the real value 2 we want is the one from `CameraAccessException` which corresponds to [CAMERA_DISCONNECTED](https://developer.android.com/reference/android/hardware/camera2/CameraAccessException#CAMERA_DISCONNECTED)

mrousavy, 2023-11-13:
@xseignard wait what? That's weird - the [Android documentation of the `onError(.., code)` method](https://developer.android.com/reference/android/hardware/camera2/CameraDevice.StateCallback#onError(android.hardware.camera2.CameraDevice,%20int)) says that we need to compare to the `CameraDevice.StateCallback.____` error codes:
[image]

mrousavy, 2023-11-13:
Maybe `ERROR_MAX_CAMERAS_IN_USE` means we didn't properly close the previous camera before opening a new one? Not sure but I think the way we read the error here is correct, at least that's what the docs say.

xseignard, 2023-11-13:
MMM, you're right, sorry for the confusion. That's weird they transform error codes in between the call to `openCamera` and the `onError` handler 🤔

dante-cervantes-rocketlab, 2023-11-15:
im still getting this issue with latest version, any updates?

mrousavy, 2024-01-15:
This is related to #2378 (new issue with logs)

closed 2024-01-15 by mrousavy