margelo/react-native-vision-camera #1959: 🐛 ReanimatedError: Property '__scanFaces' doesn't exist, js engine: reanimated

michelorth, 2023-10-09:
### What's happening?

When activating the camera using

frameProcessor={frameProcessor}

Gives the following error: ReanimatedError: Property '__scanFaces' doesn't exist, js engine: reanimated

if I remove the frameProcessor it works

My dependencies are:

"react-native-worklets-core": "^0.2.2",
"vision-camera-face-detector": "^0.1.8",
"react-native-vision-camera": "^2.16.2",
"react-native-reanimated": "^3.5.3",

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const scannedFaces = scanFaces(frame);
    runOnJS(setFaces)(scannedFaces);
  }, []);

<Camera
        ref={camera}
        style={StyleSheet.absoluteFill}
        onInitialized={onInitialized}
        frameProcessor={frameProcessor}
        frameProcessorFps={5}
        takePhoto={() => takePhoto(props.isConnectedInternet)}
        id="camera"
        preset="medium"
        device={device}
        isActive={isActive}
        enableZoomGesture={false}
        orientation="portrait"
        photo
        zoom={0}
      />
```

### Relevant log output

```shell
ReanimatedError: Property '__scanFaces' doesn't exist, js engine: reanimated
```

### Camera Device

```json
{
  "minZoom": 1,
  "isMultiCam": false,
  "supportsLowLightBoost": false,
  "supportsDepthCapture": false,
  "neutralZoom": 1,
  "supportsRawCapture": false,
  "supportsFocus": false,
  "position": "front",
  "devices": [
    "wide-angle-camera"
  ],
  "hasTorch": false,
  "supportsParallelVideoProcessing": true,
  "hasFlash": true,
  "name": "Front Camera",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
  "maxZoom": 189
}
```

### Device

Iphone 12 - 17.0.3

### VisionCamera Version

2.16.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sacha-c, 2023-10-10:
I get a similar issue after updating to v2.16.2 when trying to use `vision-camera-code-scanner` for qr code scanning. The error I get is `__scancodes` does not exist. Only occurs on iOS

I've tried updating to v3 and using Frame Processors there, but looks like the minimum API version for android is 29 when using frame processors which unfortunately too much of a leap for my project

ronix1020, 2023-10-10:
RN Face Detector also need a update, any alternative?

nhatnguyen95, 2023-10-11:
I have same issues, It works on previous versions but not on "react-native-vision-camera": "2.16.2"

Spikus, 2023-10-11:
It's looks' with updated `react-native-reanimated ` globals in plugin not works anymore. 
Same issue for `__scanOCR` https://github.com/aarongrider/vision-camera-ocr

Can't update to V3 because no working OCR plugin for v3

suwu150, 2023-10-12:
> It's looks' with updated `react-native-reanimated ` globals in plugin not works anymore. Same issue for `__scanOCR` https://github.com/aarongrider/vision-camera-ocr
> 
> Can't update to V3 because no working OCR plugin for v3

https://github.com/mrousavy/react-native-vision-camera/issues/1987

When I introduced these plugins, I encountered a compilation error

Simoon-F, 2023-10-13:
same.

```sh
 "react-native-reanimated": "^3.5.4",
 "react-native-vision-camera": "^2.16.2",
```

sommcz, 2023-10-13:
To test dependency on Xcode15 I've downgraded my Xcode to 14.3.1 (latest 14) and check app with following versions:

This version has no issue:
 "react-native-reanimated": "3.5.4",
 "react-native-vision-camera": "2.16.1" and  "react-native-vision-camera": "2.16.0",

This version has issue using both Xcode14 and 15.
 "react-native-reanimated": "3.5.4",
 "react-native-vision-camera": "2.16.2"

Simoon-F, 2023-10-14:
I plan to upgrade the package 'react-native-vision-camera' to '3.3.1 'and use the new version's built-in frame processor, which will reduce the time I have to deal with many problems.

In fact, I have already replaced it, and the effect is very good at present.

mrousavy, 2023-10-16:
So this is happening because the face-detector hasn't upgraded to V3 yet. Wait for the plugin from Aaron to upgrade to V3, then try again - there is a PR with that.

Not a VisionCamera issue

closed 2023-10-16 by mrousavy

abdulRafay2003, 2024-01-17:
any solution for this?.This issue still exist