margelo/react-native-vision-camera #1964: 🐛 [iOS] Adding border around Camera's direct parent causes offset preview

High5Apps, 2023-10-10:
### What's happening?

Using `react-native-vision-camera@3.3.1` on an iPhone 13 Mini, adding a border to the direct parent of a `Camera` component causes the preview to be offset and slightly smaller than it should be, despite the `Camera` having a `style` of `StyleSheet.absolutefill` (see first screenshot). Setting `borderWidth: 0` prevents the issue (see second screenshot). I believe the offset size is related to the `borderWidth`, since doubling the `borderWith` seems to double the offset (see third screenshot). This only happens on my iPhone 13 Mini, not on my Android S10, leading me to believe that this is an iOS-only issue.

Here's a **workaround** for others hitting this issue (see fourth screenshot). Just add an additional `View` component without a border as the direct child of your bordered component, and as direct parent of the `Camera`.
```javascript
<View style={{ borderWidth: 8, borderColor: 'red', ... }}>
  {/* Add the view below as a workaround */}
  <View style={StyleSheet.absoluteFill}>
    <Camera ... />
  </View>
</View>
```

With `borderWidth: 8`:
[image]

With `borderWidth: 0`:
[image]

With `borderWidth: 16`:
[image]

With `borderWith: 8` and workaround:
[image]

### Reproduceable Code

```tsx
// Modifying the 3.3.1 example app CameraPage.ts line 172
// https://github.com/mrousavy/react-native-vision-camera/blob/a498144fd5786c6dba1d78730d362e0f675cbaf7/package/example/src/CameraPage.tsx#L172
// The 2 Views are added, the ReanimatedCamera is unchanged
// Note that in order to get the 3.3.1 example app to build, I needed to comment out the line 289 in CameraView+AVCaptureSession.swift
// https://github.com/mrousavy/react-native-vision-camera/blob/a498144fd5786c6dba1d78730d362e0f675cbaf7/package/ios/CameraView%2BAVCaptureSession.swift#L289

<View style={{ flex: 1, justifyContent: 'center' }}>
  <View style={{ alignSelf: 'center', borderWidth: 8, borderColor: 'red', height: 150, width: 150 }}>
    <ReanimatedCamera ... />
  </View>
</View>
```

### Relevant log output

```shell
No relevant log outputs or errors, just a style issue
```

### Camera Device

```json
{
  "maxZoom": 121.875,
  "minZoom": 1,
  "hasTorch": true,
  "isMultiCam": false,
  "hardwareLevel": "full",
  "position": "back",
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsRawCapture": false,
  "supportsFocus": true,
  "neutralZoom": 1,
  "name": "Back Camera",
  "hasFlash": true,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "supportsLowLightBoost": false,
  "sensorOrientation": "portrait",
  "formats": []
}
```

### Device

iPhone 13 Mini (iOS 16.6.1)

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-10:
Lol this is an interesting issue

davidredshaw, 2023-11-10:
I have a similar issue with 3.6.4 on an iPhone XS Max

Page Structure:

It's a flexbox layout - the header and controls are 75 pixels high and the camera is 100% with flexShrink = 1, so I expect it to occupy the full space between the header above and controls below:

[image]

Camera Function

[image]

Stylesheet

[image]

Result: the magenta background is the camera background

[image]

If I change the Camera Function as follows, wrapping the camera in two containers (AbsoluteFill is needed on the camera and the immediate parent):

[image]

The display is as expected:

[image]

ClaireL-spindance, 2024-04-10:
@davidredshaw I had the same issue and was able to fix it by wrapping the Camera component in a View with flex 1. Try changing it to
```
<View style={{flex: 1, backgroundColor: 'green`}}>
    <Camera style={{flex:1}} ... />
</View>
```

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

High5Apps, 2024-04-23:
Hey Marc- congrats on the v4 release, and thanks for all of your great work on this library!

After upgrading my app from `react-native-vision-camera@3.9.2` to `react-native-vision-camera@4.0.1`, I can confirm that the general issue of borders causing an offset camera preview remains, although the details of the issue and workaround have changed.

**I would recommend reopening this issue.**

On iOS with `v4`, the extra-view workaround mentioned above still works fine. Without the workaround, the preview offset still occurs (see screenshot 1 below).

On Android with `v4`, I'm now seeing similar offset preview issues (see screenshot 2 below). Recall that in `v3` this seemed like an iOS-only issue. Interestingly, the offset on Android is in the opposite direction as on iOS. On iOS the camera preview is offset to the bottom right, but on Android it is offset to the top left (compare screenshots 1 and 2 below)

Furthermore, the extra-view workaround mentioned above wasn't sufficient on Android. Instead I now need to wait for the `Camera` to be `initialized` before setting the `absoluteFill` style.

I tried to see if the waiting-workaround would be enough to prevent the issue on iOS too, but it still needed the extra-view workaround. Fortunately, the new combined workaround seems to be working on both iOS and Android. I tested this on an iPhone 13 mini and a Samsung Galaxy S10.

## New workaround for v4

```jsx
function MyBorderedCamera() {
  const [cameraInitialized, setCameraInitialized] = useState(false);

  // ...

  return (
    <View style={{ borderWidth: 8, borderColor: 'red', ... }}>
      {/* Add the View below as a workaround on iOS */}
      <View style={StyleSheet.absoluteFill}>
        <Camera
          // ...

          // Wait for the camera to be initialized as a workaround on Android
          onInitialized={() => setCameraInitialized(true)}
          style={cameraInitialized && StyleSheet.absoluteFillObject}
        />
      </View>
    </View>
  );
}
```

### Screenshot 1: On iPhone 13 Mini running v4 (without the workaround)
[image]

### Screenshot 2: On Samsung Galaxy S10 running v4 (without the workaround)
[image]

High5Apps, 2025-03-12:
## New workaround for v4.6.4 and React Native 77 (New Architecture)

I recently updated to the versions mentioned above and started seeing an issue with the previous workaround where the iOS camera preview was blank and all white. However even with the blank camera preview, the actual scanning was still working.

It seems that the previous Android `cameraInitialized` workaround in `style` was causing the new issue on iOS. I tried to completely remove the `cameraInitialized` workaround, but then the original offset border issue reappeared 1 out of 3 times or so on Android, so it couldn't be removed completely. So the new workaround adds the `Platform.OS` check to keep things working on both platforms.

Additionally, it seems that the iOS `View` workaround from above is no longer needed. Even without it, the offset border issue didn't happen across 20ish tries, so I removed it from the new workaround below.

```jsx
function MyBorderedCamera() {
  const [cameraInitialized, setCameraInitialized] = useState(false);

  // ...

  return (
    <View style={{ borderWidth: 8, borderColor: 'purple', ... }}>
      <Camera
        // ...

        // Wait for the camera to be initialized as a workaround on Android
        onInitialized={() => setCameraInitialized(true)}
        style={(cameraInitialized || Platform.OS !== 'android') && StyleSheet.absoluteFill}
      />
    </View>
  );
}
```