margelo/react-native-vision-camera #1965: 🔧 Error: Exception in HostFunction: system/frame-processors-unavailable: Frame Processors are disabled!, js engine: hermes

KraljSamo, 2023-10-10:
### How were you trying to build the app?

I would like to use useFrameProcessor on Android however as soon as add this, I get the 

`Error: Exception in HostFunction: system/frame-processors-unavailable: Frame Processors are disabled!, js engine: hermes`

This only happens on Android. I am using POCO F3 phone. 

### Full build logs

```tsx
> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core found, Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[armeabi-v7a]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[x86]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[x86_64]
C/C++: VisionCamera: Frame Processors enabled!

> Task :app:installDebug
Installing APK 'app-debug.apk' on 'M2012K11AG - 13' for :app:debug
Installed on 1 device.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/8.0.1/userguide/command_line_interface.html#sec:command_line_warnings

BUILD SUCCESSFUL in 15s
482 actionable tasks: 20 executed, 462 up-to-date
info Connecting to the development server...
info Starting the app on "ba8478a5"...
Starting: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] cmp=com.testapp/.MainActivity }
✨  Done in 23.36s.
```

### Project dependencies

```json
"dependencies": {
    "@react-native-async-storage/async-storage": "^1.19.3",
    "@react-native-community/masked-view": "^0.1.11",
    "@react-native-community/netinfo": "^9.4.1",
    "@react-native-community/slider": "^4.4.3",
    "@react-native-firebase/analytics": "^18.5.0",
    "@react-native-firebase/app": "^18.5.0",
    "@react-native-firebase/crashlytics": "^18.5.0",
    "@react-native-firebase/remote-config": "^18.5.0",
    "@react-navigation/native": "^6.1.8",
    "@react-navigation/native-stack": "^6.9.14",
    "@reduxjs/toolkit": "^1.9.7",
    "accounting": "^0.4.1",
    "axios": "^1.5.1",
    "dayjs": "^1.11.10",
    "haversine-distance": "^1.2.1",
    "react": "18.2.0",
    "react-hook-form": "^7.47.0",
    "react-native": "0.72.5",
    "react-native-autocomplete-input": "^5.3.2",
    "react-native-barcode-mask": "^1.2.4",
    "react-native-compressor": "^1.8.12",
    "react-native-device-info": "^10.11.0",
    "react-native-document-scanner-plugin": "^0.9.1",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.13.1",
    "react-native-get-location": "^4.0.0",
    "react-native-keychain": "^8.1.2",
    "react-native-permissions": "^3.9.3",
    "react-native-progress": "^5.0.0",
    "react-native-root-toast": "^3.5.1",
    "react-native-safe-area-context": "^4.7.2",
    "react-native-screens": "^3.25.0",
    "react-native-sound": "^0.11.2",
    "react-native-splash-screen": "^3.3.0",
    "react-native-svg": "^13.14.0",
    "react-native-vector-icons": "^10.0.0",
    "react-native-vision-camera": "^3.3.1",
    "react-native-worklets-core": "^0.2.2",
    "react-redux": "^8.1.3",
    "redux-persist": "^6.0.0",
    "redux-saga": "^1.2.3"
  },
```

### VisionCamera Version

3.3.1

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-10:
What? You created a build error, and your logs say: `BUILD SUCCESSFUL`?

closed 2023-10-10 by mrousavy

FluffyDiscord, 2023-10-11:
No idea why this was closed suddenly. App builds successfully and says that frame processors are enabled, but then throws an console error when I try to use one. This is definitely a issue.

[image]

```log
> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-worklets-core:buildCMakeDebug[armeabi-v7a]
C/C++: ninja: Entering directory `/var/www/app/node_modules/react-native-worklets-core/android/.cxx/Debug/54161e6c/armeabi-v7a'
C/C++: In file included from /var/www/app/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:14:
C/C++: /var/www/app/node_modules/react-native-worklets-core/android/../cpp/decorators/WKTJsiSetImmediateDecorator.h:85:53: warning: format specifies type 'unsigned long' but the argument has type 'size_t' (aka 'unsigned int') [-Wformat]

> Task :react-native-worklets-core:buildCMakeDebug[x86]
C/C++: ninja: Entering directory `/var/www/app/node_modules/react-native-worklets-core/android/.cxx/Debug/54161e6c/x86'
C/C++: In file included from /var/www/app/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:14:
C/C++: /var/www/app/node_modules/react-native-worklets-core/android/../cpp/decorators/WKTJsiSetImmediateDecorator.h:85:53: warning: format specifies type 'unsigned long' but the argument has type 'size_t' (aka 'unsigned int') [-Wformat]

> Task :react-native-vision-camera:configureCMakeDebug[armeabi-v7a]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[x86]
C/C++: VisionCamera: Frame Processors enabled!

> Task :react-native-vision-camera:configureCMakeDebug[x86_64]
C/C++: VisionCamera: Frame Processors enabled!
```

mrousavy, 2023-10-12:
> No idea why this was closed suddenly

Because it is the wrong issue template. It is a build error, but it should've been a bug. Those issue templates exist for a reason, which is that I have the needed information to be able to help you (e.g. logs).

> App builds successfully and says that frame processors are enabled, but then throws an console error when I try to use one. This is definitely an issue.

Then it should be a bug report.

Either way, that specific error is thrown when the build flag `VISION_CAMERA_ENABLE_FRAME_PROCESSORS` is `false` (or not set), so I wonder why that is happening. 

This is where the logs appear in your build:

https://github.com/mrousavy/react-native-vision-camera/blob/54871022f43f7530bdaade67ca52f36e416d0c67/package/android/build.gradle#L68-L69

And this is where I pass that to C++:

https://github.com/mrousavy/react-native-vision-camera/blob/54871022f43f7530bdaade67ca52f36e416d0c67/package/android/build.gradle#L101-L107

Then this will be used as a C++ preprocessor flag:

https://github.com/mrousavy/react-native-vision-camera/blob/54871022f43f7530bdaade67ca52f36e416d0c67/package/android/CMakeLists.txt#L15

I don't see any way this could go wrong except if somehow the caches go out of sync. Maybe try completely clearing your caches including the `.cxx` and `build` folders

FluffyDiscord, 2023-10-12:
I had react-native 0.71. After updating to 0.72.5, I no longer get that error.

maslovay, 2023-10-16:
i have this issue even on react-native 0.72.5

KraljSamo, 2023-10-18:
Yes my bad for opening the wrong type of issue. 

So I have attempted everything, clean build, clearing cache, the issue is still here. 
Updating to version 3.4.0 also did not solve the issue. 

I have also attempted to build the project with the same react native version from scratch and there it works fine. My app was also recently updated to RN 0.72.5 so the configuration files should be identical, however there I get the above error.

mrousavy, 2023-10-19:
If it logs "VisionCamera: Frame Processors enabled", then it should compile in those Frame Processor logic.

Are you all 100% sure you did a clean and rebuild of the project? Maybe the C++ sources are still cached.

KraljSamo, 2023-10-19:
Cleared all in react-native-clean-project script. 
Also ran 
```
rm -rf node_modules
watchman watch-del-all
rm -rf $TMPDIR/react-native-packager-cache-*
rm -rf $TMPDIR/metro-bundler-cache-* 
yarn cache clean && yarn install 
yarn start --reset-cache
```

Issue still persist :S

KraljSamo, 2023-10-19:
Maybe it helps that I also need to apply this patch every time I clean install the app.  
https://github.com/mrousavy/react-native-vision-camera/issues/1770

`set(ENABLE_FRAME_PROCESSORS ON)`

mrousavy, 2023-10-19:
I don't get how this can be off when it even logs that it's on...