margelo/react-native-vision-camera #1966: 🐛 Black screen when swiping back with react navigation

DeluxeOwl, 2023-10-10:
### What's happening?

The Camera goes into a black screen when swiping to go back to a previous screen.

Here is a video

https://github.com/mrousavy/react-native-vision-camera/assets/54327698/48154a19-b97b-4339-951f-e8ad299cc323

### Reproduceable Code

```tsx
export default function CameraScreen({ navigation, route }: CameraScreenProps) {
  const isFocused = useIsFocused();
  const appState = useAppState();
  const isActive = isFocused && appState === "active";

  const device = useCameraDevice("back");
  if (!device) {
    return (
      <Layout>
        <View className="flex-1 items-center justify-center">
          <Header>Error: No camera found.</Header>
        </View>
      </Layout>
    );
  }

  return (
    <Camera
      style={{ width: 500, height: 500 }}
      device={device}
      isActive={isActive}
    />
  );

My navigation setup:
```ts
return (
    <NavigationContainer>
      <StatusBar style="light" />
      <RootStack.Navigator
        initialRouteName="TabScreen"
        screenOptions={{ headerShown: false }}
      >
        <RootStack.Screen name="CameraScreen" component={CameraScreen} />
        <RootStack.Screen name="TabScreen" component={TabScreen} />
      </RootStack.Navigator>
    </NavigationContainer>
  );
```
And in `TabScreen`
```t
<TabNav.Navigator
      initialRouteName="HomeScreen"
      //...
    >
      <TabNav.Screen name="HomeScreen" component={HomeScreen} />
      <TabNav.Screen name="DetailsScreen" component={DetailsScreen} />
      <TabNav.Screen name="ScanScreen" component={ScanScreen} />
      <TabNav.Screen name="SkiaScreen" component={SkiaScreen} />
    </TabNav.Navigator>
```
```

### Relevant log output

```shell
10-10 15:49:07.794  5781  5781 I CameraView: Configuring Camera Device...
10-10 15:49:07.794  5781  5781 I CameraSession: Configuring Session for Camera 0...
10-10 15:49:07.794  5781  5781 I CameraOutputs: Preparing Outputs for Camera 0...
[... 402 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung S22 Ultra

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-10-10:
Hm, I think the Preview View is being nuked by Android OS and I don't have enough time to properly stop the Capture Session in there?

irfancini, 2023-10-14:
Same here. Got a workaround though. Adding this.

```javascript
if (Platform.OS === 'android' && !isFocused) {
    return <View />;
}
```

DeluxeOwl, 2023-10-14:
@irfancini's answer prompted me to try again with `isFocused`. This actually lead me to the fact that `useIsFocused` returns `true` all the time when navigating back (wth...) . 

I've opened an issue for `react-navigation` with this information.

For now, I replaced the `DetailsScreen` in my example with the `CameraScreen` (inside the tab navigator) and it works properly. 

Not sure if there is an actual bug as well in vision camera, @mrousavy knows better

mrousavy, 2023-10-16:
Idk yea I think this will be fixed by #1996 once I finish that, but for now I got other priorities.

Might be because the PreviewView's `onSurfaceDestroyed` is called and we quickly (and synchronously) need to destroy the Capture Session, otherwise it errors out.

closed 2023-10-24 by mrousavy

mrousavy, 2023-10-24:
I just fixed this in #2049, and released it to npm (v3.6.0) 🎉 

If you appreciate my work and dedication to fix your issues and make VisionCamera better, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to say thanks. Thanks!

helloJincheng, 2023-11-08:
I still have this issue after upgrading to 3.6.4

bruno-centanaro-qubika, 2023-11-20:
Same issue in 3.6.6

bruno-centanaro, 2023-11-20:
Same issue in 3.6.8 but only on Android, iOS works fine

crly, 2023-11-22:
> Same issue in 3.6.8 but only on Android, iOS works fine

same problem

budowski, 2023-12-16:
Happens the same with version 3.6.17 - managed to find the issue:
When the camera is mounted more than once (e.g. add a console print right before the code that returns the `Camera` element) - it doesn't work.
Quick workaround (hack) - to set the `isActive` property to true only a little after the component was mounted (tried doing it with react-navigation v6's `useFocusEffect` - still had to add a delay):

```
  const [cameraIsActive, setCameraIsActive] = useState(false);
```

```
  useEffect(() => {
    setTimeout(() => {
      setCameraIsActive(true);
    }, 100);
  }, []);
```

```
      return (
      <Camera
        isActive={cameraIsActive}
      />);
```

MohamedShawkyBayoumi, 2024-01-01:
> Happens the same with version 3.6.17 on Android - managed to find the issue: When the camera is mounted more than once (e.g. add a console print right before the code that returns the `Camera` element) - it doesn't work. Quick workaround (hack) - to set the `isActive` property to true only a little after the component was mounted (tried doing it with react-navigation v6's `useFocusEffect` - still had to add a delay):
> 
> ```
>   const [cameraIsActive, setCameraIsActive] = useState(false);
> ```
> 
> ```
>   useEffect(() => {
>     setTimeout(() => {
>       setCameraIsActive(true);
>     }, 1000);
>   }, []);
> ```
> 
> ```
>       return (
>       <Camera
>         isActive={cameraIsActive}
>       />);
> ```

Thanks, this resolves the issue

MSchmidt, 2024-01-29:
This is still happening when unmounting the component on Android. No navigation library involved. Just plain mounting and unmounting based on state. I've created a demo project here: https://github.com/MSchmidt/CameraTester

mrousavy, 2024-01-29:
Thanks for the sponsorship @MSchmidt, I'll take a look!

crly, 2024-04-09:
Same issue in 3.9.2 but only on Android, iOS works fine

mrousavy, 2024-04-09:
Can you check out if VisionCamera V4 fixed this issue?

woowalker, 2024-07-24:
@mrousavy 3.8.2 still, 4.5.0 fixed