margelo/react-native-vision-camera #1967: ❓ Autofocus v3

ErrorPro, 2023-10-10:
### Question

Since the upgrade from v2 to v3 camera view no longer auto-focuses. I've checked the doc and found a new method to focus on a specific point but in my case I can't use a gesture handler to pass it. Is there a way to have autofocus somehow? Thanks!

### VisionCamera Version

3.3.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

stevengoldberg, 2023-10-11:
It looks like this was addressed but hasn't made it into a release yet https://github.com/mrousavy/react-native-vision-camera/pull/1943

ErrorPro, 2023-10-11:
Yeah but it still seems like you would have to call the function and it's a bit different from what is mentioned in the topic. It'd be nice to not call any function and have autofocus working. I've checked on Android and it works as expected.

enthcoder, 2023-10-14:
Hi @ErrorPro 
I have encountered the same issue during Android development. 
Have you resolved this problem? 
Could you please inform me of the solution for this issue?

Thanks.

mrousavy, 2023-10-14:
Hey!

First of all, as with pretty much every question here, this should've been a bug report. Stop skipping the bug report templates, I cannot help you if you intentionally leave out details such as logs, device or format. That's what the bug report template is for!

Anyways, here's a potential fix: https://github.com/mrousavy/react-native-vision-camera/pull/1999

mrousavy, 2023-10-14:
Actually I don't know if this changes anything. I believe the default values are already continuously auto-focusing, I just ran the app on V3 (current main) and it properly auto-focuses all the time, no matter how close I move.

mrousavy, 2023-10-16:
So in V3 it should still auto-focus as normally. Idk what's going on in your case, but please open an issue if this is not working for you so we can take a closer look at this.

For me, VisionCamera Example app auto-focuses perfectly fine on my iPhone 15 Pro.

Binz1mp, 2023-10-27:
for me, it happened in ios after I get `format` in nothing but `device`.
e.g.
```js
const device = useCameraDevice(cameraPosition);
const format = useCameraFormat(device, []);
...
<Camera
  ref={cameraRef}
  device={device}
  style={{...StyleSheet.absoluteFillObject}}
  isActive={true}
  photo={true}
  orientation='portrait'
  resizeMode='cover'
  format={format}
/>
```
I solved the problem after having `fps = 60` opion in `format` like this:
```js
...
const format = useCameraFormat(device, [
  {fps : 60}
]);
...
```
This could change `Camera`'s width and height somehow, but anyway. That was may first-aid.