margelo/react-native-vision-camera #1968: ❓ fps not working on V3

alejandroTecDam, 2023-10-10:
### Question

I've created a project from scratch to test the fps feature but after save the video, it is at 30 fps. 
I'm using a Google Pixel 6, so the camera is compatible with 60 fps and I've record a video with the native camera at 60 fps and works.

### What I tried

```typescript
function App(): JSX.Element {
  const camera = useRef<Camera>(null);
  const device = useCameraDevice('back');
  const [recording, setRecording] = useState(false);

  const onPress = () => {
    if (!camera.current) return;
    if (recording) {
      camera.current.stopRecording();
    } else {
      camera.current.startRecording({
        onRecordingFinished: async video => {
          const path = video.path;
          await CameraRoll.save(`file://${path}`, {
            type: 'video',
          });
        },
        onRecordingError: error => console.error(error),
      });
    }
    setRecording(!recording);
  };

  if (device == null) {
    return (
      <View>
        <Text>No device</Text>
      </View>
    );
  } else {
    return (
      <View
        style={{
          flex: 1,
        }}>
        <Camera
          ref={camera}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          video={true}
          audio={true}
          fps={60}
        />
        <TouchableOpacity
          onPress={onPress}
          style={{
            position: 'absolute',
            bottom: 10,
            right: 10,
            padding: 20,
            backgroundColor: recording ? 'red' : 'green',
          }}>
          <Text>Start/Stop</Text>
        </TouchableOpacity>
      </View>
    );
  }
}
``` 

### VisionCamera Version

3.3.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-14:
You created a question, when this should've been a bug report.

The bug report template is explicitly there so that you fill in details that I need to help you, such as logcat logs, the device and format you selected, and more.

You simply chose to ignore this, so I can't help you.

closed 2023-10-14 by mrousavy

alejandroTecDam, 2023-10-16:
I didn't know if this was a bug or just that I didn't know how to do it