margelo/react-native-vision-camera #1970: ❓ A way to find out the resolution of the barcode scanner

r-aurejac, 2023-10-10:
### Question

Hi, I'm trying to implement the barcode scanner, and I need to know its resolution in order to convert the scanned barcodes coordinates into my screen's reference frame, and vice versa.
I've deduced that it's 720x1280 on my Pixel 4a but I doubt it's the same on all devices.

I couldn't find anything to get it automatically, so I tried playing with the video/photo resolution to set one manually, but that didn't change the coordinates of the scanned codes.

So I'd like to know if there's a way of finding out or setting it. 

Thanks to the author and contributors of this library for their superb work.

### VisionCamera Version

3.3.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

r-aurejac, 2023-10-11:
On all the Android devices I've tested, the scanner plane is in landscape mode (origin in the top right-hand corner of the screen) with coordinates ranging up to 1280 in width (x) and 720 in height (y).

On iOS, on the other hand, the scanner is in portrait mode and its dimensions correspond to the plane of the application screen (retrieved with Dimensions.get("window")), so i don't need to resize them.

KraljSamo, 2023-10-11:
One way would be to `useFrameProcessor` and use the frame to get the resolution. Atleast that is how I was handling this on the Camera V2 version. On V3 this works, but I cant make it to work on Android. I get the Frame Processors are disabled! error even though I enabled them.

r-aurejac, 2023-10-11:
@KraljSamo Yes, that's how I used to work when I used the plugin. 
What bothers me is that it requires a dependency on react-native-worklets-core and I'd like to do without it if possible.
By the way, have you configured this lib, maybe that's where your problem comes from?

KraljSamo, 2023-10-12:
I think I have. I mean it must be working fine, since code scanning works well. Just this useFrameProcessor hooks does not work on Android. On iOS it works fine. 
I tried building a completely new project and there this same hook works on Android as well.

JulienTexier, 2023-10-13:
Hi everyone,

First of all, thank you @mrousavy for the exceptional work on that library.

I've been struggling with the coordinates problem to show the frame on top of the scanned code on Android, because of the problem described by @r-aurejac. 

I came up with a solution to convert the frame on Android to show it correctly in portrait mode. This might be useful for anyone struggling to get the proper coordinates on Android.

```
const pxDiff = Math.max(PixelRatio.get() - 1, 1);

export default function QRCodeScanner({ onQRCodeScanned }: Props) {
  const device = useCameraDevice('back');
  const isFocused = useIsFocused();
  const appState = useAppState();
  const isActive = isFocused && appState === 'active';
  const dimensions = useWindowDimensions();

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: (codes) => {
      codes.forEach((qrCode) => {
        if (!qrCode.value || !qrCode.frame) return;

        if (Platform.OS === 'ios') {
          const bounds = getBoundsIOS(qrCode.frame);
          onQRCodeScanned({ qrCode: qrCode.value, bounds });
        } else {
          const bounds = getBoundsAndroid({
            frame: qrCode.frame,
            windowWidth: dimensions.width,
          });
          onQRCodeScanned({ qrCode: qrCode.value, bounds });
        }
      });
    },
  });

  if (device == null) return <NoCameraDeviceError />;

  return (
    <View style={{ flex: 1 }}>
      {/**
       * IMPORTANT: status bar needs to be hidden for android devices
       * otherwise, it will affect the origin of the coordinate system
       */}
      <StatusBar hidden={Platform.OS === 'android'} />
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={isActive}
        codeScanner={codeScanner}
      />
    </View>
  );
}

type Frame = {
  x: number;
  y: number;
  width: number;
  height: number;
};

/**
 * Android devices have a different coordinate system than iOS devices.
 * It is based on the landscape orientation (even if the device is in portrait mode).
 * So we need to translate the frame coordinates to correspond to the origin of the portrait orientation (top left corner).
 * Additionally, the frame pixel values are represented in a different scale than the screen pixel values.
 * So we need to scale the frame coordinates to match the screen pixel values.
 * @param windowWidth The width of the screen in pixels.
 * @param frame The frame coordinates of the QR code.
 */
const getBoundsAndroid = ({
  windowWidth,
  frame,
}: {
  windowWidth: number;
  frame: Frame;
}) => {
  const { x, y, width: fWidth, height: fHeight } = frame;
  const bounds = {
    top: x / pxDiff,
    left: (windowWidth * pxDiff - (y + fHeight)) / pxDiff,
    width: fWidth / pxDiff,
    height: fHeight / pxDiff,
  };

  return bounds;
};

/**
 * Everything works as expected on iOS devices.
 * iOS rules!
 * @param frame The frame coordinates of the QR code.
 */
const getBoundsIOS = (frame: Frame) => {
  return {
    top: frame.y,
    left: frame.x,
    width: frame.width,
    height: frame.height,
  };
};

```

JulienTexier, 2023-10-16:
What exactly do you mean by `format.videoWidth`/`format.videoHeight`?

Would it be the camera layout format? I've noticed in your YouTube live that you made the iOS coordinates relative to the camera view. It would definitely be an improvement to have Android doing the same thing.

KraljSamo, 2023-10-16:
Yes that would be great. So that we could just show the overlay on the correct place on the screen and not need to do math gymnastics between the view height/width and frame height/width.

mrousavy, 2023-10-16:
> Would it be the camera layout format?

Yes. That's also the coordinate system used by Frame Processors, and by the Skia integration if that'd make it into VisionCamera main one day.

> I've noticed in your YouTube live that you made the iOS coordinates relative to the camera view

Maybe helper functions could be useful for that. Converting to UI Coordinates is only possible on the UI Thread on iOS, so this would be a performance delay.

r-aurejac, 2023-10-23:
@mrousavy In V3.5.1, I've noticed that we can now set the scanner resolution using videoHeight/videoWidth as you described it, but this only works on iOS. 
On Android, whatever I do, I always get coordinates based on a 720 by 1280 resolution.

Is this the expected behaviour?

spsaucier, 2023-10-23:
@r-aurejac Your device may have only that resolution for video. Try my snippet here & see if it works out: https://github.com/mrousavy/react-native-vision-camera/issues/2014#issuecomment-1769294150

r-aurejac, 2023-10-23:
@spsaucier I don't think so, I've tested it on several more or less recent devices. Even though I've forced the video resolution to 1080*1920 or whatever with useFormat(), I can see that the codes are between 0 and 720 in y and between 0 and 1280 in x (these are approximations on my part). Whereas on iOS I can see the difference if I change the resolution.

I've tried to adapt my sources with your code snippet but unfortunately that doesn't solve my problem.

mrousavy, 2023-10-24:
Yea on Android it's always 1280x720 - I don't think we need a larger resolution for that. Maybe we can even use a smaller resolution to be more memory efficient.

r-aurejac, 2023-10-24:
@mrousavy OK, thank you very much. 
Personally, I don't have a problem with the resolution itself. I just wanted to make sure that it was this value every time, so that I don't get fooled by certain Android devices where there might have been a different resolution.
But yes, I agree with you, we probably don't need as much precision.

mrousavy, 2024-01-15:
Can you create a feature request for this instead of a question? I can create funcs to convert from one coordinate system to the other.

mrousavy, 2024-01-15:
Actually we don't need a feature request - this is already implemented in VisionCamera! The CodeScanner now has a `CodeScannerFrame` being passed alongside. https://react-native-vision-camera.com/docs/api/interfaces/CodeScannerFrame

closed 2024-01-15 by mrousavy

pke, 2024-04-07:
thanks @JulienTexier for your work. 
However I think this kind of conversion should be done inside the library and the frame coordinates should be correct and "ready to use" in the even handler, especially since it should hide the quirks of Androids coordinate system (which isn't an issue in your daily RN life).

Could this code also work with the status bar enabled?

pke, 2024-04-09:
@JulienTexier I tried to use your function in my app but the coordinates do seem to be off still. The phone is locked in Portrait and I wonder how you paint the bounds on the screen? When I try to absolute position an image on the coordinates its not over the scanned code but always off to the top right.

Win546, 2024-05-16:
same here, i'd like to transform the camera coordinates of scanned code, to  the coordinate of the phone screen, because right now they are really off, how can obtain the height and width resolution of the camera that i'm currently use ?