margelo/react-native-vision-camera #1973: ❓ [V3][CodeScanner] CodeScanner only capture bar code in center region of camera view

zzz08900, 2023-10-11:
### Question

OK here's what I'm trying to do: 
I just upgraded from RN0.69 + vision-camera 2.15.x to RN0.72 + vision-camera 3.3.1 and replaced vision-camera-code-scanner with the built-in CodeScanner of vision-camera v3.
Everything works (or mostly works), but it seems  the built-in CodeScanner of vision-camera v3 only capture bar code in center region of camera view and the height of the scanned bar code is always somewhere between 1 and 2. 
The old vision-camera-code-scanner frame processor captures barcode anywhere in the camera view and I usually get height reading ranged from 30 to 60.

Is this behavior by design and I don't need to manually filter code locations anymore, or I did something wrong and screwed things up?

### VisionCamera Version

3.3.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-11:
Hey, this probably should've been a bug report, not sure.

1. Is there anything wrong? Does it not scan any codes?
2. Which OS? iOS? Android?
3. Which phone?
4. Share logs

zzz08900, 2023-10-11:
@mrousavy
1. When I move the phone around and place the barcode to the corners of cameraView, the CodeScanner does not detect the bar code. It can only detect bar codes located closed to the center of cameraView. Not sure its by design or a bug.
2. Everything described so far was observed on iOS, haven't tried Android yet.
3. iPhone 12 with iOS 16
4. Trying to get the logs atm.

r-aurejac, 2023-10-11:
I've observed the same behaviour on my iPhone X but not on the various Android devices I've tested with vision-camera 3.3.1

zzz08900, 2023-10-11:
@r-aurejac probably because on Android vision-camera-v3 still uses MLKit while iOS there's some sort of Apple magic.

zzz08900, 2023-10-11:
When I place barcode on the corners of camera view, nothing meaningful appears on the log.
When I place barcode close to the center of the camera view, the log is filled with code content like these:

2023-10-11 16:44:13.127471+0800 [AppName][4168:1302039] [javascript] { type: 'code-128',
  frame: 
   { width: 187.76904520019892,
     x: 50.38565383851528,
     height: 2.384361356496811,
     y: 354.0796965956688 },
  value: '00004/10' }

Scroll through the logs I found another line that might be useful:

Main Thread Checker: UI API called on a background thread: -[UIView layer]
PID: 4168, TID: 1300835, Thread name: (none), Queue name: mrousavy/VisionCamera.codeScanner, QoS: 33
Backtrace: [omitted]

GabrielvonPlaten, 2023-10-11:
I can attest that this behavior only occurs on iOS. I tested the newest version on an iPhone 14 and 15 and it's the same for both. I need to put the barcode right in the center for the scanner to detect them.
This does not happen on Android.

I also get similar logs @zzz08900 provided above.

mrousavy, 2023-10-11:
The main thread thing is definitely a bug and should be fixed by yourlatest PR

zzz08900, 2023-10-11:
Manually setting regionOfInterest does not seem to help with the issue.
Furthermore, I just noticed barcodes can always be detected if placed close to vertical center of the screen. This, along with the strange height reading makes me wonder if something is wrong with the height processing of CodeScanner?

mrousavy, 2023-10-12:
Huh, that's interesting lol. Is there a bug report for that Apple bug? Are they aware of it?

zzz08900, 2023-10-12:
> Huh, that's interesting lol. Is there a bug report for that Apple bug? Are they aware of it?

No idea. I'm not really an iOS/swift guy so can't help with Apple-related problems.
One thing I did notice though was if I rotate the bar code by 90 degrees and place it on top/bottom center of the camera view, it also gets detected. So it could be that apple designed things that way.

mrousavy, 2023-10-14:
Actually could it maybe be that the CodeScanner orientation is wrong? Can you remove the code in `CameraView+Orientation.swift` right here: https://github.com/mrousavy/react-native-vision-camera/blob/37507345786b3b0f0b63831d499f6de3ba587190/package/ios/CameraView%2BOrientation.swift#L34-L43

And see if the detection results are any better?

zzz08900, 2023-10-14:
Disabling the `func updateOrientation()` didn't help. The detection results are the same as before.

Here's my procedure:
1. modify `node_modules/react-native-vision-camera/ios/CameraView+Orientation.swift`, commenting out everything in `func updateOrientation()` 
2. run `pod install`
3. use XCode 15 to rebuild the project.

GabrielvonPlaten, 2023-10-16:
I think this issue has nothing to do with the package but the OS itself.
I installed expo-camera: 13.6.0 and expo-barcone-scanner: 12.7.0 and the issue still persists. We can only scan barcodes in the center of the camera, but QR codes can be scanned anywhere. This issue does not exist in Android.

This was tested on React Native CLI Project.

mrousavy, 2023-10-16:
Great, thank you for testing @GabrielvonPlaten! Then it's an iOS bug.

closed 2023-10-16 by mrousavy