margelo/react-native-vision-camera #1974: 🔧 react-native-vision-camera:compileDebugKotlin FAILED

OPEN-9, 2023-10-11:
### How were you trying to build the app?

yarn android

### Full build logs

```tsx
> Task :react-native-vision-camera:compileDebugKotlin FAILED
e: /Users/open/Documents/www/sjlinkapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (87, 116): Unresolved reference: resolveView
e: /Users/open/Documents/www/sjlinkapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/FrameProcessorRuntimeManager.kt: (59, 76): Unresolved reference: resolveView

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.lambda$executeIfValid$1(ExecuteActionsTaskExecuter.java:200)
        at org.gradle.internal.Try$Failure.ifSuccessfulOrElse(Try.java:263)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeIfValid(ExecuteActionsTaskExecuter.java:198)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:179)
        at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:109)
        at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter.java:46)
        at org.gradle.api.internal.tasks.execution.ResolveTaskExecutionModeExecuter.execute(ResolveTaskExecutionModeExecuter.java:62)
        at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:57)
        at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:56)
        at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:36)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:77)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:55)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:200)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:195)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:75)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:153)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:62)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.lambda$call$2(DefaultBuildOperationExecutor.java:76)
        at org.gradle.internal.operations.UnmanagedBuildOperationWrapper.callWithUnmanagedSupport(UnmanagedBuildOperationWrapper.java:54)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.call(DefaultBuildOperationExecutor.java:76)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:52)
        at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:41)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:372)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:359)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:352)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:338)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.lambda$run$0(DefaultPlanExecutor.java:127)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.execute(DefaultPlanExecutor.java:191)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.executeNextNode(DefaultPlanExecutor.java:182)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:124)
        at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
        at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:48)
        at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:56)
Caused by: org.gradle.workers.internal.DefaultWorkerExecutor$WorkExecutionException: A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
        at org.gradle.workers.internal.DefaultWorkerExecutor$WorkItemExecution.waitForCompletion(DefaultWorkerExecutor.java:336)
        at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForItemsAndGatherFailures(DefaultAsyncWorkTracker.java:142)
        at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForItemsAndGatherFailures(DefaultAsyncWorkTracker.java:94)
        at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForAll(DefaultAsyncWorkTracker.java:80)
        at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForCompletion(DefaultAsyncWorkTracker.java:68)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$3.run(ExecuteActionsTaskExecuter.java:563)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:75)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:68)
[... 131 lines cut ...]
```

### VisionCamera Version

"react-native-vision-camera": "2.16.2"

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

OPEN-9, 2023-10-11:
buildscript {
    ext {
        buildToolsVersion = "29.0.3"
        // Note: Below change is necessary for pause / resume audio feature. Not for Kotlin.
        minSdkVersion = 24
        compileSdkVersion = 33
        targetSdkVersion = 33
        kotlinVersion = '1.6.20'

        ndkVersion = "23.1.7779620"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.1.0")
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlinVersion"
        classpath 'com.google.gms:google-services:4.3.10'
        classpath 'com.google.firebase:firebase-crashlytics-gradle:2.8.1'
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

OPEN-9, 2023-10-11:
### Update the above configuration and run the **yarn android** again.
`{
        buildToolsVersion = "33.0.0"
        // Note: Below change is necessary for pause / resume audio feature. Not for Kotlin.
        minSdkVersion = 26
        compileSdkVersion = 33
        targetSdkVersion = 33
        kotlinVersion = '1.7.21'
        VisionCameraCodeScanner_targetSdkVersion = 31
        VisionCameraCodeScanner_compileSdkVersion = 31

        ndkVersion = "23.1.7779620"
    }`

### During the packaging process, an error was reported.

> Task :react-native-vision-camera:compileDebugKotlin FAILED
e: /Users/open/Documents/www/sjlinkapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (87, 116): Unresolved reference: resolveView
e: /Users/open/Documents/www/sjlinkapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/FrameProcessorRuntimeManager.kt: (59, 76): Unresolved reference: resolveView

zzz08900, 2023-10-11:
Last time I tried, you need the latest version or RN to run vision-camera v2.16.2 or v3. 

I have no idea if any version of vision-camera works with RN 0.64 - probably no such version exists.

thomarnauld, 2023-10-16:
"react-native": "0.72.5",
    "react-native-reanimated": "3.5.4",
    "react-native-vision-camera": "2.16.2",
    "vision-camera-code-scanner": "^0.2.0",
    "react-native-worklets-core": "^0.2.0",

    work on release fine but to make it work on debug you have to delete .gradle before build

mrousavy, 2023-10-16:
Maybe some cache issue. Try upgrading to VisionCamera V3

closed 2023-10-16 by mrousavy

nguyendinhdoan, 2023-11-08:
I'm having the same issue

nbolender, 2023-11-09:
@nguyendinhdoan This worked for me: https://github.com/mrousavy/react-native-vision-camera/issues/1547#issuecomment-1787296748

vikramsinh-iottive, 2025-01-27:
> Task :react-native-vision-camera:compileDebugKotlin

> Task :react-native-vision-camera:compileDebugKotlin FAILED
85 actionable tasks: 80 executed, 5 up-to-date                                                                                                                       
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
e: Incompatible classes were found in dependencies. Remove them from the classpath or use '-Xskip-metadata-version-check' to suppress errors
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:8:27 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/7b40a01955689d9c36c09139156fe4bb/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:22:38 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/7b40a01955689d9c36c09139156fe4bb/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:29:7 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/7b40a01955689d9c36c09139156fe4bb/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:29:37 Class 'kotlinx.coroutines.android.HandlerDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.   
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/ea29b801ba52254bc4a97ba54bf160d1/transformed/jetified-kotlinx-coroutines-android-1.9.0.jar!/kotlinx/coroutines/android/HandlerDispatcher.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:29:37 Class 'kotlinx.coroutines.android.HandlerDispatcherKt' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0. 
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/ea29b801ba52254bc4a97ba54bf160d1/transformed/jetified-kotlinx-coroutines-android-1.9.0.jar!/kotlinx/coroutines/android/HandlerDispatcherKt.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:30:18 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/7b40a01955689d9c36c09139156fe4bb/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:30:38 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/7b40a01955689d9c36c09139156fe4bb/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:30:38 Class 'kotlinx.coroutines.ExecutorsKt' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/7b40a01955689d9c36c09139156fe4bb/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/ExecutorsKt.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:61:24 Class 'kotlinx.coroutines.sync.Mutex' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/BAPS/.gradle/caches/transforms-3/7b40a01955689d9c36c09139156fe4bb/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/sync/Mutex.class
e: file:///D:/vikram/20025/Vikram/TTL360/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:61:24 Class 'kotlinx.coroutines.sync.MutexKt' was compiled with an incompatible vers [... cut ...]
    at async runOnAllDevices (D:\vikram\20025\Vikram\TTL360\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:85:5)
    at async Command.handleAction (D:\vikram\20025\Vikram\TTL360\node_modules\@react

Below is my package file 

  "dependencies": {
    "@react-native-firebase/app": "^21.7.1",
    "@react-native-firebase/ml": "^21.7.1",
    "react": "18.2.0",
    "react-native": "0.71.0",
    "react-native-vision-camera": "^4.6.3",
    "vision-camera-barcode-scanner": "^0.2.2"
  },

Please solve the above issue