margelo/react-native-vision-camera #1976: 🐛MI6 qrcode canner can't get any response on listener

wublack, 2023-10-11:
### What's happening?

I wanted to scan qrcode by the lib,but it did not work.on Mi6 Android 9 MIUI11.0.5.The same code run vivo phone(android 13) is ok!

### Reproduceable Code

```tsx
const codeScanner = useCodeScanner({
        codeTypes: ['qr'],
        onCodeScanned: (codes) => {
            if (codes.length > 0) {
                console.log(`${codes[0].value} `)
                
            } else {
                console.log('codes', codes)
            }

        }
    })
const { hasPermission, requestPermission } = useCameraPermission()
    const device = useCameraDevice('back', {
        physicalDevices: [
            'ultra-wide-angle-camera',
            'wide-angle-camera',
            'telephoto-camera'
        ]
    })
    device.formats = undefined
    console.log(JSON.stringify(device,null,2))
    const isFocused = useIsFocused();const { hasPermission, requestPermission } = useCameraPermission()
    const device = useCameraDevice('back', {
        physicalDevices: [
            'ultra-wide-angle-camera',
            'wide-angle-camera',
            'telephoto-camera'
        ]
    })
    device.formats = undefined
    console.log(JSON.stringify(device,null,2))
    const isFocused = useIsFocused();const { hasPermission, requestPermission } = useCameraPermission()
    const device = useCameraDevice('back', {
        physicalDevices: [
            'ultra-wide-angle-camera',
            'wide-angle-camera',
            'telephoto-camera'
        ]
    })
    device.formats = undefined
    console.log(JSON.stringify(device,null,2))
    const isFocused = useIsFocused();
return(
<Camera  {...props} isActive={isFocused} device={device} codeScanner={codeScanner}></Camera>
)
```

### Relevant log output

```shell
2023-10-11 15:21:58.401 10801-13716/com.* I/CameraSession: Camera became un-available: 0
2023-10-11 15:21:58.401 10801-13716/com.* I/CameraSession: Camera became available: 1
2023-10-11 15:21:58.401 10801-10801/com.* I/CameraView: Props changed: [cameraId, isActive, codeScannerOptions]
2023-10-11 15:21:58.409 10801-10801/com.* I/PreviewView: Using Preview Size 1920 x 1080.
2023-10-11 15:21:58.410 10801-10801/com.* D/PreviewView: onMeasure(0, 0)
2023-10-11 15:21:58.410 10801-10801/com.* D/PreviewView: coverSize :: 1920x1080 (0.5625), 0x0 (NaN)
2023-10-11 15:21:58.410 10801-10801/com.* D/PreviewView: Fitted dimensions set: 0x0
2023-10-11 15:21:58.410 10801-10801/com.* I/CameraView: Configuring Camera Device...
2023-10-11 15:21:58.411 10801-10801/com.* I/CameraSession: Setting Format (fps: null | videoStabilization: null | hdr: null | lowLightBoost: null)...
2023-10-11 15:21:58.413 10801-10801/com.* I/CameraSession: Setting isActive: false (isRunning: false)
2023-10-11 15:21:58.414 10801-10801/com.* I/CameraSession: Setting isActive: true (isRunning: false)
2023-10-11 15:21:58.416 10801-10801/com.* D/PreviewView: onMeasure(1080, 1848)
[... 89 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Mi6

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

wublack, 2023-10-11:
> Well I guess there's your problem 'The given Camera device (id: 0) has been disconnected'
> 
> You only got one camera so try just use that one.

> Well I guess there's your problem好吧，我想这就是你的问题 'The given Camera device (id: 0) has been disconnected' “给定的相机设备（id：0）已断开连接”
> 
> You only got one camera so try just use that one.您只有一台相机，所以尝试只使用那一台。

> Well I guess there's your problem 'The given Camera device (id: 0) has been disconnected'
> 
> You only got one camera so try just use that one.
 NO！I don't think so! When i use this API " useCameraDevices().length" get camera devices ,return two

wublack, 2023-10-11:
> Well I guess there's your problem 'The given Camera device (id: 0) has been disconnected'
> 
> You only got one camera so try just use that one.

NO！I don't think so! When i use this API " useCameraDevices().length" get camera devices ,return two

mrousavy, 2023-10-11:
Did you [Setup the CodeScanner](https://react-native-vision-camera.com/docs/guides/code-scanning#setup) properly? It needs google play services to download the MLKit model.

mrousavy, 2023-10-11:
Try changing this line in `node_modules/react-native-vision-camera/android/build.gradle`:

https://github.com/mrousavy/react-native-vision-camera/blob/18e69269d49f38de11693cbec61ee9e280a1f6e9/package/android/build.gradle#L147

to 

```
implementation 'com.google.mlkit:barcode-scanning:17.2.0'
```

then rebuild your app and see if that works then.

wublack, 2023-10-12:
> Did you [Setup the CodeScanner](https://react-native-vision-camera.com/docs/guides/code-scanning#setup) properly? It needs google play services to download the MLKit model.

Thanks! I have configured！It don't work!

wublack, 2023-10-12:
> To anyone who has no idea what @mrousavy is suggesting:
> 
> > Did you [Setup the CodeScanner](https://react-native-vision-camera.com/docs/guides/code-scanning#setup) properly? It needs google play services to download the MLKit model.
> 
> I think by default Mi6 does not have Google Play services installed/enabled so the it needs to an apk with preloaded models otherwise MLKit won't work.

only Mi6 has this problem,other phone is work(vivo phone).

mrousavy, 2023-10-14:
They're doing nothing really different than I am as far as I can tell - I#m just using a newer version of the library. https://github.com/react-native-camera/react-native-camera/blob/master/android/src/mlkit/java/org/reactnative/barcodedetector/RNBarcodeDetector.java

@wublack are you 100% sure that you changed the line in the `build.gradle` file, synced your project (gradle sync in Android Studio), and then rebuilt the app and it still didn't work?

wublack, 2023-10-16:
I am pretty sure !  I remove the "node_module" folder ,try again it's didn't work! If the build.gradle file is not config, error listener will also prompt, right! But there is no prompt in the log
[image]

mrousavy, 2023-10-16:
So I guess MLKit just doesn't work on that phone then? I don't know, doesn't seem like there's anything I can do here.

I could use ZXing instead of MLKit on Android, but I heard that's slower.

@wublack you can try to build a native Java Android app and try if MLKit works on that phone by [following this guide](https://developers.google.com/ml-kit/vision/barcode-scanning/android).

wublack, 2023-10-17:
> So I guess MLKit just doesn't work on that phone then? I don't know, doesn't seem like there's anything I can do here.
>
> I could use ZXing instead of MLKit on Android, but I heard that's slower. [...]

Thanks! I try it，but it's worked on Mi6 ,using [camerax+mlkit](https://github.com/wublack/Mlkitdemo.git) ,I feel very frustrated !

wublack, 2023-10-17:
Thanks! It's my code bug ! `const isFocused = useIsFocused(); `   the Bugs will appear when i using constant isFocused. The components has not refresh!

closed 2023-10-17 by wublack

mrousavy, 2023-10-17:
🙃

hengkx, 2023-10-17:
> Try changing this line in `node_modules/react-native-vision-camera/android/build.gradle`: [...]

Thank you very much. The QR code can be recognized, but the screen will appear black when back