margelo/react-native-vision-camera #1977: 🔧 [VisionCamera] react-native-worklets-core not found, Frame Processors disabled!

caolong0204, 2023-10-11:
### How were you trying to build the app?

Everything worked well after i upgrade version lib react-native-vision-camera to 3.3.1
I did new implementing way and ios run ok but i meet error in android

 I tried build with command yarn android

### Full build logs

```tsx
> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core not found, Frame Processors disabled!

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.3.3/userguide/command_line_interface.html#sec:command_line_warnings
6 actionable tasks: 6 up-to-date
Warning: This version only understands SDK XML versions up to 2 but an SDK XML file of version 3 was encountered. This can happen if you use versions of Android Studio and the command-line tools that were released at different times.
Warning: unexpected element (uri:"", local:"extension-level"). Expected elements are <{}codename>,<{}layoutlib>,<{}api-level>
Warning: unexpected element (uri:"", local:"base-extension"). Expected elements are <{}codename>,<{}layoutlib>,<{}api-level>

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:mergeDevelopmentDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:developmentDebugRuntimeClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Searched in the following locations:
       - file:/Users/tanaakkhcm/Documents/work/projects/Uqey/mobile/cloneRentCar/node_modules/react-native/android/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/tanaakkhcm/Documents/work/projects/Uqey/mobile/cloneRentCar/node_modules/jsc-android/dist/com/facebook/react/react-android/maven-metadata.xml
       - https://dl.google.com/dl/android/maven2/com/facebook/react/react-android/maven-metadata.xml
       - https://www.jitpack.io/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/DigiKey0001-01/FREEKEY-DK-SDK_SDX_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/DigiKey0001-01/FREEKEY-DK-SDK_PRD_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/DigiKey0001-01/DK-SDK_SDX_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/DigiKey0001-01/DK-SDK_PRD_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/TOKAIRIKA-QEYBASE-SDK/CRYPTO-SDK-SDX_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/TOKAIRIKA-QEYBASE-SDK/QEYBASE-SDK-SDX_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/TOKAIRIKA-QEYBASE-SDK/QEYBASE-SDK-PRD_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/HITSERIES-FREEKEY-CRYPTO/QEYBASE-SDK-STG_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - https://raw.githubusercontent.com/DigiKey0001-01/FREEKEY-DK-SDK_SDX-DKPRD_Android/main/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/tanaakkhcm/Documents/work/projects/Uqey/mobile/cloneRentCar/node_modules/@notifee/react-native/android/libs/com/facebook/react/react-android/maven-metadata.xml
     Required by:
         project :app > project :react-native-vision-camera

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 22s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDevelopmentdebug -PreactNativeDevServerPort=8081
Warning: This version only understands SDK XML versions up to 2 but an SDK XML file of version 3 was encountered. This can happen if you use versions of Android Studio and the command-line tools that were released at different times.
Warning: unexpected element (uri:"", local:"extension-level"). Expected elements are <{}codename>,<{}layoutlib>,<{}api-level>
Warning: unexpected element (uri:"", local:"base-extension"). Expected elements are <{}codename>,<{}layoutlib>,<{}api-level>

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:mergeDevelopmentDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:developmentDebugRuntimeClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Searched in the following locations:
       - file:/Users/tanaakkhcm/Documents/work/projects/Uqey/mobile/cloneRentCar/node_modules/react-native/android/com/facebook/react/react-android/maven-metadata.xml
[... 39 lines cut ...]
```

### VisionCamera Version

3.3.1

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

OPEN-9, 2023-10-11:
2.16.2 I also encountered this error.

zzz08900, 2023-10-11:
Last time I tried, you need the latest version or RN to run vision-camera v2.16.2 or v3.
For older RN, stick to 2.15.x

mrousavy, 2023-10-11:
Did something happen with the RN upgrade? Many people are experiencing this, idk if there's something wrong with my `build.gradle`?

mrousavy, 2023-10-11:
Oh yea I think you need RN 0.71 or higher!