margelo/react-native-vision-camera #1979: 🐛 Failed to build/run android project V3

yasiragi, 2023-10-11:
### What's happening?

Hi, When trying to build app using V3 Along with Reanimated 3+ I'm unable to build neither run the app on Android phone haven't tried it on iPhone yet.

### Reproduceable Code

```tsx
Could not determine the dependencies of task ':app:lintVitalReportDevRelease'.
> Could not resolve all task dependencies for configuration ':app:devReleaseRuntimeClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Searched in the following locations:
       - file:/Users/yasiraziz/Projects/carspect-qc/node_modules/react-native/android/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/yasiraziz/Projects/carspect-qc/node_modules/jsc-android/dist/com/facebook/react/react-android/maven-metadata.xml
       - https://dl.google.com/dl/android/maven2/com/facebook/react/react-android/maven-metadata.xml
       - https://www.jitpack.io/com/facebook/react/react-android/maven-metadata.xml
     Required by:
         project :app > project :react-native-vision-camera

"react-native-vision-camera": "^3.3.1"
"react-native": "0.70.6"
"react-native-reanimated": "^3.3.0"
```

### Relevant log output

```shell
Could not determine the dependencies of task ':app:lintVitalReportDevRelease'.
> Could not resolve all task dependencies for configuration ':app:devReleaseRuntimeClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Searched in the following locations:
       - file:/Users/yasiraziz/Projects/carspect-qc/node_modules/react-native/android/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/yasiraziz/Projects/carspect-qc/node_modules/jsc-android/dist/com/facebook/react/react-android/maven-metadata.xml
       - https://dl.google.com/dl/android/maven2/com/facebook/react/react-android/maven-metadata.xml
       - https://www.jitpack.io/com/facebook/react/react-android/maven-metadata.xml
     Required by:
         project :app > project :react-native-vision-camera
```

### Camera Device

```json
-
```

### Device

android

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

caolong0204, 2023-10-11:
i encountered a similar problem

zzz08900, 2023-10-11:
Last time I tried, you need the latest version or RN to run vision-camera v2.16.2 or v3.
For older versions of RN, stick to 2.15.x

mrousavy, 2023-10-11:
This is a build error, but you created a bug report instead. Use the other template.

interstates21, 2023-11-13:
Same issue

yasiragi, 2023-11-13:
@interstates21 its working fine with latest version of React Native. Please try out!

interstates21, 2023-11-15:
We have 0.70.6

For some reason it's very hard to migrate the existing project from 0.70.6 to 0.71 🥲

If it's a relatively small effort - it would be great to fix this crash for 0.70.*, as neither of react-native-vision-camera versions compile for us

vauruk, 2024-01-30:
I have the same error, and I wrote all these issues but any solution works to me.
---package.json -----
    "react-native": "0.70.15",
    "react": "18.1.0",
    "react-native-vision-camera": "^3.8.2",
    "react-native-worklets-core": "^0.2.4",

--> error ------
* What went wrong:
Could not determine the dependencies of task ':react-native-vision-camera:compileDebugAidl'.
> Could not resolve all task dependencies for configuration ':react-native-vision-camera:debugCompileClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Required by:
         project :react-native-vision-camera