margelo/react-native-vision-camera #1982: ❓ Issue with Ultra-wide-angle Camera Detection on Google Pixel Series

algopert, 2023-10-11:
### Question

Hi,

I hope you're doing well. It seems that your current library is unable to detect the ultra-wide-angle camera on Google Pixel 6 Pro and Google Pixel 7 Pro, despite their inclusion of this camera. 

Also I tried it with your sample app as well.

This issue persists across the entire Google phone series, and I've tried both versions 2 and 3 of the library without success. 
Could you please provide some insight into why this detection is currently impossible?

Thank you for your assistance.

### VisionCamera Version

ver 2, 3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-11:
Hi!

Thanks, doing great, hbu? 😄 

VisionCamera has a very solid Devices API that is capable of detecting all Camera devices on your phone that are available for use in your app. Even USB Cameras. If there's a device missing in that list, it is pretty likely that this device is simply not available for use by other apps, and the vendor (Samsung, Huawei, ...) just locked this and doesn't allow you to use this specific Camera device. In other words; if VisionCamera can't detect the Camera, there's nothing you can do about it. It's not even available in a native Android app (except for their stock Samsung/Huawei/... Camera app).

To confirm this, download [camera-samples/Camera2Video](https://github.com/android/camera-samples/tree/main/Camera2Video) and run it on your phone. If the Camera device is not in there either, then again, nothing you can do.

I'm sure that for specific devices such as telephoto or ultra-wide, they usually don't allow you to use those devices directly, but only in combination with other devices (e.g. a "dual-camera" consisting of a normal wide-angle and the ultra-wide-angle), because they only work like that. 

Try logging all devices you get:

```js
const devices = Camera.getAvailableCameraDevices()
devices.forEach((d) => {
  d.formats = null
})
console.log(JSON.stringify(devices, null, 2))
```

closed 2023-10-11 by mrousavy

mrousavy, 2023-10-24:
Try 3.6.3 and let me know if the multi-cam now shows up for you. If you appreciate my efforts & dedication to VisionCamera, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to say thanks. :)

algopert, 2023-10-24:
Thanks for your good news. 
I will.

RiteshJariwala, 2023-12-07:
Hello mrousavy,
I have pixel 7 pro and sometimes its does not find the back camera.
I have  screen open with the button to load camera screen. On first time when i hit button and find out using Camera.getAvailableCameraDevices() i got all back and front camera fine. I close a screen and open it again it does not find back cameras, though it find front camera just fine.

Also when it open up back camera first time its all grainy on the screen. I am using react-native-vision-camera v 3.6.13.

Its there any problem with the google pixel series phones?

mrousavy, 2023-12-07:
No, there shouldn't be - can you try to see if that issue also occurs in the example app here?

RiteshJariwala, 2023-12-08:
> No, there shouldn't be - can you try to see if that issue also occurs in the example app here?

Yeah Marc,

I downloaded your example app and ran it on my google pixel 7 pro. it ran fine without a problem. 

Now problem is i use a codeScanner in my project and when i use codeScanner it throwing me error like as mentioned in this thread:

https://github.com/mrousavy/react-native-vision-camera/issues/1801

I tried to comment that line mentioned as solution, but nothing is working. 

What kind of parameters to pass to the Camera object in case of when we use codeScanner would you please create a sample  for that too test it.

AdamWP, 2024-10-26:
I had a similar issue with a Pixel 7A. 

We can get the minZoom from Camera.getAvailableCameraDevices();. Then, setting the zoom to minZoom in the camera settings will make use of the ultra-wide-angle-camera