margelo/react-native-vision-camera #1984: CodeScanner does not work on earlier versions of Android

igorkalmykov28, 2023-10-11:
### Question

I'm currently facing an issue where the QR code scanning doesn't work on my device with Android 7.1.2 installed.

Has anyone encountered this issue with support for this library on earlier Android versions and can provide any solutions or guidance?

### VisionCamera Version

3.3.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-10-12:
Try changing this line in `node_modules/react-native-vision-camera/android/build.gradle`:

https://github.com/mrousavy/react-native-vision-camera/blob/18e69269d49f38de11693cbec61ee9e280a1f6e9/package/android/build.gradle#L147

to 

```
implementation 'com.google.mlkit:barcode-scanning:17.2.0'
```

then rebuild your app and see if that works then.

igorkalmykov28, 2023-10-12:
@mrousavy I tried to do it, but unfortunately it didn't help.

mrousavy, 2023-10-14:
Can't help you. No logs or build errors or nothing attached.

closed 2023-10-14 by mrousavy