margelo/react-native-vision-camera #1985: 🐛 [V3][iOS][CodeScanner] CodeScanner very slow on Android

spsaucier, 2023-10-11:
### What's happening?

On a Moto G Stylus 2022 Android device, when `codeScanner` is used, the displayed view drops to about 5 fps and any `codeScanner` functionality is slow & delayed.

When codeScanner is disabled (or on iOS at all times), the framerate is excellent.

This is occurring on `react-native-vision-camera@3.3.1`.

### Reproduceable Code

```tsx
Tested in the Example App, with photo/video/audio/frameProcessor all disabled and this codeScanner function:

  const onCodeScanned = useCallback((codes: Code[]) => {
    console.log(codes.length)
  }, [])
```
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Moto G Stylus 2022

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

irekrog, 2023-10-12:
Same issue on Samsung Galaxy A51

FluffyDiscord, 2023-10-12:
can confirm, same on samsung galaxy s23 ultra

GabrielvonPlaten, 2023-10-12:
Exact same issue here on a Samsung Galaxy S21 FE.

zzz08900, 2023-10-12:
How about you guys change the title to "[V3][Android]" ? 
Putting iOS in the title of an Android-only problem gives me headaches.

AlexDM0, 2023-10-12:
Same issue here on a samsung A23

Danushka96, 2023-10-12:
Same Issue OnePlus 9. The first few seconds work fine, then start to drop the fps

zzz08900, 2023-10-12:
FYI there's this frame processor plugin that works with v3:
https://github.com/mgcrea/vision-camera-barcode-scanner

I haven't fully tested myself but it looks pretty promising. Anyone hit by this issue could try using the frame processor plugin above on Android as a workaround.

FluffyDiscord, 2023-10-12:
Yes, that one works nicely. I had to upgrade react native to 0.72.5 though.

celik75, 2023-10-12:
same problem, pr fix of https://github.com/mrousavy/react-native-vision-camera/commit/915ef331d3b3f3630f4b8d5a8cdd7f5fcf30c335 didn't fixed it on andoid

zzz08900, 2023-10-12:
> Yes, that one works nicely. I had to upgrade react native to 0.72.5 though.

Well I just tried it on Android and it crashed the whole app after ~1 sec of code scanning.
Guess I'll be staying with v2 a little longer.

mrousavy, 2023-10-14:
> same problem, pr fix of https://github.com/mrousavy/react-native-vision-camera/commit/915ef331d3b3f3630f4b8d5a8cdd7f5fcf30c335 didn't fixed it on android.
> tested on 2 diffrent androids

@celik75 how did you test that?

sacha-c, 2023-10-16:
> > same problem, pr fix of https://github.com/mrousavy/react-native-vision-camera/commit/915ef331d3b3f3630f4b8d5a8cdd7f5fcf30c335 didn't fixed it on android.
> > tested on 2 diffrent androids
> 
> @celik75 how did you test that?

Also just tested using the [latest commit on main](https://github.com/mrousavy/react-native-vision-camera/commit/e8df7148fa50214f47ff1a41dfa40e7255bc3f8d) and the issue remains
I tested by using [gitpkg](https://gitpkg.vercel.app/) and setting my dependency to `https://gitpkg.now.sh/mrousavy/react-native-vision-camera/package?e8df7148fa50214f47ff1a41dfa40e7255bc3f8d`
The behavior I noticed using this commit is that the camera is fluid at first but becomes _progressively_ laggy (during the first ~2 seconds) & it stabilizes with the same level of slowness as v3.3.1. On v3.3.1 however it was slow from the start

mrousavy, 2023-10-16:
I don't get this. I think the Camera2 APIs are just terrible at this.

@sacha-c can you maybe try this change? Idk if that'll work better

[image]

sacha-c, 2023-10-16:
@mrousavy Gave it a shot, and with this the lagging is fixed! It works smoothly and remains that way.

However after scanning a QR code the camera goes black, this did not used to occur on v3.3.1. 
Looking briefly through the logs I see this after scanning a QR code which may be related
```sh
2023-10-16 13:55:17.455  2699-3982  Camera3-OutputStream    cameraserver                         E  returnBufferCheckedLocked: Stream 0: Error queueing buffer to native window: No such device (-19)
2023-10-16 13:55:17.455  2699-3982  Camera3-Device          cameraserver                         E  Can't return buffer to its stream: No such device (-19)
```
It looks like it may be related to the latest codeScanning changes (not just the diff above, but the diff between v3.3.1..master), but if you would like a separate issue for this let me know if you would like me to create it.

mrousavy, 2023-10-16:
Yea please create a separate issue. Thanks

closed 2023-10-16 by mrousavy

mrousavy, 2023-10-16:
Fixes this in #2009!

spsaucier, 2023-10-16:
Render performance is much better now, thank you. Still seeing the reads buffer/queue and execute for a while after the reads, but vastly improved!

Danushka96, 2023-10-16:
Running smoothly on Android after the new version update (3.4.0)!

mrousavy, 2023-10-17:
Amazing, great to hear my fix works for you guys.

> Still seeing onCodeScanned buffer/queue and execute for a few seconds after the reads, but vastly improved!

Create a separate issue if that's a problem, I don't quite get what this means tbh